istoreos/istoreos #1229: 【BUG】iStore软件包 驿站-边缘计算 可能影响其它软件包安装

bcseputetto, 2024-01-03:
## 1.关于你要提交的问题
iStore里的`驿站-边缘计算`会影响其它软件包的安装。

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
只要安装了`驿站-边缘计算`，安装其它任何软件包都可能会出现
配置一次 owxbyz
```console
Configuring owxbyz.
Collected errors:
 * pkg_run_script: package "owxbyz" postinst script returned status 1.
 * opkg_configure: owxbyz.postinst returned 1.
```
然后所有iStore里的软件可能都会因为这个原因报失败，装是可以装上。
但第三方软件包因为判断问题会安装失败。

### (2) 路由器型号和固件版本
版本`2023122916`
在 `x86_64` 下测试，`aarch64`未测试

### (3) 详细日志和/或截图

[image]

ks-bao, 2024-01-05:
我让玩物科技的人来负责哈。

ionewu, 2024-02-02:
> ## 1.关于你要提交的问题
> iStore里的`驿站-边缘计算`会影响其它软件包的安装。
> 
> Q：是否用关键词搜索了issue? (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> 只要安装了`驿站-边缘计算`，安装其它任何软件包都会出现 配置 owxbyz 接着报错
> 
> ```
> Configuring owxbyz.
> Collected errors:
>  * pkg_run_script: package "owxbyz" postinst script returned status 1.
>  * opkg_configure: owxbyz.postinst returned 1.
> ```
> 
> 然后所有iStore里的软件可能都会因为这个原因报失败，装是可以装上。 但第三方软件包可能会因为判断问题而终止脚本后，安装失败。
> 
> 卸载`驿站-边缘计算`后问题可以得到解决
> 
> ### (2) 路由器型号和固件版本
> 版本`2023122916` 在 `x86_64` 下测试，`aarch64`未测试
> 
> ### (3) 详细日志和/或截图
> ![owxbyz](https://private-user-images.githubusercontent.com/29139573/293957821-a7d872ba-8621-4240-a433-b04ca315feb2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4NDQxMTQsIm5iZiI6MTcwNjg0MzgxNCwicGF0aCI6Ii8yOTEzOTU3My8yOTM5NTc4MjEtYTdkODcyYmEtODYyMS00MjQwLWE0MzMtYjA0Y2EzMTVmZWIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAyVDAzMTY1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjZGY4NTk4OWRlNTcxZGQ0MGMzOTZmYTU3N2QzMzNmOWMyNDkzNzI0OTFiYzliZjNkYWZmMzk5YWJiYzVjNGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FK-PQlQ9k_-vzWCFNwPG9RRd8g108fe8OqY80kkUbKg)

我们测试未发现相关问题，也未收到其他用户有类似反馈。
请卸载重装一下驿站，再看看是否还有相同问题，如仍有问题，请联系我。

bcseputetto, 2024-02-02:
@ionewu 
[image]

bcseputetto, 2024-02-02:
@ionewu 
[image]
装完驿站后，再装其它的，都会报这个错误，这百分百重现的，你们怎么会测试不出来？

ionewu, 2024-02-02:
@bcseputetto  方便的话，辛苦留一下联系方式，我们单独联系。
我们大最用户均未遇到类似问题，希望可以具体问题具体分析

bcseputetto, 2024-02-02:
@ionewu 请你给一个邮箱，我往你邮箱发联系方式，QQ，微信，还是tg？

ionewu, 2024-02-02:
> @ionewu 请你给一个邮箱，我往你邮箱发联系方式，QQ，微信，还是tg？

好的，辛苦发邮件到：zhaoganjie@ionewu.com

bcseputetto, 2024-02-02:
@ionewu 已经发了联系方式到邮箱了，注意查看

bcseputetto, 2024-02-02:
远程调试了，等待玩物科技更新。
更新前暂时可以用以下办法临时解决
```console
echo "exit 0" >> /usr/lib/opkg/info/owxbyz.postinst
```

closed 2024-02-02 by bcseputetto

jjm2473, 2024-02-02:
这个错误不会导致其他插件安装失败，只是如果玩物没配置过，会一直报这个错误。

bcseputetto, 2024-02-02:
> 这个错误不会导致其他插件安装失败，只是如果玩物没配置过，会一直报这个错误。

是的不会导致其它软件安装失败，但会一直报这个错误，我有 .run 软件包有判断错误退出脚本，所以可能会导致 .run 安装包安装失败，所以最好消除影响

jjm2473, 2024-02-02:
有些人之所以没能复现这个错误，是因为他们已经配置好了玩物。要复现也很简单，删掉`/etc/config/owxbyz`，然后重新安装玩物（卸载重装）