istoreos/istoreos #1230: 外接usb硬盘盒掉盘

ghostlyheart, 2024-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前可能因为usb供电不足总是会出现usb设备盘符变更sdb变为sdc一段时间又变为sdb后来增加了独立供电usb拓展，现在运行一会网心云直接消失了，只有在升级系统和重装系统后才会重新出现

### (2) 路由器型号和固件版本

A：x86 J4125  固件版本22.03.5 2023122916

### (3) 详细日志和/或截图

A：[ 4218.159649] usb 2-2.3: device not accepting address 10, error -62
[ 4218.319623] EXT4-fs error: 179329 callbacks suppressed
[ 4218.319627] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4218.319737] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830863: comm sh: reading directory lblock 0
[ 4218.320457] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6706632: comm softdog.sh: reading directory lblock 0
[ 4218.321192] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6819755: comm sh: reading directory lblock 0
[ 4218.321408] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4218.321742] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6819755: comm sh: reading directory lblock 0
[ 4218.322101] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4218.322107] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4218.322112] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4218.322119] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830852: comm sh: reading directory lblock 0
[ 4220.879903] EXT4-fs warning: 2218 callbacks suppressed
[ 4220.879907] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.885809] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.890910] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.890932] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.890938] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.893878] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.897319] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.897479] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.897777] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4220.903405] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4223.329812] EXT4-fs error: 172772 callbacks suppressed
[ 4223.329816] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6707077: comm softdog.sh: reading directory lblock 0
[ 4223.329888] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6819755: comm sh: reading directory lblock 0
[ 4223.330191] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.330207] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.330214] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.330218] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.330223] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830852: comm sh: reading directory lblock 0
[ 4223.330227] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830863: comm sh: reading directory lblock 0
[ 4223.330380] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.330386] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4223.619739] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[ 4225.891530] EXT4-fs warning: 2108 callbacks suppressed
[ 4225.891535] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.891782] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.892645] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.897939] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.898187] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.900904] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.906454] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.910929] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.910980] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4225.910989] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4228.339762] EXT4-fs error: 168732 callbacks suppressed
[ 4228.339766] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830863: comm sh: reading directory lblock 0
[ 4228.340398] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830863: comm sh: reading directory lblock 0
[... 32 lines cut ...]
[ 4233.354341] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4234.659651] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command
[ 4235.910912] EXT4-fs warning: 1930 callbacks suppressed
[ 4235.910916] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.910938] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.910944] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.914022] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.918064] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.918208] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.919262] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.926247] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.926561] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4235.926870] EXT4-fs warning (device sdb1): dx_probe:792: inode #6707078: lblock 0: comm softdog.sh: error -5 reading directory block
[ 4238.359721] EXT4-fs error: 155217 callbacks suppressed
[ 4238.359724] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6830863: comm sh: reading directory lblock 0
[ 4238.360486] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4238.360618] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6819755: comm sh: reading directory lblock 0
[ 4238.360955] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6707077: comm softdog.sh: reading directory lblock 0
[... cut ...]
[ 4238.360987] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6706625: comm softdog.sh: reading directory lblock 0
[ 4238.360993] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6706625: comm softdog.sh: reading directory lblock 0
[ 4238.361303] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6831120: comm sh: reading directory lblock 0
[ 4238.361761] EXT4-fs error (device sdb1): __ext4_find_entry:1579: inode #6819755: comm sh: reading directory lblock 0
[ 4239.939829] xhci_hcd 0000:00:15.0: Timeout while waiting for setup device command

hx801217, 2024-01-09:
硬盘盒问题，我使用联想的廉价硬盘盒也有这问题，换了拆机的希捷、西数硬盘盒就不会发生这问题了。尤其是PCDN这样大量的读写场景，个人体验还是选AS祥硕主控芯片的硬盘盒，杰微和威盛的主控都不行。

ghostlyheart, 2024-01-09:
> 硬盘盒问题，我使用联想的廉价硬盘盒也有这问题，换了拆机的希捷、西数硬盘盒就不会发生这问题了。尤其是PCDN这样大量的读写场景，个人体验还是选AS祥硕主控芯片的硬盘盒，杰微和威盛的主控都不行。

感谢解惑

closed 2024-01-09 by ghostlyheart