istoreos/istoreos #1235: 简单网络管控 白名单控制的问题

houhai2049, 2024-01-05:
## 1.  详细叙述
使用istore 插件 ：简单网络管控 的功能，在管控 -> 访问控制页面中设置 “限制模式”为白名单，添加mac设备后一切正常；当路由重启后再白名单中的设备无法正常上网，编辑访问控制页面后保存 白名单恢复正常(电脑能正常上网)

### (2) 路由器型号和固件版本
x86硬件；固件版本：22.03.5 2023122916

### (3) 详细日志和/或截图
[系统日志.txt](https://github.com/istoreos/istoreos/files/13837343/default.txt)

jjm2473, 2024-01-05:
每次重启路由器白名单都失效吗？

houhai2049, 2024-01-05:
> R

是的，已经试了很多次，每次重启就失效

houhai2049, 2024-01-17:
> 每次重启路由器白名单都失效吗？

又发现开启白名单后端口映射失效了

jjm2473, 2024-01-18:
我测试了，没发现白名单会失效。端口映射倒是确实被影响了。

jjm2473, 2024-01-19:
istore更新了，看看还有没有问题

houhai2049, 2024-01-29:
已经更新到最新的版本iStoreOS 22.03.6 2024012613，问题依然存在：
1. 设置白名单重启后白名单内的mac地址均无法上网
2. 端口映射依然不起作用
3. 简单网络管控功能 不在白名单中的设备依然可以通过ipv6上网

houhai2049, 2024-01-29:
> 我测试了，没发现白名单会失效。端口映射倒是确实被影响了。

试一下是不是ipv6的能访问，ipv4无法访问；我这里百度、钉钉都可以，但是不支持ipv6的都无法连接

jjm2473, 2024-02-01:
> 已经更新到最新的版本iStoreOS 22.03.6 2024012613，问题依然存在：

靠，我说的是istore里面的插件，谁跟你说istoreos？istore是应用商店，istoreos是固件，两个东西不一样OK？

jjm2473, 2024-02-01:
更新istore里面的插件！

closed 2024-02-23 by jjm2473