istoreos/istoreos #1236: 启用 FullCone-NAT导致ksoftirqd/2占用CPU过高，致使网络不通，机器死机

bupahaibao, 2024-01-05:
启用 FullCone-NAT导致ksoftirqd/2占用CPU过高，致使网络不通，机器死机

(1) 具体问题
设备为R86S，下面挂了两台网心云，开启了FullCone-NAT。短则半小时，长则不过一天。当活动链接数量上升到4万往上时，设备iostreos首页就会看到CPU占用超过了50%，甚至百分之八九十，极端情况能达到100%，设备就会卡死重启。不过大多数情况是延迟，丢包，网络不通。使用top -n 10命令查看CPU占用会发现ksoftirqd/2占用能达到25%，也就是把CPU4个核心中的一个核心用到顶了，基本这时候网络也就不同了，大概率要死机了。通过查询资料（https://github.com/coolsnowwolf/lede/issues/6126）发现按照这个帖子关闭FullCone-NAT即可正常。网上搜索”openwrt ksoftirqd“会发现一大堆这样的问题，但是近三年无人解决。

(2) 路由器型号和固件版本
R86S 
固件版本：[istoreos-22.03.5-2023122916-x86-64-squashfs-combined-efi.img.gz

(3) 详细日志和/或截图
由于我是小白，不懂得日志这些东西，不过发生这种情况日志都一切正常。晚上相关帖子表述也可证明我的说法。希望开发者能够参考晚上相关问题，尽早解决这个bug。

bupahaibao, 2024-01-05:
[image]
[image]
可以看到，开启了FullCone-NAT，ksoftirqd/2占用开始缓慢升高，因为我挂着网心云，活动链接肯定会慢慢上升，最终ksoftirqd/2到达25%，设备开始丢包死机。

bupahaibao, 2024-01-05:
[image]

bupahaibao, 2024-01-05:
[image]

bupahaibao, 2024-01-05:
可以看到，开启了FullCone-NAT，ksoftirqd/2占用开始缓慢升高，因为我挂着网心云，活动链接肯定会慢慢上升，最终ksoftirqd/2到达25%，设备开始丢包死机。

jjm2473, 2024-01-07:
用什么方式可以产生这么多连接数？也就是测试方法

bupahaibao, 2024-01-07:
> 用什么方式可以产生这么多连接数？也就是测试方法

这个怎么说呢，作为一个小白，我只能说买两个网心云挂上，总会产生这么多连接。或者我建议你把istoreos商店里的“网心云”“驿站”“蓝山流量宝”等等类似的docker都给部署上，挂机块硬盘，过几天也应该能有如此多的连接数量。你还可以寻找一下UDP并发测试工具之类的。

jjm2473, 2024-01-08:
关掉fullconenat以后，连接数还会跑到4万以上吗

bupahaibao, 2024-01-08:
> 关掉fullconenat以后，连接数还会跑到4万以上吗

不会。
但是解决这个问题的方向，首先我认为你们要优化一下这个udp连接的老化时间。建议向爱快借鉴一下这一块儿。
第二，尽快提升你们的这个Linux内核。以解决你们这个网卡软中断亲和力的问题。所有的中断请求都在一个核心上堆积，多核支持太差。
对于网心云这一块儿你们还是要借鉴一下lede大神大神那边那个turbo acc加速工具里面那个fullcone-nat高性能模式。

jjm2473, 2024-01-08:
中断亲和问题跟linux版本无关吧？非x86的的都是有固定脚本配置。x86的性能比较高，一般单核也足够处理网络，如果实在有问题，网络-接口-全局设置里面打开“数据包引导”也能分散CPU压力（配置以后重启下）。

“udp连接的老化时间”如果内核有参数可调，那可以试试，如果是linux内核本身的问题，就只能反馈给linux社区了。

fullcone-nat高性能模式可以考虑支持，之后移植了先给你测试下有没有效果。

bupahaibao, 2024-01-08:
> 中断亲和问题跟linux版本无关吧？非x86的的都是有固定脚本配置。x86的性能比较高，一般单核也足够处理网络，如果实在有问题，网络-接口-全局设置里面打开“数据包引导”也能分散CPU压力（配置以后重启下）。
> 
> “udp连接的老化时间”如果内核有参数可调，那可以试试，如果是linux内核本身的问题，就只能反馈给linux社区了。
> 
> fullcone-nat高性能模式可以考虑支持，之后移植了先给你测试下有没有效果。

太好了，乐意效劳。

jjm2473, 2024-01-09:
@bupahaibao 你用的是efi还是非efi固件？

jjm2473, 2024-01-09:
@bupahaibao  这是efi测试固件 https://we.tl/t-QDwEQ32FZQ

bupahaibao, 2024-01-09:
> @bupahaibao 你用的是efi还是非efi固件？

EFI的。

bupahaibao, 2024-01-09:
> @bupahaibao 这是efi测试固件 https://we.tl/t-QDwEQ32FZQ

好的，十分感谢，我得周末回家试试。我担心在外尝试家里断网，家中人无法上网。

bupahaibao, 2024-01-09:
> @bupahaibao 这是efi测试固件 https://we.tl/t-QDwEQ32FZQ

请问这个不会清除其他配置，以及影响后续正式版的升级更新吧？

jjm2473, 2024-01-09:
@bupahaibao 兼容的，除非你自己选择不保留配置

bupahaibao, 2024-01-09:
> @bupahaibao 兼容的，除非你自己选择不保留配置

[image]
[image]
冒险远程升级了一下，选择了高性能nat模式，确实起作用。你可以在稳定版中发布了。平时活动链接数这么多早就死机了，更新你这个高性能nat之后CPU很淡定。

bupahaibao, 2024-01-09:
> @bupahaibao 这是efi测试固件 https://we.tl/t-QDwEQ32FZQ

[image]
[image]
可以，可以，现在破了10万连接数也不疼不痒。这个高性能nat的原理是啥呀？

jjm2473, 2024-01-09:
@bupahaibao 我也不知道原理是啥，只知道没有使用外部的内核模块，而是对内核打补丁，从代码来看，也没有自己再维护一个列表，可能因为这个，所以性能较高。代码都是从lean仓库拷贝的，可能要问作者了 https://github.com/coolsnowwolf/lede/commit/193db6f18680f2354aa7daea11c2f61cc037e343#diff-5a72816901f7a9d19f59ef8e75113af09ddbbd179f8eb33a51fdf6ac6031d908 。

bupahaibao, 2024-01-09:
> @bupahaibao 我也不知道原理是啥，只知道没有使用外部的内核模块，而是对内核打补丁，从代码来看，也没有自己再维护一个列表，可能因为这个，所以性能较高。代码都是从lean仓库拷贝的，可能要问作者了 [coolsnowwolf/lede@193db6f](https://github.com/coolsnowwolf/lede/commit/193db6f18680f2354aa7daea11c2f61cc037e343) 。

可以的，可以的，啥时候你更新了这个高性能nat正式版。更新的那个日志着重说一下呀，到时候我看到了我就更新到正式版去。如果你们不打算集成到正式版里，我就留你这个测试版养老了。

bupahaibao, 2024-01-09:
已通过作者给出的内测版解决问题。

closed 2024-01-09 by bupahaibao

jjm2473, 2024-01-09:
如果测试没问题，一般下个版本发布时就会带上

bupahaibao, 2024-01-10:
> 如果测试没问题，一般下个版本发布时就会带上

没问题，完全没问题。

TomFoxLee, 2024-01-15:
> > @bupahaibao 这是efi测试固件 https://we.tl/t-QDwEQ32FZQ
> 
> ![Screenshot_20240109_194649_com example routerapp](https://private-user-images.githubusercontent.com/67363931/295196598-bcda6027-bd7a-4b65-9311-3318443d6e15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyNzk2OTIsIm5iZiI6MTcwNTI3OTM5MiwicGF0aCI6Ii82NzM2MzkzMS8yOTUxOTY1OTgtYmNkYTYwMjctYmQ3YS00YjY1LTkzMTEtMzMxODQ0M2Q2ZTE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE1VDAwNDMxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkY2I2M2NhN2U5MzlhYjY4NjRjODExZmM3NjE3Y2NiYjk5ZmY1ZjBjNTE5ZDJjOTA2ODhiOTNmZGZmZWNjOTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._cKSZ0oNDExQ7KbikIyPg6gMHzvsCgTZfLs758kVPUY) ![Screenshot_20240109_194703_com example routerapp](https://private-user-images.githubusercontent.com/67363931/295196641-d53674c1-15ef-4e7c-aec4-f584b16fbc7c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUyNzk2OTIsIm5iZiI6MTcwNTI3OTM5MiwicGF0aCI6Ii82NzM2MzkzMS8yOTUxOTY2NDEtZDUzNjc0YzEtMTVlZi00ZTdjLWFlYzQtZjU4NGIxNmZiYzdjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE1VDAwNDMxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmZmVhN2YxYWQ0MzhkNTdlMzUzNDExZTZlNTIxOWMwYzY0MDRmODg5ZTc5ZjcxMDg2MGJlMDk4NDQxZjcwMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-9kBnqCyR1uThRWg1x4B6zJM965CG_9glkx40IU8Ry8) 可以，可以，现在破了10万连接数也不疼不痒。这个高性能nat的原理是啥呀？

請問一下，這個活動連接數上限，是在什麼地方修改啊？我的上限顯示是65535。雖然平常好像用不到這麼多連接數，但還是想知道在哪裡調整。。。

jjm2473, 2024-01-15:
@TomFoxLee 在`/etc/sysctl.d/`里面，你把12-开头的文件复制一份13-开头，然后编辑新文件即可。不过一般不需要修改连接数限制，如果真有那么多连接数，还是考虑下是不是受到攻击了。

wobushizhx, 2024-12-07:
请问，我的版本
iStoreOS 22.03.7 2024102509，也是出现ksoftirqd进程cpu占用很高，按照前排老哥的方法：网络-接口-全局设置里面打开“数据包引导”也做了，没有明显的改善，请问现在有更好的办法吗？