istoreos/istoreos #1237: 是否可以添加dhcp指定客户端获取特定网关的功能

whetherTsmile, 2024-01-05:
现在 只能通过在 /etc/config/dhcp 文件里手动添加 tag 
config host
	option name 'ps5'
	option dns '1'
	option mac 'xxxx'
	option ip '192.168.5.26'
	option tag 'openclash'

config tag 'openclash'
	list dhcp_option '3,192.168.5.2'
	list dhcp_option '6,192.168.5.2'
可以在ui上增加这个选项吗， 添加新设备比较方便

jjm2473, 2024-01-08:
lan口的dhcp配置不是本来就有个自定义dhcp选项吗？

jjm2473, 2024-01-08:
哦，你是需要对不同的客户端指定不同的网关和dns

jjm2473, 2024-01-08:
你到openwrt官方项目反馈吧，这些功能都是openwrt自带的，istoreos也没改。
https://github.com/openwrt/openwrt

whetherTsmile, 2024-01-15:
> 你到openwrt官方项目反馈吧，这些功能都是openwrt自带的，istoreos也没改。 https://github.com/openwrt/openwrt

好的 感谢

closed 2024-01-15 by whetherTsmile

jjm2473, 2024-01-18:
我看openwrt官方已经加上了host关联tag，但是却没tag的配置页面

jjm2473, 2024-01-19:
iStoreOS 实现了 tag 配置页面，现在功能完整了。

jjm2473, 2024-01-19:
后台具体位置“网络”-“DHCP/DNS”-“标签”，配置前仔细阅读其中的说明。

顺便提一下iStoreOS中怎么默认指派非本机网关给内网主机（例如iStoreOS作为旁路由时，指派主路由作为网关），在“网络”-“接口”，在“lan”后面点击编辑，在弹出的对话框选择“DHCP 服务器”-“高级设置”，其中的“DHCP 选项”与前面提到的“标签”中的“DHCP 选项”一样。

“DHCP 选项”的语法：
* “3,192.168.100.2”表示指派 192.168.100.2 作为网关。（注意3和英文逗号）
* “6,192.168.100.2”表示指派 192.168.100.2 作为 DNS 服务器。（注意6和英文逗号）
* 其他选项不常用就不解释了

典型使用场景（必须iStoreOS提供DHCP服务，以下皆假设主路由IP是192.168.100.1，旁路由IP是192.168.100.2，掩码都是24位）：
* iStoreOS作为旁路由，提供翻墙服务，内网部分主机需要翻墙，大部分主机都不需要翻墙
  1. 打开“网络”-“接口”，在“lan”后面点击编辑，在弹出的对话框选择“DHCP 服务器”-“高级设置”，其中的“DHCP 选项”增加两条“3,192.168.100.1”和“6,192.168.100.1”
  2. 打开“网络”-“DHCP/DNS”-“标签”，在“添加”按钮左边输入“t_proxy”，点击“添加”按钮，然后也是“DHCP 选项”输入两条“3,192.168.100.2”和“6,192.168.100.2”，保存应用
  3. 打开“网络”-“DHCP/DNS”-“静态地址分配”，点击“添加”（如果之前已经配置了要翻墙的主机，那编辑相应条目即可），输入主机名称（可以用英文大小写，数字，下划线，必填），选择要翻墙的主机的MAC；如果不需要固定IP，“IPv4 地址”可以选择未指定；最关键的是在“标签”选项中选择“t_proxy”（如果没出现，保存下刷新页面再回来选）。然后保存应用即可。对于每一个需要翻墙的设备都要添加。内网主机重新插网线或连wifi，应该就会获取到指定的网关和DNS了。
* iStoreOS作为主路由，不提供翻墙服务，由旁路由提供翻墙服务，内网部分主机需要翻墙，大部分主机都不需要翻墙。步骤跟上面一样，但是跳过第1步（因为默认网关就是主路由）。

whetherTsmile, 2024-05-31:
> iStoreOS 实现了 tag 配置页面，现在功能完整了。

我超，看的晚了，作者牛逼，我直接一个升级

jjm2473, 2025-06-30:
关掉ipv6