istoreos/istoreos #1240: 关于移动电信双wan口的各种小bug。

bupahaibao, 2024-01-06:
1、首页流量只显示wan口信息，不显示wan1口信息。（建议增加wan1口显示模块）
[image]
2、使用两个wan口，首页DNS总是会跳错，提示未联通。
[image]
3、使用两个wan口，istoreos商店里的实时流量不起作用，进入实时流量页面永远都是0，不出任何数据。
[image]
4、使用两个wan口，istoreos商店里面的软件下载总是无法连通网址。（建立设置无论几条线，商店都走第一个wan口，或者告诉我们商店用设备本身的哪个地址访问的，我们自己做策略）
[image]
5、使用两个wan口，两个lan口。upnp功能无法指定比如说lan对应到wan，lan1对应到wan1。默认会把lan和lan1都给映射到第一个wan口，很烦。

jjm2473, 2024-01-08:
这是你没配置好路由啊，双wan不是拨两个号就行了，还需要配置策略路由。

bupahaibao, 2024-01-08:
> 这是你没配置好路由啊，双wan不是拨两个号就行了，还需要配置策略路由。

理论上是不需要配置路由的。两条线路不同网段均会在路由表里体现，不需要人为干预。（证据就是两条线不安装mwan3也可以正常上网，流量会根据路由表走，会有默认规则的优先级，比如说跳数）但是现在istore商店在双线路之下确实出现了问题，我可以手动制定策略，所以我想问问istore商店默认使用哪个地址对外访问？lan的网关？wan口地址？还是lo0地址？走的ipv4还是v6？搞不清楚这些，我也没法指定策略。

jjm2473, 2024-01-08:
既然你知道路由表和跳数，那么看 `route -n` 命令的输出应该就知道了

jjm2473, 2024-01-08:
双wan上网肯定不会有问题，你这有问题肯定是配置错误。双wan会出问题的地方，在从外网访问路由器时会出现，如果不配置策略路由的话。还有，你要lan1对wan1，那也要策略路由，因为要根据源ip路由。

bupahaibao, 2024-01-08:
> 双wan上网肯定不会有问题，你这有问题肯定是配置错误。双wan会出问题的地方，在从外网访问路由器时会出现，如果不配置策略路由的话。还有，你要lan1对wan1，那也要策略路由，因为要根据源ip路由。

这个策略只能在mwan3里面去匹配流量并指定，还真不好自己设置静态路由。我的wan、wan2获取的IPv4地址不是公网，是运营商给的大内网地址，但是也是会随机变动的，我无法在路由上指定下一跳，除非你们这个系统支持指定下一跳出口。
当时很明显，在你们的lan口设置下一跳网关不支持直接写接口。
[image]

hx801217, 2024-01-09:
不是固件本身的问题，请参考以下链接文章进行负载均衡设置。
https://www.gebi1.com/thread-295572-1-1.html

bupahaibao, 2024-01-09:
> 不是固件本身的问题，请参考以下链接文章进行负载均衡设置。 https://www.gebi1.com/thread-295572-1-1.html

我问的问题不是你发的这个帖子。并且你发的这个我设计的比这个精细多了。并且你这个帖子上还没有ipv6的负载。

jjm2473, 2024-01-10:
所以用mwan3，mwan3应该支持指定接口作为出口吧，不需要网关ip

jjm2473, 2024-01-10:
> 你们的lan口设置下一跳网关不支持直接写接口

你看看哪个openwrt支持？而且这是网关，网关是全局的好吧，不是你在lan口指定的网关就会路由来自lan的流量。我怀疑你都不了解网关的意义，正常情况下，只应该在出口的网口上配置网关，因为网口掉线的时候这个网关就应该失效，而不是把网关配置到lan口。

常规的路由是只根据目的ip判断的，你说的需求都需要使用策略路由实现。要么使用mwan3，要么在“网络”-“路由”里面配置策略路由。

jjm2473, 2024-01-10:
我怀疑你一开始提到的问题，全都是因为你在lan口配置网关导致的。

bupahaibao, 2024-01-10:
> 我怀疑你一开始提到的问题，全都是因为你在lan口配置网关导致的。

lan口网关我都是一直留空的。如果是单wan口，lan网关那里会自动显示wan以及wan口ip的字眼。
其次，lan设置那里填写网关是全局啥意思？lan可不代表全局，我还建立了lan区域同等级的许多个dmz区域，lan可代表不了其他区域。我肯定不理解lan口设置那个网关是啥意思，因为lan口区域下辖管的设备网关默认都是lan的地址。那么lan设置里面的网关设置也只能是lan的下一跳地址，类似默认路由指下一跳的功能。
另外你说哪个openwrt默认路由支持下一跳指向出口？openwrt我确实没见过，但是华为华山企业路由器都支持这种路由的指定，将流量明确指向下一跳出口。由于wan口地址多变动，99%的家庭wan口地址都会变，路由指向出口这也是刚需，别人openwrt不行，你就不行吗，你就不能第一个吃螃蟹吗。istoreos肯定要做大做强，如果你是开发者甚至是作者，你们考虑问题应该考虑我说的和不合理，应不应该，而不是让我去看看别人家有没有，来作为自己不作为的理由。
当然可以依靠mwan3去解决这个问题，但是mwan3多久没更新了，有的时候生不生效都很难评估。我设置好了规则，两个网心云分别走wan口电信和wan2移动，但是网心云控制台都显示电信。最终我还是新创建了wan2口防火墙，来进行逻辑上隔离电信和移动流量达成目的的。mwan3本就是一个年久失修的插件，对IPv6的支持像一坨屎。制定的规则生不生效，mwan3检测接口在不在线就靠命。各种问题罄竹难书。你们不能依靠这种玩意。

bupahaibao, 2024-01-10:
> 这些基本的网络功能都是openwrt自带的，如果网络功能不满足你的需求，那你可以去openwrt反馈这个问题。
> 
> > 99%的家庭wan口地址都会变
> 
> 确实都会变啊，但是有几个人需要配置下一跳出口，有0.1%吗？也就你这种多wan的情况下才有可能需要策略路由。

那你换一种思路，是不是多wan口的人都100%需要。这不一下就有改的动力了？多播的人都需要，对不对，都需要。就像LEDE大神的turbo acc功能，多少人因为这个功能用他的固件？抢人的都是这些看着可有可无的小细节。

jjm2473, 2024-01-10:
> lan设置那里填写网关是全局啥意思？

不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由，这个路由规则并不会针对来自你配置的逻辑接口上的流量，这个路由规则跟配置的逻辑接口的关系是逻辑接口下线时会删掉此路由规则。也就是说，网关是全局生效的，不会因为你配置在哪个接口上就只处理这个接口的流量。

> 多wan的人都需要

前面也提到了，mwan3就是为了应对多wan场景的，既然mwan3有问题，也应该让openwrt修复，而不是又造个轮子。

bupahaibao, 2024-01-10:
> > lan设置那里填写网关是全局啥意思？
> 
> 不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由，这个路由规则并不会针对来自你配置的逻辑接口上的流量，这个路由规则跟配置的逻辑接口的关系是逻辑接口下线时会删掉此路由规则。也就是说，网关是全局生效的，不会因为你配置在哪个接口上就只处理这个接口的流量。
> 
> > 多wan的人都需要
> 
> 前面也提到了，mwan3就是为了应对多wan场景的，既然mwan3有问题，也应该让openwrt修复，而不是又造个轮子。

你还真是一头犟驴。“不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由”那如果是这样的话，请教一下lan口和lan1口那个网关的地方我都写了电信移动不同的wan口地址。难道并不是两个网关地址？对应管控的难倒不是lan和lan1口的流量吗？如果不是，那就是你系统有问题。你所谓理解的全局，仅仅是在全局route表里添加一个默认网关路由。但是这条默认路由它不叫全局路由。全局路由是[0.0.0.0 0 下一跳]这叫全局，囊括所有。如果说我的lan是192.168.1.0/24，lan1是192.168.2.0/24。那么我在这两个接口下分别加对应的网关，路由表里添加的路由是[192.168.1.0 0.0.0.255 0 下一跳]和[192.168.2.0 0.0.0.255 0 下一跳]这两条路由。这两条路由可不叫全局路由。强调路由表叫全局路由表也毫无意义，哪个路由器IPv4还能分成两个路由表吗？每个区域设置每个区域的网关就单独管控这个区域，毫无疑问，如果不是，就是你们系统有问题。
其次，什么叫重新造轮子呀。咱就是说华三，华为，思科这些路由器都能指下一跳接口。我想你们这个能不能支持，对不对？那你说愿意做，就愿意做，不愿意做你就明确说不做呗。我有啥损失，跟我有啥好犟的。那确实我也可以找其他办法曲线救国嘛。

bupahaibao, 2024-01-10:
> > lan设置那里填写网关是全局啥意思？
> 
> 不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由，这个路由规则并不会针对来自你配置的逻辑接口上的流量，这个路由规则跟配置的逻辑接口的关系是逻辑接口下线时会删掉此路由规则。也就是说，网关是全局生效的，不会因为你配置在哪个接口上就只处理这个接口的流量。
> 
> > 多wan的人都需要
> 
> 前面也提到了，mwan3就是为了应对多wan场景的，既然mwan3有问题，也应该让openwrt修复，而不是又造个轮子。

你还真是一头犟驴。“不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由”那如果是这样的话，请教一下lan口和lan1口那个网关的地方我都写了电信移动不同的wan口地址。难道不是两个网关地址，对应管控的不是lan和lan1口的流量吗？如果不是，那就是你系统有问题。你所谓理解的全局，仅仅是在全局route表里添加一个默认网关路由。但是这条默认路由它不叫全局路由。全局路由是[0.0.0.0 0 下一跳]这叫全局，囊括所有。如果说我的lan是192.168.1.0/24，lan1是192.168.2.0/24。那么我在这两个接口下分别加对应的网关，路由表里添加的路由是[192.168.1.0 0.0.0.255 0 下一跳]和[192.168.2.0 0.0.0.255 0 下一跳]这两条路由。这两条路由可不叫全局路由。强调路由表叫全局路由表也毫无意义，哪个路由器IPv4还能分成两个路由表吗？每个区域设置每个区域的网关就单独管控这个区域，毫无疑问，如果不是，就是你们系统有问题。
其次，什么叫重新造轮子呀。咱就是说华三，华为，思科这些路由器都能指下一跳接口。我想你们这个能不能支持，对不对？那你说愿意做，就愿意做，不愿意做你就明确说不做呗。我有啥损失，跟我有啥好犟的。那确实我也可以找其他办法曲线救国嘛。

bupahaibao, 2024-01-10:
> > lan设置那里填写网关是全局啥意思？
> 
> 不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由，这个路由规则并不会针对来自你配置的逻辑接口上的流量，这个路由规则跟配置的逻辑接口的关系是逻辑接口下线时会删掉此路由规则。也就是说，网关是全局生效的，不会因为你配置在哪个接口上就只处理这个接口的流量。
> 
> > 多wan的人都需要
> 
> 前面也提到了，mwan3就是为了应对多wan场景的，既然mwan3有问题，也应该让openwrt修复，而不是又造个轮子。

你还真是一头犟驴。“不管哪个逻辑接口上配置的网关，都会在全局的route表里面添加一个默认网关路由”那如果是这样的话，请教一下lan口和lan1口那个网关的地方我都写了电信移动不同的wan口地址。难道不是两个网关地址，对应管控的不是lan和lan1口的流量吗？如果不是，那就是你系统有问题。你所谓理解的全局，仅仅是在全局route表里添加一个默认网关路由。但是这条默认路由它不叫全局路由。全局路由是[0.0.0.0 0 下一跳]这叫全局，囊括所有。如果说我的lan是192.168.1.0/24，lan1是192.168.2.0/24。那么我在这两个接口下分别加对应的网关，路由表里添加的路由是[192.168.1.0 0.0.0.255  下一跳]和[192.168.2.0 0.0.0.255  下一跳]这两条路由。这两条路由可不叫全局路由。强调路由表叫全局路由表也毫无意义，哪个路由器IPv4还能分成两个路由表吗？每个区域设置每个区域的网关就单独管控这个区域，毫无疑问，如果不是，就是你们系统有问题。
其次，什么叫重新造轮子呀。咱就是说华三，华为，思科这些路由器都能指下一跳接口。我想你们这个能不能支持，对不对？那你说愿意做，就愿意做，不愿意做你就明确说不做呗。我有啥损失，跟我有啥好犟的。那确实我也可以找其他办法曲线救国嘛。

jjm2473, 2024-01-11:
我说的“网关”是你在lan口的“IPv4网关”填的这个值，不是“IP/掩码”产生的路由。都说了，“网关”是全局的，这是openwrt，甚至是linux的默认行为，因为如果不是全局，那就涉及到策略路由了，早期linux都没策路由这种东西。策略路由也有其他页面可以设置。

我刚想到，其实你的需求可以不用mwan3实现。wan1接口的高级设置可以选择你要覆盖的路由表，例如填一个新的1001路由表，默认网关就会写到1001路由表，然后在“网络”-“路由”里面，通过路由规则，将你的lan1关联到新的路由表1001，这样就能实现策略路由，而且也不需要写死网关的ip地址。wan2，lan2也一样操作。而且这个支持ipv6路由表.

zctmdc, 2024-01-24:
弱弱的问 是这样吗

[image]

[image]

jjm2473, 2024-01-24:
@zctmdc 如果你打算自己配置策略路由，那就不要使用mwan3了，以免规则冲突。

zctmdc, 2024-01-25:
> @zctmdc 如果你打算自己配置策略路由，那就不要使用mwan3了，以免规则冲突。

好的，截图里面  二选一的对吧