istoreos/istoreos #1241: 请教关于各场景ipv6写法问题。

bupahaibao, 2024-01-06:
众所周知ipv6前64位运营商分配的，是会不定时变更的，后64位是根据设备mac算出来的，是不会变的。所以我们在某些场景想要设定某个设备的ipv6流量，就要用到后面64位固定的地址。我举个例子，比如A设备ipv6是"2409:8a20:e0c:fee0:xxxx:xxxx:xxxx:xxxx"

如果我想在防火墙中设置wan区域流量可以访问这个A设备的ipv6，那么我只需放行所有端口到这个ipv6地址的流量，但是前64位不断变化怎么办呢？有办法，可以这样写“::xxxx:xxxx:xxxx:xxxx/::ffff:ffff:ffff:ffff”这样就可以表示A设备的ipv6地址，前面无论怎么变都无所谓。

但是在mwan3中又让我迷惑起来了，继续沿用上面的方法不成功了，显示格式不对，请问有没有懂的，知道该怎么正确的书写？
[image]

bcseputetto, 2024-01-07:
可以这么写
::aaaa:bbbb:cccc:eeee/0
[image]

bcseputetto, 2024-01-07:
你测试一下
`::aaaa:bbbb:cccc:dddd/0`
有没有效果，如果没有效果的话可能要向openwrt那边提交issue

bupahaibao, 2024-01-07:
> 你测试一下 `::aaaa:bbbb:cccc:dddd/0` 有没有效果，如果没有效果的话可能要向openwrt那边提交issue
> 
> 我向openwrt luci仓库提交了 [issue](https://github.com/openwrt/luci/issues/6813)

这应该没效果吧。掩码是0，那就是所有地址啊。你想说这是反掩码，但是mwan3写正常地址的时候都是正掩码，没有写反掩码的传统吧。并且也没有说写前半部分IP配正掩码，写后半部分IP配反掩码的传统。我认为mwan3这个工具还是没有充分考虑IPv6的问题，应该在地址格式规范上与防火墙保持一致。

bcseputetto, 2024-01-08:
> 这应该没效果吧。掩码是0，那就是所有地址啊。你想说这是反掩码，但是mwan3写正常地址的时候都是正掩码，没有写反掩码的传统吧。并且也没有说写前半部分IP配正掩码，写后半部分IP配反掩码的传统。我认为mwan3这个工具还是没有充分考虑IPv6的问题，应该在地址格式规范上与防火墙保持一致。
@bupahaibao 请协助测试一下是否mwan3支持跟防火墙一样的写法，而只是luci限制了此格式。
编辑 /etc/config/mwan3
```console
config rule 'default_rule_v6'
        option family 'ipv6' #互联网协议
        option proto 'all' #协议
        option src_ip '::aaaa:bbbb:cccc:dddd/::ffff:ffff:ffff:ffff' #源地址
        option dest_ip '::/0' #目标地址
        option sticky '0' #粘滞模式
        option use_policy 'balanced' #分配的策略
```
编辑完后保存文件。
执行下面的命令重启 mwan3
```console
/etc/init.d/mwan3 restart
```

bupahaibao, 2024-01-08:
> > 这应该没效果吧。掩码是0，那就是所有地址啊。你想说这是反掩码，但是mwan3写正常地址的时候都是正掩码，没有写反掩码的传统吧。并且也没有说写前半部分IP配正掩码，写后半部分IP配反掩码的传统。我认为mwan3这个工具还是没有充分考虑IPv6的问题，应该在地址格式规范上与防火墙保持一致。
> 
> @bupahaibao 请协助测试一下是否mwan3支持跟防火墙一样的写法，而只是luci限制了此格式。 编辑 /etc/config/mwan3
> 
> ```
> config rule 'default_rule_v6'
>         option family 'ipv6' #互联网协议
>         option proto 'all' #协议
>         option src_ip '::aaaa:bbbb:cccc:dddd/::ffff:ffff:ffff:ffff' #源地址
>         option dest_ip '::/0' #目标地址
>         option sticky '0' #粘滞模式
>         option use_policy 'balanced' #分配的策略
> ```
> 
> 编辑完后保存文件。 执行下面的命令重启 mwan3
> 
> ```
> /etc/init.d/mwan3 restart
> ```
> 
> 测试一下这种写法能否让mwan3正常工作。如果能正常工作，则只是luci限制了数据格式。
> 
> ![image](https://private-user-images.githubusercontent.com/29139573/294874580-afff8781-9169-4065-9626-6eb53e665a89.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MjY4OTIsIm5iZiI6MTcwNDcyNjU5MiwicGF0aCI6Ii8yOTEzOTU3My8yOTQ4NzQ1ODAtYWZmZjg3ODEtOTE2OS00MDY1LTk2MjYtNmViNTNlNjY1YTg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDE1MDk1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlNjIzMmYxYWVhZTgxYTQ4Yzg3MTZlYjA5ZmJmNGViODBjNDQ0NWU0OWE2NmM3YWRhMjMwOWZmODJkMTAxZGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7A3RTrr3MhEwTCEPXtuIvxpCl1bg1MvB-btS-DLNNo8) ![image](https://private-user-images.githubusercontent.com/29139573/294874671-6019d5fa-9b32-4350-a091-6e32e3571d7a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MjY4OTIsIm5iZiI6MTcwNDcyNjU5MiwicGF0aCI6Ii8yOTEzOTU3My8yOTQ4NzQ2NzEtNjAxOWQ1ZmEtOWIzMi00MzUwLWEwOTEtNmUzMmUzNTcxZDdhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDE1MDk1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3MzczMDM1NTc4ZWIyNTg2ODc2NjhjMWY3N2UyNTlkOWUwMTEzMGFjN2NjNjdmNDhkMWY4MDA1NGIwNjA4M2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RRPtqsgL30kXomP0y7dSub35CcEXDKqVn2XPgHpJQhw) ![image](https://private-user-images.githubusercontent.com/29139573/294875205-185f4bac-489f-4f28-903c-88503ba8b66f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MjY4OTIsIm5iZiI6MTcwNDcyNjU5MiwicGF0aCI6Ii8yOTEzOTU3My8yOTQ4NzUyMDUtMTg1ZjRiYWMtNDg5Zi00ZjI4LTkwM2MtODg1MDNiYThiNjZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA4VDE1MDk1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlmZjQwZWQ3YmZmZTg3NGNjOGYxYTg5N2VhN2M2MDAyM2UyNjcyOWNiMzRhNjdjYTkwM2EyMGRkNmMwY2FjZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.crCARaRvm-HazaqfdkC5L4whj1puzEMSyY94CYII7iU)

妙啊，我试一下，不过我在出差。得等到周末。你这招釜底抽薪妙啊！直接改配置文件。

bupahaibao, 2024-01-09:
> @bupahaibao 请协助测试一下是否mwan3支持跟防火墙一样的写法，而只是luci限制了此格式。 编辑 /etc/config/mwan3
>
> ```
> config rule 'default_rule_v6'
>         option family 'ipv6' #互联网协议
>         option proto 'all' #协议
>         option src_ip [...]

兄弟，我试了一下，结果是：如果硬是在mwan3配置文件中这样修改，会导致整个mwan3负载均衡配置失效，所有流量都会走跳数低的wan口。我两台网心云，一台走wan电信，一台走wan1移动。按照你的方法尝试以后，都被识别为电信了。

bcseputetto, 2024-01-09:
@bupahaibao 用 tracker 测试IPv6路由 是最直观的办法

bcseputetto, 2024-01-09:
@bupahaibao 用tracert 测试IPv6路由是最直观的测试办法

bupahaibao, 2024-01-09:
> @bupahaibao 用tracert 测试IPv6路由是最直观的测试办法

我要吐血了，后面怎么改mwan3，两个网心云都识别电信。我只能建立两个出口防火墙了，wan和wan2。规则设定dmz_dianxin出wan区域，dmz_yidong出wan2，区域。我就一直觉得mwan3这玩意不靠谱，生不生效全看天意。还是防火墙靠谱。我已经躺平了，总之那种修改mwan3文件的做法也是不生效的。坐等你那个帖子lcui大神给出正解。

jjm2473, 2024-01-12:
openwrt可以配置策略路由，默认网关也可以写到非main路由表，功能可以替代掉mwan3了，可能配置复杂些。

具体的位置：策略路由在 网络-路由，以及 网络-接口 里面各个接口的高级配置“覆盖路由表”相关的选项。

bupahaibao, 2024-01-12:
> openwrt可以配置策略路由，默认网关也可以写到非main路由表，功能可以替代掉mwan3了，可能配置复杂些。
> 
> 具体的位置：策略路由在 网络-路由，以及 网络-接口 里面各个接口的高级配置“覆盖路由表”相关的选项。

捣鼓了。三条线路。我想让前两条各走各的。第三条负载均衡。问题是我用路由策略做了前两条各走各的。Mwan3里面第三条的负载均衡也就失效。不晓得为什么。

jjm2473, 2024-01-13:
既然自己配置策略路由了，就不要再用mwan3，卸载掉

bupahaibao, 2024-01-13:
> 既然自己配置策略路由了，就不要再用mwan3，卸载掉

但是做策略并不能让应该负载均衡的地址网速叠加呀。mwan3可以网速叠加。

jjm2473, 2024-01-13:
mwan3也是利用策略路由的，可能会跟你配置的冲突。你可以用`ip rule`和`ip route`命令检查下路由和策略有没有问题，后台的状态-路由页面也能看到。

bupahaibao, 2024-01-13:
> mwan3也是利用策略路由的，可能会跟你配置的冲突。你可以用`ip rule`和`ip route`命令检查下路由和策略有没有问题，后台的状态-路由页面也能看到。

是这样，mwan3中设置负载均衡是这两个优先级权重都设置成了一样，所以他能做到上行下行带宽的叠加。我能跑到1800M的网速。
但如果卸载了，这个东西纯粹用策略路由来做。本来应该负载均衡的流量，他会根据路由策略选择优先级高的一条走。无法实现负载均衡啊。你把跳数和权重都设置成一样，会引发路由问题。

bupahaibao, 2024-01-13:
> mwan3也是利用策略路由的，可能会跟你配置的冲突。你可以用`ip rule`和`ip route`命令检查下路由和策略有没有问题，后台的状态-路由页面也能看到。

Openwrt的路由也很好看懂。wan和wan2建立链接之后，IPv4路由表会生成表1和表3。IPv6路由表会生成表2和表4。我们设置策略的时候，你想他走wan1，就在IPv4规则里面把选定区域的流量加入表一就行，在IPv6规则里面，把选定区域的流量加入表2。这样就完成了指定不同区域的流量走不同的wan口。而你需要负载均衡的区域，就完全不动，保持默认。不设立任何规则。这样的话他两个口都可以走，但实际情况是流量在选择的时候，它会选择优先级高的一条路，不会做负载均衡。所以想请教一下，你说通过策略路由来实现负载均衡，具体应该怎么操作呢？

jjm2473, 2024-01-15:
不然mwan3怎么实现负载均衡的？看看mwan3生成的规则