istoreos/istoreos #1244: ip command not work

TTcheng, 2024-01-07:
## ip命令无法运行，并报错

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

我从[22.03.5 - 20230728]升级到[22.03.5 - 20231229]发现ip命令无法正常运行

### (2) 路由器型号和固件版本

A：
hardware：天宝AMD 300U
Istoreos：22.03.5 - 20231229

### (3) 详细日志和/或截图

A：errors belows here
Error relocating /sbin/ip: bpf_object__next_map: symbol not found
[image]

jjm2473, 2024-01-08:
你肯定是自己升级了系统软件包，导致不兼容。
istore安装系统便利工具，然后执行其中的“修复系统软件”，等自动重启后应该就好了。

closed 2024-01-08 by jjm2473

TTcheng, 2024-01-13:
我尝试过两次，无法解决我的问题

TTcheng, 2024-01-13:
我单独执行了修复系统软件的脚本，命令提示了修复的包是 ip-full，但是重启之后依然没用

jjm2473, 2024-01-13:
那就恢复出厂设置吧，以后不要再乱升级系统软件包

TTcheng, 2024-01-15:
相关软件包是OpenClash引入的，希望官方早点支持

jjm2473, 2024-01-15:
怎么可能呢，ip-full本来就集成到固件里的，就算OpenClash需要这个包，也不需要安装。只有可能是手动安装的。