istoreos/istoreos #1245: 编译错误

ghmole, 2024-01-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错，好像是那个patch对

### (2) 路由器型号和固件版本

A：HLink-H68K

### (3) 详细日志和/或截图

A：
patching file drivers/net/ethernet/stmicro/stmmac/stmmac_ptp.h
patching file drivers/net/ethernet/stmicro/stmmac/stmmac_selftests.c
patching file include/linux/stmmac.h
Hunk #5 FAILED at 205.
1 out of 5 hunks FAILED -- saving rejects to file include/linux/stmmac.h.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/rockchip/rk35xx/patches-5.10/0019-drivers-update-stmmac.patch!
make[4]: *** [Makefile:33: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-rockchip_rk35xx/linux-5.10.201/.prepared_b4e85a5b280a483109f7751479adec10] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#8.83#7.09#11.79
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:234: world] Error 2

jjm2473, 2024-01-08:
修复了，昨天加了个补丁没rebase。

closed 2024-01-08 by jjm2473

ghmole, 2024-01-09:
Applying /workdir/openwrt/target/linux/generic/backport-5.15/791-v6.2-01-net-phy-Add-driver-for-Motorcomm-yt8521-gigabit-ethernet.patch using plaintext: 
patching file MAINTAINERS
patching file drivers/net/phy/Kconfig
patching file drivers/net/phy/motorcomm.c
Hunk #1 FAILED at 1.
Hunk #2 FAILED at 38.
Hunk #3 FAILED at 111.
Hunk #4 FAILED at 121.
4 out of 4 hunks FAILED -- saving rejects to file drivers/net/phy/motorcomm.c.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/backport-5.15/791-v6.2-01-net-phy-Add-driver-for-Motorcomm-yt8521-gigabit-ethernet.patch!
make[4]: *** [Makefile:29: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.125/.prepared_b89c6d4515652b5d165d5b0946645869] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#7.85#6.51#10.15
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:236: world] Error 2

jjm2473, 2024-01-10:
你用的啥分支啊，内核5.15？你是不是乱合并的