istoreos/istoreos #1247: 编译21.02版本21.02分支的最新提交时，gcc 8.4.0出错

Roxy09099, 2024-01-09:
(1) 具体问题

A：编译21.02版本21.02分支的最新提交时，gcc 8.4.0出错

Log:
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libbacktrace   -o print-tree.o -MT print-tree.o -MMD -MP -MF ./.deps/print-tree.TPo /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-tree.c
/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c: In function 'void print_value(pretty_printer*, const_rtx, int)':
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1658:25: warning: spurious leading punctuation sequence '<' in format [-Wformat-diag]
 1658 |         pp_printf (pp, "<%wx,%wx>",
      |                         ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1658:33: warning: spurious trailing punctuation sequence '>' in format [-Wformat-diag]
 1658 |         pp_printf (pp, "<%wx,%wx>",
      |                                 ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1667:23: warning: quoted '%s' directive in format [-Wformat-diag]
 1667 |       pp_printf (pp, "\"%s\"", XSTR (x, 0));
      |                       ^~~~~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1667:22: note: if using '%qs' is not feasible, use '"%-s"' instead
 1667 |       pp_printf (pp, "\"%s\"", XSTR (x, 0));
      |                      ^~~~~~~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1670:23: warning: grave accent '`' in format [-Wformat-diag]
 1670 |       pp_printf (pp, "`%s'", XSTR (x, 0));
      |                       ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1670:22: note: use the apostrophe directive '%'' instead
 1670 |       pp_printf (pp, "`%s'", XSTR (x, 0));
      |                      ^~~~~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1670:26: warning: unterminated quote character ''' in format [-Wformat-diag]
 1670 |       pp_printf (pp, "`%s'", XSTR (x, 0));
      |                          ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1678:25: warning: spurious trailing punctuation sequence '(' in format [-Wformat-diag]
 1678 |       pp_printf (pp, "%s(", GET_RTX_NAME (GET_CODE (x)));
      |                         ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1678:25: warning: unbalanced punctuation character '(' in format [-Wformat-diag]
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1692:25: warning: spurious leading punctuation sequence ':' in format [-Wformat-diag]
 1692 |         pp_printf (pp, ":%s", GET_MODE_NAME (GET_MODE (x)));
      |                         ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1696:23: warning: spurious leading punctuation sequence '#' in format [-Wformat-diag]
 1696 |       pp_printf (pp, "#");
      |                       ^
/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c: In function 'void print_pattern(pretty_printer*, const_rtx, int)':
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1746:25: warning: spurious trailing space in format [-Wformat-diag]
 1746 |       pp_printf (pp, "%s ", GET_RTX_NAME (GET_CODE (x)));
      |                         ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1800:31: warning: spurious trailing space in format [-Wformat-diag]
 1800 |             pp_printf (pp, "%s      ", save_print_rtx_head);
      |                               ^~~~~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1815:23: warning: unquoted keyword 'asm' in format [-Wformat-diag]
 1815 |       pp_printf (pp, "asm {%s}", XSTR (x, 0));
      |                       ^~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1815:30: warning: spurious trailing punctuation sequence '}' in format [-Wformat-diag]
 1815 |       pp_printf (pp, "asm {%s}", XSTR (x, 0));
      |                              ^
/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c: In function 'void print_insn(pretty_printer*, const rtx_insn*, int)':
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1914:34: warning: unbalanced punctuation character '>' in format [-Wformat-diag]
 1914 |         pp_printf (pp, "debug %s => ", name);
      |                                  ^~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1914:36: warning: spurious trailing space in format [-Wformat-diag]
 1914 |         pp_printf (pp, "debug %s => ", name);
      |                                    ^
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1967:32: warning: unquoted sequence of 2 consecutive punctuation characters '("' in format [-Wformat-diag]
 1967 |               pp_printf (pp, " (\"%s\")", label);
      |                                ^~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1967:37: warning: spurious trailing punctuation sequence '")' in format [-Wformat-diag]
 1967 |               pp_printf (pp, " (\"%s\")", label);
      |                                     ^~~
/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c: In function 'void print_insn_with_notes(pretty_printer*, const rtx_insn*)':
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1999:27: warning: unquoted sequence of 6 consecutive space characters in format [-Wformat-diag]
 1999 |         pp_printf (pp, "%s      %s ", print_rtx_head,
      |                           ^~~~~~
Error: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/print-rtl.c:1999:35: warning: spurious trailing space in format [-Wformat-diag]
 1999 |         pp_printf (pp, "%s      %s ", print_rtx_head,
      |                                   ^
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/host/include  -I/workdir/openwrt/build_dir [... cut ...] l -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/. -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../include -I/workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/gcc-8.4.0/gcc/../libcpp/include -I/workdir/openwrt/staging_dir/host

jjm2473, 2024-01-10:
试试换成 gcc 7.x

Roxy09099, 2024-01-10:
> 试试换成 gcc 7.x

我需要如何修改这个文件？
toolchain/gcc/Config.version
config GCC_VERSION_7
	default y if GCC_USE_VERSION_7
	bool

config GCC_VERSION_9
	default y if GCC_USE_VERSION_9
	bool

config GCC_VERSION_10
	default y if GCC_USE_VERSION_10
	bool

config GCC_VERSION
	string
	default "7.5.0"		if GCC_VERSION_7
	default "9.3.0"		if GCC_VERSION_9
	default "10.2.0"	if GCC_VERSION_10
	default "8.4.0"

config GCC_USE_IREMAP
	bool
	default y if GCC_USE_VERSION_7
	default n

jjm2473, 2024-01-10:
不需要修改文件，`make menuconfig`就可以选择gcc版本

Roxy09099, 2024-01-11:
> 不需要修改文件，`make menuconfig`就可以选择gcc版本

make[3]: Entering directory '/home/longtoan99/opt/istoreos/package/transplant/xtables-addons'
mkdir -p /home/longtoan99/opt/istoreos/dl
SHELL= flock /home/longtoan99/opt/istoreos/tmp/.xtables-addons-2.15.tar.xz.flock -c '  	/home/longtoan99/opt/istoreos/scripts/download.pl "/home/longtoan99/opt/istoreos/dl" "xtables-addons-2.15.tar.xz" "e1e05c1ec5188d7b0252e260df70c562acaff5174110609cc93b9ef931b8d663" "" "https://inai.de/files/xtables-addons/"    '
rm -f /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15/.built
touch /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15/.built_check
make  -C /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15 KCFLAGS="-fmacro-prefix-map=/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/longtoan99/opt/istoreos/staging_dir/host/include -I/home/longtoan99/opt/istoreos/staging_dir/hostpkg/include -I/home/longtoan99/opt/istoreos/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Dec 18 03:31:58 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/longtoan99/opt/istoreos/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/longtoan99/opt/istoreos/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/antfs.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ath10k-ct.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/batman-adv.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/button-hotplug.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/dahdi-linux.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/dmx_usb_module.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/exfat.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gl-mifi-mcu.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mac80211.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/macremapper.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mt76.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mtd-rw.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/nat46.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/netatop.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ntfs3-oot.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/oaf.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/openvswitch.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/r8125.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/r8168.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/rtl8812au-ct.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/rtpengine.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/siit.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/trelay.symvers /home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/usb-serial-xr_usb_serial_common.symvers" KERNELRELEASE=5.4.188 DESTDIR="/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15/ipkg-install" DEPMOD="/bin/true" all
make[4]: Entering directory '/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15'
make  all-recursive
make[5]: Entering directory '/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15'
Making all in extensions
make[6]: Entering directory '/home/longtoan99/opt/istoreos/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/xtables-addons-2.15/extensions'
Xtables-addons 2.15 - Linux 5.4.188
[... cut ...]
    ERROR: package/transplant/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/transplant/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/longtoan99/opt/istoreos'
make[1]: *** [package/Makefile:110: /home/longtoan99/opt/istoreos/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/longtoan99/opt/istoreos'
make: *** [/home/longtoan99/opt/istoreos/include/toplevel.mk:234: world] Error 2
longtoan99@AMD:~/opt/istoreos$

closed 2024-01-12 by jjm2473