istoreos/istoreos #1251: MT7921无线网卡设置问题

2253845067, 2024-01-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：正常设置MT7921无线网卡，安卓和电脑可以正常连接，苹果设备能搜到WiFi，却连不上，在其他固件上用同样的WiFi设置可以正常连接，  istoreos不管是自编译还是官方都没办法正常连接，想知道下苹果设备应该如何设置才可以正常连接上

### (2) 路由器型号和固件版本

A：
设备 H69K

固件版本 istoreos-22.03.5-2023122916-h6xk-squashfs-combined.img.gz

### (3) 详细日志和/或截图

A：
[image]

[istoreos-22.03.5-2023122916-h6xk-squashfs-combined.img.gz](https://fw0.koolcenter.com/iStoreOS/h6xk/istoreos-22.03.5-2023122916-h6xk-squashfs-combined.img.gz)

jjm2473, 2024-01-12:
高级设置试试选择国家为“China”，还有信道选择157试试

jjm2473, 2024-01-12:
你的苹果手机哪一年产的，支持wifi6吗？你先试试把wifi的AX改成AC

2253845067, 2024-01-12:
> 你的苹果手机哪一年产的，支持wifi6吗？你先试试把wifi的AX改成AC

手机是14pm，平板是ipad2021，有时间我试试看ax改ac，现在用的大雕的固件，一样的设置可以正常连接

2253845067, 2024-01-15:
切换到AC模式就可以正常连接了，速度也正常
[image]

closed 2024-01-15 by 2253845067

jjm2473, 2024-01-15:
那就是wifi6的兼容性问题，尤其老设备很容易遇到。