istoreos/istoreos #1253: 提两个问题，其中一个是bug，另外一个建议。

haobinnan, 2024-01-12:
路由器硬件：电犀牛r66s，用的整合后的新固件。iStoreOS 22.03.5 2023122916 / LuCI istoreos-22.03 branch git-23.363.24414-7aedb5e

1、问题：点击 “网络” -> “接口” -> “wan_6的编辑”
界面的上方提示错误
TypeError
Cannot read properties of null (reading '.name')

2、建议：“主页”可以显示“已连接设备”的个数，希望能提供一个列出具体那些设备在线。

haobinnan, 2024-01-12:
[image]

jjm2473, 2024-01-12:
wan_6是pppoe产生的虚拟接口，没法编辑，相关配置都在pppoe里面。

“已连接设备”的列表点一下“设备路由”就能看到，其实就是ARP表的条目数量。

“动态 DNS”没有后台服务，只要你没配置，就不会有任何影响。就算配置了，也只是几个脚本定时更新DNS而已。

“易有云”默认启动主要是为了方便易有云APP能扫描到局域网的这个设备，还有文件管理页面也依赖易有云的服务。

closed 2024-01-12 by jjm2473

haobinnan, 2024-01-12:
@jjm2473 “已连接设备”的列表点一下“设备路由”就能看到，其实就是ARP表的条目数量。 这里显示的数量要比 主界面显示的多。

jjm2473, 2024-01-12:
@haobinnan arp是包含所有接口的，你只看lan口的是不是对得上

haobinnan, 2024-01-12:
对不上发自我的荣耀手机-------- 原始邮件 --------发件人： Liangbin Lian ***@***.***>日期： 2024年1月12日周五 21:26收件人： istoreos/istoreos ***@***.***>抄送： binnan_hao ***@***.***>, Mention ***@***.***>主    题： Re: [istoreos/istoreos] 提两个问题，其中一个是bug，另外一个建议。 (Issue #1253)
@haobinnan arp是包含所有接口的，你只看lan口的是不是对得上

haobinnan, 2024-01-12:
> @haobinnan arp是包含所有接口的，你只看lan口的是不是对得上

对不上 检查了 好几次

jjm2473, 2024-01-13:
你截图看看两边的数据分别是什么

haobinnan, 2024-01-14:
> 你截图看看两边的数据分别是什么

我知道什么原因了，因为我设置了静态arp（mac与ip绑定），所以导致不一致了。不过还是推荐咱们单独弄一个列表出来，还能显示主机名，这样方便查看。

jjm2473, 2024-01-15:
主机名依赖的是DHCP过程中主机提交的名称，ARP又不一定是DHCP分配的主机才有，其实都会对不上。

jjm2473, 2024-01-16:
没啥必要，真有很多客户端，也只能去“设备路由”或者DHCP分配记录看，或者“状态”-“概览”页面也有DHCP分配信息。DHCP记录一般有主机名，只要客户端提交了。

haobinnan, 2024-01-16:
好吧 暂时这样吧 感谢。

fenglinyes, 2024-07-12:
X86 
iStoreOS 22.03.6 2024062810
为什么PPPOE拨号后，没有wan_6接口出现呢。可以确定通过光猫拨号上网有V6地址和公网V4地址。通过软路由拨号就没有V6地址

haobinnan, 2024-07-12:
istoreos商店中记得有一个什么工具可以打开ipv6  你先用这个试试。