istoreos/istoreos #1256: 是否可以在istoreos内加入SFE和BBR两种加速技术

bailuxianfeiying, 2024-01-13:
[image]
对于像H28K这样的arm设备，这两个加速技术应该可以增强网络性能

jjm2473, 2024-01-13:
SFE不就是offload吗，防火墙页面就可以开。BBR在路由器上一般没什么用，反而可能导致莫名其妙的问题，例如samba速度异常

bailuxianfeiying, 2024-01-13:
好像sfe和offload有点区别，sfe可能性能更好？

jjm2473, 2024-01-13:
那你有空可以对比下

VwEl, 2024-02-26:
> 那你有空可以对比下

sfe性能会更高点，所以大伙呼声这么高，就加一个呗

HernandoR, 2024-03-03:
related #352

jjm2473, 2024-03-04:
SFE可能会引入兼容性问题。offload是linux内核和openwrt官方支持的，同样也是软件实现的流卸载。如果sfe性能比offload好，那就提供性能对比数据。或者反馈给openwrt社区，让openwrt支持sfe，既然这东西这么好他们应该不会拒绝。

BBR内核支持倒是简单，之后可以加上，不过不会默认启用。

jjm2473, 2024-03-08:
如果是x86或者rk33xx系列，或者rpi4，在软件包里面搜索 kmod-tcp-bbr 安装就可以使用bbr了（安装后可能要重启下，自动切换到bbr，不需要设置），但是如果遇到samba速度或其他网络速度异常，不要反馈。

jjm2473, 2024-03-08:
h28k就算不开offload都能跑满千兆PPPoE了

closed 2024-04-07 by jjm2473