istoreos/istoreos #1257: [Bug Report]关于x86-64环境下对于22.03.6-2024011211版本的兼容。

OcrSu, 2024-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A： 在G5400 / 华硕H310M DDR4的x86-64 环境下从

22.03.5-2023122916 升级到 22.03.6-2024011211 版本。

会导致系统损坏，无法访问后台，网页OTA升级了半个小时也没有成功。
使用显示器连接主机，使用quickstart 中第一个指令查看网关信息 会进入到一个可以输入文本的状态，但无法退出，也无法重启。

此后我下载了istoreos-22.03.6-2024011211-x86-64-squashfs-combined.img.gz 并刻录到我的U盘上。
发现其功能是不正常的，比如
1. 启动后没有提示按Enter进入控制台
2. 安装完成后无法自动重启
3. 手动重启进入主机的iStoreOS控制台后，使用quickstart 中第一个指令查看网关信息仍然会进入到一个可以输入文本的状态，但无法退出，也无法重启的状态。

最后呢，我下载了22.03.5-2023122916版本，并刻录到我的U盘内。
此时一切正常，可以正常安装自动重启，网卡信息也能读取，总之没有任何Bug。

### (2) 路由器型号和固件版本

A：
istoreos-22.03.6-2024011211-x86-64-squashfs-combined.img.gz

### (3) 详细日志和/或截图

A：退坑了，没留下任何截图，等iStoreOS什么时候修好这个bug我再来吧。数据没了太心痛了。

jjm2473, 2024-01-13:
你是不是把非efi固件刷到efi系统里了？

OcrSu, 2024-01-13:
> 你是不是把非efi固件刷到efi系统里了？

晚点再确认一下，这台机子做主路由几个月了，今天更新才出的问题，我也忘记当时下载的哪个固件了。

jjm2473, 2024-01-13:
OTA升级失败不会破坏系统，下载错误的固件刷上去才会破坏系统。

OcrSu, 2024-01-13:
> OTA升级失败不会破坏系统，下载错误的固件刷上去才会破坏系统。

确实不是对原有的环境进行破坏，我想表述的破坏是：无法访问后台 即使用quickstart 中第一个指令查看网关信息 会进入到一个可以输入文本的状态，但无法退出，也无法重启。

我晚点再重新下回来试试，如果仍然能复现我就在这里多提供一点信息（比如截图等）

jjm2473, 2024-01-13:
quickstart里面选择 `q、 QUIT` 然后回车就退出了

OcrSu, 2024-01-13:
> quickstart里面选择 `q、 QUIT` 然后回车就退出了

他是类似于vim编辑器状态，并没有在quickstart的5个选项交互界面中。

我试过了vim的各种退出方法，都没办法生效。

jjm2473, 2024-01-13:
你后面测试时是不是同时有多个系统存在，例如u盘有istoreos，硬盘里也有，那就不确定会从哪里启动了。不过最重要的是不能efi跟非efi混刷，那肯定会刷坏引导，以后看看能不能加上检测，防止混刷。

OcrSu, 2024-01-13:
> 你后面测试时是不是同时有多个系统存在，例如u盘有istoreos，硬盘里也有，那就不确定会从哪里启动了。不过最重要的是不能efi跟非efi混刷，那肯定会刷坏引导，以后看看能不能加上检测，防止混刷。

还没测试呢，刚吃饱饭在刻录EFI的U盘镜像。我这台机子只有 240G的istoreos的系统盘和 2tb的数据盘。等会测试好了反馈给你

jjm2473, 2024-01-13:
如果出现莫名其妙的界面，例如你说的类型vim的界面，拍照看看是啥

OcrSu, 2024-01-13:
> 如果出现莫名其妙的界面，例如你说的类型vim的界面，拍照看看是啥

我测试好了，跟efi没关系，方便加一下qq/vx吗。我录制了个视频

OcrSu, 2024-01-13:
> 如果出现莫名其妙的界面，例如你说的类型vim的界面，拍照看看是啥

U盘和 install在硬盘后 都是这样的情况。EFI和非EFI 固件都试过了。

[image]
[image]

jjm2473, 2024-01-13:
看起来是程序卡住了。你的网卡知道是什么型号的吗？

jjm2473, 2024-01-13:
重启以后，试试执行`ip link`看会不会卡住

OcrSu, 2024-01-13:
> 看起来是程序卡住了。你的网卡知道是什么型号的吗？

主板板载网卡是：
Realtek® RTL8111H, 1 x 千兆网卡

PCIE网卡为 EB-LINK的 RTL8125BG 单电口。

试过了，ip link也会卡住，跟图中情况一样。

OcrSu, 2024-01-13:
> > 看起来是程序卡住了。你的网卡知道是什么型号的吗？
> 
> 主板板载网卡是： Realtek® RTL8111H, 1 x 千兆网卡
> 
> PCIE网卡为 EB-LINK的 RTL8125BG 单电口。
> 
> （下图有相关信息）
> 
> 试过了，ip link也会卡住，跟图中情况基本相同，但ip link后不能用 print功能按键 实现脱离这个状态，而上文图中的可以。
> 
> ![image](https://private-user-images.githubusercontent.com/75876201/296449010-b2d73e0a-dd49-4ebd-bec5-6fe91a6b935d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUxNDY3MzMsIm5iZiI6MTcwNTE0NjQzMywicGF0aCI6Ii83NTg3NjIwMS8yOTY0NDkwMTAtYjJkNzNlMGEtZGQ0OS00ZWJkLWJlYzUtNmZlOTFhNmI5MzVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEzVDExNDcxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNhMWVjZTJiZGU1MjJmZTUzZDI2NjFiMDdlNWRiN2Q2NDA2ZjEwYmQ3NDRmOTA2NzdkY2RhNzVmYjcyYTEzNzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O7RmMhpEQ8p6fTbza7bX96rjhLrk475yzTtxfIprG4g)

我这图中怎么显示 8221B，难道是驱动识别错误？？

jjm2473, 2024-01-13:
那应该是内核驱动出了问题，导致所有读取网络信息的进程都会卡住。你试试先关机拔掉PCIE网卡，开机看看是不是正常了

OcrSu, 2024-01-13:
> 那应该是内核驱动出了问题，导致所有读取网络信息的进程都会卡住。你试试先关机拔掉PCIE网卡，开机看看是不是正常了

太对了哥，完全没有bug了，看来就是驱动上出问题了。

不过之前版本都是没有问题的，希望修复一下！

jjm2473, 2024-01-13:
我看了下，如果是8125网卡，应该会先匹配r8125驱动才对，8111H则应该会被r8168驱动匹配。r8169应该其他两个不匹配才会用上。

jjm2473, 2024-01-13:
你看看pcie网卡重新插上还有没有问题，我怀疑信号问题导致id读错了，所以驱动也加载错误。

jjm2473, 2024-01-13:
你还可以试试卸载掉r8169驱动，然后关机插上网卡，这时就只剩下r8125驱动能匹配了。

OcrSu, 2024-01-13:
s

> 你看看pcie网卡重新插上还有没有问题，我怀疑信号问题导致id读错了，所以驱动也加载错误。

插上就出事了，就变回之前的模样。仍然用的是r8169驱动。

我先去研究下怎么卸载驱动

jjm2473, 2024-01-13:
`opkg remove kmod-r8169`

OcrSu, 2024-01-13:
> `opkg remove kmod-r8169`

驱动卸载后能正常启动了，但wan口 lan口地址都没了

[image]

jjm2473, 2024-01-13:
是不是没启动完啊

jjm2473, 2024-01-13:
我查了下资料，识别成 RTL8226B_RTL8221B 是正常的，因为这是指 PHY 芯片，而不是整个 RTL8125 ，"RTL8125 is not a single PHY solution, it integrates PHY/MAC/PCIE bus controller and embedded memory."

 https://lore.kernel.org/all/1601448538-18004-1-git-send-email-willy.liu@realtek.com/T/

OcrSu, 2024-01-13:
> 是不是没启动完啊
[image]
[image]

感觉不是没启动完，因为之前一直都是这个速度。

jjm2473, 2024-01-13:
嗯，显示 “iStoreOS is ready!“ 就已经启动完了

jjm2473, 2024-01-13:
现在你的机器一个网卡都没识别到。你插上网卡了吗？执行下 `lsmod | grep r81` 看看

jjm2473, 2024-01-13:
quickstart显示的只是默认值而已，`ip link` 没有eth开头的网卡。

jjm2473, 2024-01-13:
还有执行`lspci -nnk | grep -A3 -B3 Network`看看有哪些PCI设备

OcrSu, 2024-01-13:
[image]

另一个指令的反馈稍等下

OcrSu, 2024-01-13:
[image]

OcrSu, 2024-01-13:
> 现在你的机器一个网卡都没识别到。你插上网卡了吗？执行下 `lsmod | grep r81` 看看

插上了的

主板自带的网卡和 PCIEx4 插槽上的 8125BG网卡

jjm2473, 2024-01-13:
奇怪，为什么你的r8125和r8168驱动没了

jjm2473, 2024-01-13:
哦，我想起来了，这两个驱动是有开关的

jjm2473, 2024-01-13:
执行这几条命令启用驱动：
```
uci add_list kmods.kmods.enable=r8125
uci add_list kmods.kmods.enable=r8168
uci commit kmods
reboot
```

重启后应该就正常了

OcrSu, 2024-01-13:
> 执行这几条命令启用驱动：
> 
> ```
> uci add_list kmods.kmods.enable=r8125
> uci add_list kmods.kmods.enable=r8168
> uci commit kmods
> reboot
> ```
> 
> 重启后应该就正常了

太对了哥，一切正常了。

期待后续固件更新修复这个bug，不然我每次更新都要重新执行一次这些指令，麻烦大佬了。

jjm2473, 2024-01-13:
@OcrSu 不用每次都执行的，只要你保留配置文件升级，这些设置都还有效。

不过，r8169内核自带的驱动呢，却有问题，感觉是你的电脑的特例。r8125和r8168是第三方驱动。

OcrSu, 2024-01-13:
> @OcrSu 不用每次都执行的，只要你保留配置文件升级，这些设置都还有效。
> 
> 不过，r8169内核自带的驱动呢，却有问题，感觉是你的电脑的特例。r8125和r8168反而是第三方驱动。

谢谢解答，目前确实没看到有其他人反馈这个问题，或许只是个例，不过旧版本固件是没问题的，既然问题没有彻底解决，那我改一下issue标题吧，这样万一有其他人跟我一样。

zhuozi1212, 2024-01-13:
我的istoreos也是出现这个问题
具体网卡是rtl8125bg 4口版本
通过ota升级，并且发现只要输入对网卡有相关的命令均会卡死
ifconfig | ip addr 之类的命令后，命令终端就会无限卡死（指敲回车会有下一跳但是并不会出任何内容）
[image]

jjm2473, 2024-01-13:
@zhuozi1212 你还能登录后台吗？如果可以，在后台页面的系统-杂项设置-驱动里面勾选r8125，重启下就好了。

如果不能登录后台，那么执行这几条命令启用驱动：
```
uci add_list kmods.kmods.enable=r8125
uci commit kmods
reboot
```
重启后应该就正常了。

zhuozi1212, 2024-01-13:
> @zhuozi1212 你还能登录后台吗？如果可以，在后台页面的系统-杂项设置-驱动里面勾选r8125，重启下就好了。
> 
> 如果不能登录后台，那么执行这几条命令启用驱动：
> 
> ```
> uci add_list kmods.kmods.enable=r8125
> uci commit kmods
> reboot
> ```
> 
> 重启后应该就正常了。
敲完上面的命令就可以
但是reboot后没有效果，强制断电重启就正常了

jjm2473, 2024-01-15:
已确认是 linux 内核 5.10.201 的bug，上游已修复 https://github.com/torvalds/linux/commit/59d395ed606d8df14615712b0cdcdadb2d962175

我也测试了此补丁可以修复死锁问题。

Youwenqwq, 2024-01-15:
> @zhuozi1212 你还能登录后台吗？如果可以，在后台页面的系统-杂项设置-驱动里面勾选r8125，重启下就好了。
> 
> 如果不能登录后台，那么执行这几条命令启用驱动：
> 
> ```
> uci add_list kmods.kmods.enable=r8125
> uci commit kmods
> reboot
> ```
> 
> 重启后应该就正常了。

同为8125网卡，我是PVE PCI直通过去的，也遇到了这个问题，使用该指令启动驱动后无法直接reboot，需要强制重启，重启后恢复正常。

closed 2024-01-19 by jjm2473