istoreos/istoreos #1258: 如何在istoreos内安装Windows

zhuozi1212, 2024-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有方法能在istoreos里面安装Windows吗，或者给个思路

### (2) 路由器型号和固件版本

A：x86-64版本istoreos

jjm2473, 2024-01-15:
istore里面有个虚拟机插件（搜索“webvirtcloud”），用这个就能安装windows

zhuozi1212, 2024-01-15:
> istore里面有个虚拟机插件（搜索“webvirtcloud”），用这个就能安装windows

但是我上传正常的Windows iso会显示文件太大，请问一下是装哪个版本不会这样

LingzhiGod, 2024-01-16:
装个精简版看看

ks-bao, 2024-01-16:
是上传失败，说文件太大，还是放到虚拟机里面文件太大？我自己已经把 Windows 原生官网下载的 4.7G 左右的 iso 都可以的。
你要把 iso 挂载成 CDROM，然后启动虚拟机，从 CDROM 去引导系统就行了。

zhuozi1212, 2024-01-16:
> 是上传失败，说文件太大，还是放到虚拟机里面文件太大？我自己已经把 Windows 原生官网下载的 4.7G 左右的 iso 都可以的。 你要把 iso 挂载成 CDROM，然后启动虚拟机，从 CDROM 去引导系统就行了。
我新创建一个iso存储池在/tmp/1里面但是无论怎么创建都显示不存在 
是不是我操作有问题
[image]
[image]

jjm2473, 2024-01-16:
你是没硬盘吗？还要放在/tmp？

zhuozi1212, 2024-01-16:
> 你是没硬盘吗？还要放在/tmp？

有，但是iso通常一次性就完了，而且我ram有64GiB

zhuozi1212, 2024-01-16:
ram有64GB /tmp/1竟然都说太大 不太明白实际路径在哪
[image]

zhuozi1212, 2024-01-16:
closed吧
玩明白了

closed 2024-01-17 by jjm2473