istoreos/istoreos #1263: J4125 i226 刷这个系统 做主路由，电信千兆，老是隔几天就断网，重启后就恢复

nxydyxsj123456, 2024-01-15:
如题，电信强制断线后 这个系统不会自动重拨么

CHNTDCS, 2024-01-16:
同硬件 同网络 我这啥事也没有。

jjm2473, 2024-02-18:
pppoe拨号的高级设置里面，“LCP 响应故障阈值”设置成1以上的数字，“LCP 响应间隔”设置5试试

138161, 2024-02-26:
先用电脑接，看有几个网关，我一个光猫出来两个网关
在路由器上ip -6 route看一下自己的默认网关。如果获得的是

default from 2402:f000:x:xxxx::/64 via fe80::xxxx:xxxx:xxxx:xxxx dev eth0  proto static  metric 512

这样坑爹的网关，在转发NAT包的时候会有问题，需要把去掉from 2402:f000:x:xxxx::/64这一部分的以后的默认路由添加到路由表中。可以新建一个/etc/hotplug.d/iface/99-ipv6，它的内容是
有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
iface=wan6
[ -z "$iface" -o "$INTERFACE" = "$iface" ] || exit 0
ip -6 route add `ip -6 route show default|sed -e 's/from [^ ]* //'`
logger -t IPv6 "Add IPv6 default route."

这里iface是LuCI里面Network->Interfaces里面看到的名字，一般叫wan6。这个脚本的意思是在wan6起来以后读取默认网关，把带from的内容去掉，再加到系统路由表里。记得
chmod +x /etc/hotplug.d/iface/99-ipv6

有效
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
hwaddr=`ifconfig |ip -6 route show default | head -n1 |awk '{print $5}'`
route -A inet6 add default gw $hwaddr dev eth0

aoinokodou, 2024-09-24:
> pppoe拨号的高级设置里面，“LCP 响应故障阈值”设置成1以上的数字，“LCP 响应间隔”设置5试试

我试一下，我也遇到这个问题了，看wan接口界面有ip地址，但是无法联网，重启接口后恢复，同时多出一个虚拟动态接口