istoreos/istoreos #1268: 无法进入系统>启动项，界面显示 Error XHR request timed out

bizshow, 2024-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刚开机时，动作快的话，可以进入系统>启动项，两个tab的内容都能看到。但是启动过一会（大概一两分钟）后，就无法进入了，界面显示 “Error XHR request timed out”，并且之后也一直无法进入。

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R2S
iStoreOS 22.03.6 2024011211

### (3) 详细日志和/或截图

A：[image]
如果需要系统日志的话，稍后再附上

jjm2473, 2024-01-16:
看看浏览器请求，是不是很长时间都没加载完数据

bizshow, 2024-01-16:
[image]

[image]

确实，问题出在如上图的请求中。

bizshow, 2024-01-16:
我自己通过构造同样的post请求以后，在执行到30s左右会返回这个响应：
`[{"jsonrpc":"2.0","id":2,"result":[0,{"data":"# Put your custom commands here that should be executed once\n# the system init finished. By default this file does nothing.\n\nln -s /proc/self/fd /dev/fd\n/etc/init.d/uhttpd start\nnginx\n\nexit 0\n"}]}`

jjm2473, 2024-01-16:
估计是某个插件的init脚本有bug，导致一调用就卡死。在浏览器加载的过程中，终端执行`ps ww|grep init.d`，把结果发出来看看。

huanhxkx, 2024-01-18:
我的x86小主机，也是开机就会卡死，后面刷了老版本就好了，现在北京敢更新，已更新就会无法进入系统，无论是ｕ盘启动还是刷入到硬盘里，都是一样

jjm2473, 2024-01-18:
@huanhxkx 你的是RTL8125网卡？不要使用20240112版本就行了

bizshow, 2024-01-18:
> 估计是某个插件的init脚本有bug，导致一调用就卡死。在浏览器加载的过程中，终端执行`ps ww|grep init.d`，把结果发出来看看。

[image]
附上进入时，终端里的输出

bizshow, 2024-01-18:
[image]
问题可能出在 {dockerd} /bin/sh /etc/rc.common /etc/init.d/dockerd enabled 上
多次进入时，会有多个进程在跑

jjm2473, 2024-01-18:
把你的 `/etc/init.d/dockerd` 提取出来看看，还有卡住的时候执行`ps ww|grep ubus`看看

bizshow, 2024-01-26:
> 把你的 `/etc/init.d/dockerd` 提取出来看看，还有卡住的时候执行`ps ww|grep ubus; ubus call service list '{"name":"dockerd"}'`看看

`
root@iStoreOS:~#
root@iStoreOS:~# ps ww|grep ubus; ubus call service list '{"name":"dockerd"}'
  516 ubus      1264 S    /sbin/ubusd
 3595 root      2252 S    /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
 8403 root      5380 S    /usr/sbin/uhttpd -f -h /www -r iStoreOS -x /cgi-bin -u /ubus -t 360 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80 -C /etc/uhttpd.crt -K /etc/uhttpd.key -s 0.0.0.0:443 -s [::]:443
11110 root      1252 R    grep ubus
16203 root      2576 S    {dnsmasq} /sbin/ujail -t 5 -n dnsmasq -u -l -r /bin/ubus -r /etc/TZ -r /etc/dnsmasq.conf -r /etc/ethers -r /etc/group -r /etc/hosts -r /etc/passwd -w /tmp/dhcp.leases -r /tmp/dnsmasq.d -r /tmp/hosts -r /tmp/resolv.conf.d -r /usr/bin/jshn -r /usr/lib/dnsmasq/dhcp-script.sh -r /usr/share/dnsmasq/dhcpbogushostname.conf -r /usr/share/dnsmasq/rfc6761.conf -r /usr/share/dnsmasq/trust-anchors.conf -r /usr/share/libubox/jshn.sh -r /var/etc/dnsmasq.conf.cfg01411c -w /var/run/dnsmasq/ -- /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
18197 root      2576 S    {ntpd} /sbin/ujail -t 5 -n ntpd -U ntp -G ntp -C /etc/capabilities/ntpd.json -c -u -r /bin/ubus -r /usr/bin/env -r /usr/bin/jshn -r /usr/sbin/ntpd-hotplug -r /usr/share/libubox/jshn.sh -- /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.cn.pool.ntp.org -p 1.cn.pool.ntp.org -p 2.cn.pool.ntp.org -p 3.cn.pool.ntp.org
{
	"dockerd": {
		"instances": {
			"instance1": {
				"running": true,
				"pid": 22628,
				"command": [
					"/usr/bin/dockerd",
					"--config-file=/tmp/dockerd/daemon.json"
				],
				"term_timeout": 5,
				"limits": {
					"nofile": "1048576 1048576"
				}
			}
		}
	}
}
root@iStoreOS:~#
`
卡住的时候，执行命令显示上面的内容

bizshow, 2024-01-26:
[dockerd.zip](https://github.com/istoreos/istoreos/files/14061184/dockerd.zip)
上面是 /etc/init.d/dockerd 文件

closed 2024-01-26 by bizshow

jjm2473, 2024-02-01:
你自己执行`time /etc/init.d/dockerd enabled`会卡住吗？还有执行`time ubus call service list '{"name":"dockerd"}'`看看命令执行了多长时间

star8020, 2024-02-21:
同样的问题，硬盘休眠也进不去。

jjm2473, 2024-02-22:
@star8020 你的估计是因为硬盘坏了

Somnr123, 2024-05-13:
> 看看浏览器请求，是不是很长时间都没加载完数据

同样的问题，电脑出现XHR request  
但是同网络下手机访问却没有问题[image]

jjm2473, 2024-05-13:
@Somnr123  可能电脑翻墙了？换个浏览器，或者浏览器清理缓存试试？