istoreos/istoreos #1272: iStoreOS的路由器管理界面accessibility相关问题反馈

yunshenwuji, 2024-01-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue
x
## 2. 详细叙述

### (1) 具体问题

A：
将路由器手动配置为旁路由模式的页面，提供 DHCPv4 服务、自动获取 IPV6、开启 NAT这三个复选框不是标准的CheckBox，导致我的屏幕阅读器没法读出复选框的类型和选中状态，希望用标准的复选框来实现

### (2) 路由器型号和固件版本

A：
虚拟机，iStoreOS 22.03.6 2024011211

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-01-17:
这种复选框不是安卓IOS常用的吗？为啥会不支持

yunshenwuji, 2024-01-17:
我对前端了解实在有限，不过可以看下网口配置这个页面，这里的复选框是正常的
[image]

jjm2473, 2024-01-18:
@yunshenwuji 麻烦测试下这个页面的checkbox能不能识别到 https://www.w3schools.com/howto/tryit.asp?filename=tryhow_css_custom_checkbox

yunshenwuji, 2024-01-18:
> @yunshenwuji 麻烦测试下这个页面的checkbox能不能识别到 https://www.w3schools.com/howto/tryit.asp?filename=tryhow_css_custom_checkbox

刚刚点进去测了下，链接里的这一组复选框可以正常聚焦和朗读

yunshenwuji, 2024-01-18:
感谢您关注无障碍需求，我会向其他视力障碍朋友们推荐iStoreOS的，软路由可真是太好玩了

jjm2473, 2024-01-18:
感谢支持！

jjm2473, 2024-01-19:
本周的固件更新将修复

closed 2024-01-19 by jjm2473