istoreos/istoreos #1275: r8169网卡用最新22.03.6，局域网设备跑外网跑不满，换回22.03.5正常

coolzhi, 2024-01-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
1000M带宽
内网测速（局域网传输用那个HOMEBOX插件测试）没问题，
外网测速 用路由器自身跑speedtest可以跑满
局域网设备 跑外网只能跑到500M左右

换回22.3.5最后一个版本 速度正常
A：

### (2) 路由器型号和固件版本

A：CPU N3700/网卡8169吧还是啥 不会看

### (3) 详细日志和/或截图
下面是网卡信息
A：root@ZHIHOME:~# dmesg | grep -i eth
[    6.398720] igb: Intel(R) Gigabit Ethernet Network Driver
[   14.756165] l2tp_eth: L2TP ethernet pseudowire support (L2TPv3)
[   14.772410] i40e: Intel(R) Ethernet Connection XL710 Network Driver
[   14.776129] Intel(R) 2.5G Ethernet Linux Driver
[   14.805043] r8169 0000:01:00.0 eth0: RTL8168evl/8111evl, 00:e0:4c:69:08:41, XID 2c9, IRQ 121
[   14.805063] r8169 0000:01:00.0 eth0: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   14.832776] r8169 0000:02:00.0 eth1: RTL8168evl/8111evl, 00:e0:4c:69:08:42, XID 2c9, IRQ 122
[   14.832796] r8169 0000:02:00.0 eth1: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   14.859044] r8169 0000:03:00.0 eth2: RTL8168evl/8111evl, 00:e0:4c:69:08:43, XID 2c9, IRQ 123
[   14.859060] r8169 0000:03:00.0 eth2: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   14.886335] r8169 0000:04:00.0 eth3: RTL8168evl/8111evl, 00:e0:4c:69:08:44, XID 2c9, IRQ 124
[   14.886354] r8169 0000:04:00.0 eth3: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   15.043626] 8139too: 8139too Fast Ethernet driver 0.9.28
[   15.761301] iavf: Intel(R) Ethernet Adaptive Virtual Function Network Driver
[   15.765516] usbcore: registered new interface driver ipheth
[   15.865165] usbcore: registered new interface driver kaweth
[   16.230044] pegasus: v0.9.3 (2013/04/25), Pegasus/Pegasus II USB Ethernet driver
[   16.660734] usbcore: registered new interface driver cdc_ether
[   16.736135] usbcore: registered new interface driver MOSCHIP usb-ethernet driver
[   23.261692] RTL8211E Gigabit Ethernet r8169-0-200:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-0-200:00, irq=IGNORE)
[   23.624346] r8169 0000:02:00.0 eth1: No native access to PCI extended config space, falling back to CSI
[   23.626679] r8169 0000:02:00.0 eth1: Link is Down
[   23.626722] br-lan: port 1(eth1) entered blocking state
[   23.626760] br-lan: port 1(eth1) entered disabled state
[   23.627132] device eth1 entered promiscuous mode
[   23.630209] br-lan: port 1(eth1) entered blocking state
[   23.630245] br-lan: port 1(eth1) entered forwarding state
[   23.638668] RTL8211E Gigabit Ethernet r8169-0-300:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-0-300:00, irq=IGNORE)
[   23.957032] r8169 0000:03:00.0 eth2: Link is Down
[   23.957150] br-lan: port 2(eth2) entered blocking state
[   23.957171] br-lan: port 2(eth2) entered disabled state
[   23.957527] device eth2 entered promiscuous mode
[   23.957733] br-lan: port 2(eth2) entered blocking state
[   23.957763] br-lan: port 2(eth2) entered forwarding state
[   23.963509] RTL8211E Gigabit Ethernet r8169-0-400:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-0-400:00, irq=IGNORE)
[   24.296112] br-lan: port 1(eth1) entered disabled state
[   24.297824] br-lan: port 2(eth2) entered disabled state
[   24.298441] r8169 0000:04:00.0 eth3: Link is Down
[   24.298953] br-lan: port 3(eth3) entered blocking state
[   24.298990] br-lan: port 3(eth3) entered disabled state
[   24.299241] device eth3 entered promiscuous mode
[   24.311317] RTL8211E Gigabit Ethernet r8169-0-100:00: attached PHY driver [RTL8211E Gigabit Ethernet] (mii_bus:phy_addr=r8169-0-100:00, irq=IGNORE)
[   24.669235] r8169 0000:01:00.0 eth0: Link is Down
[   25.875571] r8169 0000:02:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   25.875627] br-lan: port 1(eth1) entered blocking state
[   25.875643] br-lan: port 1(eth1) entered forwarding state
[   26.657805] r8169 0000:04:00.0 eth3: Link is Up - 1Gbps/Full - flow control rx/tx
[   26.657847] br-lan: port 3(eth3) entered blocking state
[   26.657854] br-lan: port 3(eth3) entered forwarding state
[   27.005215] r8169 0000:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control off
[   27.005259] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  264.605473] r8169 0000:03:00.0 eth2: Link is Up - 1Gbps/Full - flow control rx/tx
[  264.605542] br-lan: port 2(eth2) entered blocking state
[  264.605564] br-lan: port 2(eth2) entered forwarding state

coolzhi, 2024-01-19:
iStoreOS 22.03.5 2023121510这个版本还是会间歇性的跑不满

之前的版本没问题的

jjm2473, 2024-01-19:
在系统-杂项设置的驱动选项卡，启用r8168驱动试试，不管哪个固件版本都可以

coolzhi, 2024-01-20:
> 在系统-杂项设置的驱动选项卡，启用r8168驱动试试，不管哪个固件版本都可以

似乎没有作用，然后刷了昨天最新的版本还是不行，而且速度更差了

jjm2473, 2024-01-22:
@coolzhi 勾选以后要重启

closed 2024-02-23 by jjm2473