istoreos/istoreos #1276: 插件ua2f依赖libnetfilter-quene，libmnl 导致系统内核panic崩溃重启

Mcxunyun666, 2024-01-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在安装插件[UA2F](https://github.com/Zxilly/UA2F)，当安装到依赖libnetfilter-quene，libmnl 时会导致系统内核panic崩溃重启。

### (2) 路由器型号和固件版本

A：

x86_64，22.03.4，22.03.5
树莓派4b（1GB内存版本），22.03.4，22.03.5，22.03.6

### (3) 详细日志和/或截图

A：暂时没有捕获到

jjm2473, 2024-01-18:
不支持NFTables，跟iptables互相冲突，但是为了兼容性，我们只能使用iptables

Mcxunyun666, 2024-01-19:
> 不支持NFTables，跟iptables互相冲突，但是为了兼容性，我们只能使用iptables

也就是在后续版本可能也不打算支持nftables吗？

jjm2473, 2024-01-19:
这倒不是，要看现实情况，如果大部分插件都兼容NFTables了，到时候升级就没问题。如果你想测试NFTables的话，可以回滚掉这个commit https://github.com/istoreos/istoreos/commit/ec8367044caa19441721e90de02f41df88231e33 ，然后编译。

jjm2473, 2024-01-19:
ua2f之前没出过iptables版本吗？或许你可以问下作者

Mcxunyun666, 2024-01-19:
> ua2f之前没出过iptables版本吗？或许你可以问下作者

感谢，我先去看看

Mcxunyun666, 2024-01-21:
感觉有可能是libnetfilter-quene安装配置时拉了nft依赖然后强制安装导致的panic。回头我测试下看看，这个issue先开着留个路吧。感谢回答

_最初由 Mcxunyun666 在 https://github.com/Zxilly/UA2F/issues/115#issuecomment-1902465952 发布_

jjm2473, 2024-01-22:
你试试这个固件（x86_64_efi） https://we.tl/t-C3tnf9avZA

这个固件安装ua2f没问题，不过iptables跟nftables共用可能还有其他问题就自己摸索吧

Mcxunyun666, 2024-01-22:
> 你试试这个固件（x86_64_efi） https://we.tl/t-C3tnf9avZA
> 
> 这个固件安装ua2f没问题，不过iptables跟nftables共用可能还有其他问题就自己摸索吧

感谢，貌似可以使用了，我观察几天看看

jjm2473, 2024-02-18:
新固件已经修复崩溃问题。

closed 2024-02-18 by jjm2473