istoreos/istoreos #1277: 关于cpu占用率过高的问题。

haobinnan, 2024-01-18:
电犀牛测试视频他2000m宽带跑下载cpu占用率才 50% https://www.bilibili.com/video/BV1U24y1X7Go/
而我是1000m用咱们的固件 占用率却很高 如下图。
[image]

设备型号
FastRhino R66s
固件版本
iStoreOS 22.03.6 2024011211

haobinnan, 2024-01-18:
另外一个视频：https://www.bilibili.com/video/BV1iG4y1H7Hk/?spm_id_from=333.999.0.0

haobinnan, 2024-01-18:
这里面 提到了 “数据包负载均衡”  https://www.bilibili.com/video/BV1Kt4y1j7a1/?spm_id_from=333.999.0.0  咱们固件能否开发这个功能呢 ？

jjm2473, 2024-01-18:
“数据包负载均衡”其实就是“数据包引导”，后台的“网络”-“接口”-“全局网络选项”里面就有。

haobinnan, 2024-01-18:
> “数据包负载均衡”其实就是“数据包引导”，后台的“网络”-“接口”-“全局网络选项”里面就有，注意勾选以后重启才生效。

请问 为什么咱们不默认打开呢？有什么弊端吗？

haobinnan, 2024-01-18:
需要等没人用网的时候测试 ，我问lean 他给电犀牛编译的固件 默认开了这个选项  。说开启后单线程会有影响。

haobinnan, 2024-01-18:
> 又不是所有场景打开都是有好处的，你自己试试pppoe，内网传输，内外网转发，如果性能都更好，那就可以默认打开。你自己测试了吗？性能有提升？

然后我又问他 关于cpu占用率高的问题 他的回复是 咱们istoreos 优化问题。

haobinnan, 2024-01-19:
刚刚测试下 对“数据包引导” 选项的改变 对 cpu占用率没有影响，我这两天尽快测试下 lean与istoreos的cpu占用率情况。

coolzhi, 2024-01-19:
最近也发现 也不知道从哪个版本开始，局域网的设备跑外面跑不满。之前肯定是没问题的

haobinnan, 2024-01-19:
> 最近也发现 也不知道从哪个版本开始，局域网的设备跑外面跑不满。之前肯定是没问题的

你什么路由器

haobinnan, 2024-01-19:
@jjm2473 我测试完毕了lean的固件性能还不如咱们istoreos好呢，我测试的是下载，他的固件没有咱们固件表现的好，但是他的固件 cpu占用率明显比咱们低。

haobinnan, 2024-01-20:
> @jjm2473 我测试完毕了lean的固件性能还不如咱们istoreos好呢，我测试的是下载，他的固件没有咱们固件表现的好，但是他的固件 cpu占用率明显比咱们低。

之前说的有点不清晰，我说的lean的下载速度没有咱们固件速度快 咱们基本每次测试都能到 1200  他的 1000出头的时候少。 也不排除他的cpu占用低和这个有关系，他的基本50%出头的cpu占用率。 咱们的 就要 80%+。

jjm2473, 2024-01-22:
如果速度不一样，确实可能cpu占用率也不一样，速度不一样的具体原因就复杂了。

haobinnan, 2024-01-22:
> 如果速度不一样，确实可能cpu占用率也不一样，速度不一样的具体原因就复杂了。

是的就不知道 2000m的带宽能否胜任了。

jjm2473, 2024-01-22:
之前测试过自己搭建的pppoe只能跑到1.3G

haobinnan, 2024-01-22:
> 之前测试过自己搭建的pppoe只能跑到1.3G

看电犀牛的视频 2000m 他们也能跑上去 但是用的lean的固件

jjm2473, 2024-01-22:
所以说不知道是不是光猫不一样啊，也许人家的光猫支持巨型帧呢。不同的带宽要求，使用的技术也不一样，现代的网卡能跑到千兆都是有很多加速技术的，例如TSO，GSO等等，如果没有这些加速，跑千兆都要占用很多CPU。

haobinnan, 2024-01-22:
> 所以说不知道是不是光猫不一样啊，也许人家的光猫支持巨型帧呢。不同的带宽要求，使用的技术也不一样，现代的网卡能跑到千兆都是有很多加速技术的，例如TSO，GSO等等，如果没有这些加速，跑千兆都要占用很多CPU。

好的 那暂时先这样吧 非常感谢。

closed 2024-01-22 by haobinnan

haobinnan, 2024-02-27:
istoreos-22.03.6-2024022310-r6xs-squashfs-combined.img.gz
次版本cpu占用率已经不高了，istoreos更新了网卡驱动。

haobinnan, 2024-02-27:
[image]