istoreos/istoreos #1279: 关于核显调用的问题

szkeum, 2024-01-19:
各位大哥好，我是esxi 8.0装的istoreos x86 64，固件版本iStoreOS 22.03.6 2024011211内核版本5.10.201。已经在系统-杂项设置—驱动里启动了gpu，重启后并没有看到ls /dev/dri里有设备。我尝试在虚拟机选项里面添加了hypervisor.cpuid.v0=FALSE，和svga.present=FALSE都不行，请问是我操作哪里有问题吗？谢谢！

szkeum, 2024-01-19:
补充一下，我的两台主机核显分别是i7 7500U的UHD620和i7 1265U的Intel Iris Xe Graphics 96，均不行。

jjm2473, 2024-01-19:
不要在驱动里启用gpu，现在是自动判断的。关掉GPU驱动，然后重启下。重启以后istoreos执行`lspci -nnk ; dmesg | grep i915` 命令，把结果贴出来看看。

szkeum, 2024-01-19:
root@iStoreOS:~# lspci -nnk ; dmesg | grep i915
00:00.0 Host bridge [0600]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX Host bridge [8086:7190] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: agpgart-intel
lspci: Unable to load libkmod resources: error -12
00:01.0 PCI bridge [0604]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX AGP bridge [8086:7191] (rev 01)
00:07.0 ISA bridge [0601]: Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.1 IDE interface [0101]: Intel Corporation 82371AB/EB/MB PIIX4 IDE [8086:7111] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: ata_piix
00:07.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.7 System peripheral [0880]: VMware Virtual Machine Communication Interface [15ad:0740] (rev 10)
        Subsystem: VMware Virtual Machine Communication Interface [15ad:0740]
        Kernel driver in use: vmw_vmci
00:11.0 PCI bridge [0604]: VMware PCI bridge [15ad:0790] (rev 02)
        Subsystem: VMware PCI bridge [15ad:0790]
00:15.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:17.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:17.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
[... 42 lines cut ...]
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
02:00.0 Ethernet controller [0200]: Intel Corporation 82545EM Gigabit Ethernet Controller (Copper) [8086:100f] (rev 01)
        DeviceName: Ethernet0
        Subsystem: VMware PRO/1000 MT Single Port Adapter [15ad:0750]
        Kernel driver in use: e1000
03:00.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 620 [8086:5916] (rev 02)
        DeviceName: pciPassthru0
        Subsystem: Intel Corporation Device [8086:2015]
        Kernel driver in use: i915
[    6.067104] disable_kmod.group i915-oot
[    6.786167] i915 0000:03:00.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A
[    6.786495] i915 0000:03:00.0: [drm] VT-d active for gfx access
[    6.791154] i915 0000:03:00.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0xffff
[    6.791161] i915 0000:03:00.0: [drm] Failed to find VBIOS tables (VBT)
[    6.791167] i915 0000:03:00.0: [drm] *ERROR* DC state mismatch (0x0 -> 0x2)
[    6.804309] i915 0000:03:00.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[... cut ...]
[    7.451685] i915 0000:03:00.0: [drm] GuC firmware i915/kbl_guc_33.0.0.bin version 33.0 submission:disabled
[    7.451690] i915 0000:03:00.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0 authenticated:yes
[    7.452472] [drm] Initialized i915 1.6.0 20200917 for 0000:03:00.0 on minor 0
[    7.465106] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes
[    7.477501] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes
[    7.489713] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes
root@iStoreOS:~#

szkeum, 2024-01-19:
03:00.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 620 [8086:5916] (rev 02)
        DeviceName: pciPassthru0
        Subsystem: Intel Corporation Device [8086:2015]
        Kernel driver in use: i915
[    6.067104] disable_kmod.group i915-oot
[    6.786167] i915 0000:03:00.0: [drm] Incompatible option enable_guc=3 - GuC submission is N/A
[    6.786495] i915 0000:03:00.0: [drm] VT-d active for gfx access
[    6.791154] i915 0000:03:00.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0xffff
[    6.791161] i915 0000:03:00.0: [drm] Failed to find VBIOS tables (VBT)
[    6.791167] i915 0000:03:00.0: [drm] *ERROR* DC state mismatch (0x0 -> 0x2)
[    6.804309] i915 0000:03:00.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    7.421034] i915 0000:03:00.0: [drm] failed to retrieve link info, disabling eDP
[    7.451685] i915 0000:03:00.0: [drm] GuC firmware i915/kbl_guc_33.0.0.bin version 33.0 submission:disabled
[    7.451690] i915 0000:03:00.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0 authenticated:yes
[    7.452472] [drm] Initialized i915 1.6.0 20200917 for 0000:03:00.0 on minor 0
[    7.465106] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes
[    7.477501] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes
[    7.489713] i915 0000:03:00.0: [drm] Cannot find any crtc or sizes

jjm2473, 2024-01-22:
现在还没有 /dev/dri ？

> [drm] Initialized i915 1.6.0 20200917 for 0000:03:00.0 on minor 0

看起来drm设备都注册了

closed 2024-02-18 by jjm2473