istoreos/istoreos #1283: 克隆磁盘问题

MrBrilliant2046, 2024-01-19:
Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue
A：把istoreos 所在的硬盘（64g不知名品牌）在pe环境下用DiskGenius的克隆磁盘功能克隆到了一个海康威视256g磁盘并调整了软件盘2g->133g，之后开机卡在GRUB

### (2) 路由器型号和固件版本

A：倍控软路由,j4125,8g*2

MrBrilliant2046, 2024-01-19:
> Q：是否用关键词搜索了issue? (使用 "x" 选择)
> 
> * [x ] 没有类似的issue
>   A：把istoreos 所在的硬盘（64g不知名品牌）在pe环境下用DiskGenius的克隆磁盘功能克隆到了一个海康威视256g磁盘并调整了软件盘2g->133g，之后开机卡在GRUB
> 
> ### (2) 路由器型号和固件版本
> A：倍控软路由,j4125,8g*2

最新版本（截至1-19）（开不了机不想把硬盘换回去看版本号，硬盘金手指快完了）

MrBrilliant2046, 2024-01-19:
我发现之前的GRUB里的cfg文件中貌似有指向某一扇区的UUID，（根据Google我发现大多数都是Ubuntu卡在GRUB）请问我应该怎样修改以符合现有的硬盘和分区大小。

jjm2473, 2024-01-20:
说明不是真正的克隆，MBR区域的UUID都没克隆过去。你对比下两个硬盘的第440字节和1168字节是不是都匹配，440字节可以直接修改，但是1168属于GPT，不能直接改，改了要重新生成校验码。

还是换个方式克隆吧，先启动到之前的硬盘系统，接上第二个硬盘，执行quickstart把系统安装到新硬盘，顺便插上PE启动盘，系统安装完会重启，进PE启动盘以后，再来用diskgen克隆第三分区（如果有第四分区，那也克隆，分区数量不一致就建分区）

closed 2024-02-18 by jjm2473