istoreos/istoreos #1286: iStoreOS在线升级成最新固件问题

ydl898898, 2024-01-20:
iStoreOS在线升级成最新的22.03.6-2024011910固件后open clash 守护程序不运行dns出现错误。
怎么解决这个问题

jjm2473, 2024-01-20:
试试重装openclash

jfgg95, 2024-02-20:
openclash默认会劫持本地dns解析，具体实现是设置了一层dns转发。
你可以看一下“网络-DHCP/DNS-常规设置”下面，有没有这么一条信息
[image]
有的话删除掉应该就好了。

hsy66, 2024-02-27:
> openclash默认会劫持本地dns解析，具体实现是设置了一层dns转发。 你可以看一下“网络-DHCP/DNS-常规设置”下面，有没有这么一条信息 ![图像](https://private-user-images.githubusercontent.com/28002478/306154706-db164341-82e7-42cb-9ded-f8f0532d7d90.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNDYwMzIsIm5iZiI6MTcwOTA0NTczMiwicGF0aCI6Ii8yODAwMjQ3OC8zMDYxNTQ3MDYtZGIxNjQzNDEtODJlNy00MmNiLTlkZWQtZjhmMDUzMmQ3ZDkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI3VDE0NTUzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlNTZhY2MxODRiOTNlODBkY2RkMjdiYTBhZDNkMjVlZWUxZjhlYTljNmM4Njc3MjQ4YjIxZWQxMDY3OWM5NjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-hOosi7aaPIUTS8yyYinlNbARzYP5jhPeusHVqrgV-w) 有的话删除掉应该就好了。

确实有这么一条dns劫持，但是已经删掉了还是不行，并且把openclash中的dms劫持已经停掉了还是不可以