istoreos/istoreos #1287: 更新版本后负载均衡失效了

wionch, 2024-01-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

更新版本: iStoreOS 22.03.6 2024011910
更新版本以后, 负载均衡显示运行中. 但是点击进入web 网页会打不开, 提示: ERR_CONNECTION_REFUSED
点击节点测试, 显示超时
重装haproxy和passwall或者将passwall安装回老版本也都存在同样的问题
passwall的其他功能是正常的

但是另外安装passwall2, 里面的负载均衡却可以正常使用.

### (1) 具体问题

A：负载均衡功能失效了

### (2) 路由器型号和固件版本

A：R86S

### (3) 详细日志和/或截图

A：

bcseputetto, 2024-01-20:
请打开终端执行
```
opkg update
opkg install libpcre
```
然后再试一下

wionch, 2024-01-21:
[image]
重新安装了libpcre还是没有解决.
passwall依然还是不能进入web管理界面, 功能也是失效

但是passwall2的负载均衡是正常可用的.
[image]

bcseputetto, 2024-01-21:
那你只能重置了
终端里执行haproxy 显示什么？截图看下

wionch, 2024-01-21:
[image]
重置如何操作?

bcseputetto, 2024-01-21:
@wionch haproxy在终端可以正常执行，你先去把passwall2的都关了，然后passwall那边负载均很先关闭并保存应用，再启用并保存应用，看看

wionch, 2024-01-21:
关闭了passwall2主开关

回到passwall
关闭了负载均衡,  -> 保存并应用
启用负载均衡 -> 保存并应用

还是无法进入web管理界面
负载均衡运行中, 但是测试负载均衡节点显示超时

bcseputetto, 2024-01-21:
执行

bcseputetto, 2024-01-21:
执行
```console
netstat -anp |grep ":1188"
```
截图看下

wionch, 2024-01-21:
[image]
端口没有被启用.

bcseputetto, 2024-01-21:
执行
```console
ps |grep haproxy
```
再截图

wionch, 2024-01-21:
[image]

bcseputetto, 2024-01-21:
执行
```console
ls /tmp/etc/passwall/bin
```
确认这个目录下有haproxy
然后再执行
```console
/tmp/etc/passwall/bin/haproxy
```
看下它是否正常

wionch, 2024-01-21:
[image]

bcseputetto, 2024-01-21:
```console
ls /tmp/etc/passwall/haproxy/
```
看下是否有这个目录，且目录下有 `config.cfg` 此为passwall生成的 `haproxy` 的配置文件

wionch, 2024-01-21:
[image]

config.cfg代码如下:
```
root@iStoreOS:~# ls /tmp/etc/passwall/haproxy/
config.cfg
root@iStoreOS:~# vim /tmp/etc/passwall/haproxy/config.cfg
        external-check                                      
        insecure-fork-wanted                                
                                                            
defaults                                                    
        mode                    tcp                         
        log                     global
        option                  tcplog
        option                  dontlognull
        option http-server-close           
        #option forwardfor       except 127.0.0.0/8
        option                  redispatch         
        retries                 2                  
        timeout http-request    10s                
        timeout queue           1m                 
        timeout connect         10s                
        timeout client          1m                 
        timeout server          1m        
        timeout http-keep-alive 10s
        timeout check           10s
        maxconn                 3000
                                    
resolvers mydns                     
        resolve_retries       1     
        timeout resolve       5s    
        hold valid           600s   
    nameserver dns1 192.168.7.3:53  
    nameserver dns2 8.8.8.8:53    
                                  
listen 1181                       
        bind 0.0.0.0:1181         
        mode tcp                  
        balance roundrobin        
        option external-check 
        external-check command "/usr/share/passwall/haproxy_check.sh"
    server s1.nfstream.click:13001 127.0.0.1:2082 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13001 127.0.0.1:2083 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13002 127.0.0.1:2084 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13003 127.0.0.1:2085 weight 50  check inter 60s rise 1 fall 3
    server t-06.outrdp.xyz:14001 127.0.0.1:2086 weight 50  check inter 60s rise 1 fall 3  
    server s1.nfstream.click:13053 127.0.0.1:2087 weight 50  check inter 60s rise 1 fall 3 backup
    server s1.nfstream.click:13054 127.0.0.1:2088 weight 50  check inter 60s rise 1 fall 3 backup
    server s1.nfstream.click:13056 127.0.0.1:2089 weight 50  check inter 60s rise 1 fall 3 backup
    server t-06.outrdp.xyz:13063 127.0.0.1:2090 weight 50  check inter 60s rise 1 fall 3 backup  
    server t-06.outrdp.xyz:13064 127.0.0.1:2091 weight 50  check inter 60s rise 1 fall 3 backup  
                                                                                                 
listen console                                                                                   
        bind 0.0.0.0:1188                                                                        
        mode http                                                                                
        stats refresh 30s                                                                      
        stats uri /                                                                            
        stats admin if TRUE
        stats auth admin:123123
```

bcseputetto, 2024-01-21:
```console
/tmp/etc/passwall/bin/haproxy -f /tmp/etc/passwall/haproxy/config.cfg &
```
执行这个看下passwall路径下的haproxy能否用passwall生成的haproxy配置文件运行起来。
```console
ps |grep haproxy
#检查haproxy进程
netstat -anp |grep ":1188"
#检查haproxy监听端口
~~~

wionch, 2024-01-21:
[image]
报错了.

bcseputetto, 2024-01-21:
重启一下软路由再尝试一下，顺便passwall2的负载均衡开关你没有关吗？那个负载均衡是独立于主开关的，我推荐你最好关闭它

wionch, 2024-01-21:
passwall2我关闭并卸载了

```
root@iStoreOS:~# is-opkg remove 'luci-app-passwall2' 'luci-i18n-passwall2-zh-cn'
Removing package luci-i18n-passwall2-zh-cn from root...
Removing package luci-app-passwall2 from root...
Not deleting modified conffile /etc/config/passwall2_server.
Collected errors:
 * opkg_remove_cmd: Package luci-i18n-passwall2-zh-cn is not installed.
```
然后物理重启了软路由
还是同样的问题.

[image]

bcseputetto, 2024-01-21:
@wionch 把那两行警告完整复制出来看下，有一部分被隐藏了

bcseputetto, 2024-01-21:
```console
/tmp/etc/passwall/bin/haproxy -f /tmp/etc/passwall/haproxy/config.cfg &
```
还有你执行这个的时候，后面没有加`&`吗，没加的话，你按Ctrl+C，可能它就退出了

wionch, 2024-01-21:
抱歉, 中午出去了.

[image]

### 执行命令 `/tmp/etc/passwall/bin/haproxy -f /tmp/etc/passwall/haproxy/config.cfg &` ,显示第一个红框的部分.

### 按回车键以后, 显示第二个红框的部分

`/tmp/etc/passwall/haproxy/config.cfg`代码如下:

```
root@iStoreOS:~# vim /tmp/etc/passwall/haproxy/config.cfg

        external-check                                      
        insecure-fork-wanted                                
                                                            
defaults                                                    
        mode                    tcp                         
        log                     global
        option                  tcplog
        option                  dontlognull
        option http-server-close           
        #option forwardfor       except 127.0.0.0/8
        option                  redispatch         
        retries                 2                  
        timeout http-request    10s                
        timeout queue           1m                 
        timeout connect         10s                
        timeout client          1m                 
        timeout server          1m        
        timeout http-keep-alive 10s
        timeout check           10s
        maxconn                 3000
                                    
resolvers mydns                     
        resolve_retries       1     
        timeout resolve       5s    
        hold valid           600s   
    nameserver dns1 192.168.7.3:53  
    nameserver dns2 8.8.8.8:53    
                                  
listen 1181                       
        bind 0.0.0.0:1181         
        mode tcp                  
        balance roundrobin        
        option external-check 
        external-check command "/usr/share/passwall/haproxy_check.sh"
    server s1.nfstream.click:13001 127.0.0.1:2082 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13001 127.0.0.1:2083 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13002 127.0.0.1:2084 weight 50  check inter 60s rise 1 fall 3
    server s1.nfstream.click:13003 127.0.0.1:2085 weight 50  check inter 60s rise 1 fall 3
    server t-06.outrdp.xyz:14001 127.0.0.1:2086 weight 50  check inter 60s rise 1 fall 3  
    server s1.nfstream.click:13053 127.0.0.1:2087 weight 50  check inter 60s rise 1 fall 3 backup
    server s1.nfstream.click:13054 127.0.0.1:2088 weight 50  check inter 60s rise 1 fall 3 backup
    server s1.nfstream.click:13056 127.0.0.1:2089 weight 50  check inter 60s rise 1 fall 3 backup
    server t-06.outrdp.xyz:13063 127.0.0.1:2090 weight 50  check inter 60s rise 1 fall 3 backup  
    server t-06.outrdp.xyz:13064 127.0.0.1:2091 weight 50  check inter 60s rise 1 fall 3 backup  
                                                                                                 
listen console                                                                                   
        bind 0.0.0.0:1188                                                                        
        mode http                                                                                
        stats refresh 30s                                                                      
        stats uri /                                                                            
        stats admin if TRUE
        stats auth admin:123123
```

bcseputetto, 2024-01-21:
我给你的gmail发了邮件，请注意看下

wionch, 2024-01-21:
解决啦, 多谢帮忙.

负载均衡设置的节点有问题. 清空负载均衡中的节点, 重新设置负载均衡节点就正常了.

多谢@bcseputetto

closed 2024-01-21 by wionch