istoreos/istoreos #1289: link is down 卡启动，路由器系统无法正常启动，重新刷机后恢复，断电后重启故障依旧

Yantesoft, 2024-01-21:
## 2. 详细叙述
安装系统后，若断电关机再次开机后系统会提示link is down，且无法进入系统，CPU占用百分之百，已经尝试多次重新刷机，刷机后正常，只要设备断电重启后就会再次遇到这个问题。

系统是.6的版本，拨号上网做主路由，仅仅设置了这些。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志和/或截图

A：

jjm2473, 2024-01-22:
机型是什么？是否最新固件20240119？

Yantesoft, 2024-01-22:
系统是[istoreos-22.03.6-2024011910-x86-64-squashfs-combined.img.gz](https://fw0.koolcenter.com/iStoreOS/x86_64/istoreos-22.03.6-2024011910-x86-64-squashfs-combined.img.gz)

Yantesoft, 2024-01-22:
网卡是两个： Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller。

Yantesoft, 2024-01-22:
J1900的小主机

Yantesoft, 2024-01-22:
不确定是否是真的CPU占用100%，反正就是键盘无响应，键盘拔掉以后再插入键盘的灯也不亮了，应该是系统内核卡死了，我看日志说修复了网卡死锁的问题，可能卡死是因为别的吧，现在状况是有概率能启动，有概率是卡死。

jjm2473, 2024-01-22:
0119已经修复了驱动的bug，不会再卡死。你说的断电重启是软重启，或者poweroff以后重新上电就会触发吗

Yantesoft, 2024-01-22:
按下自带的重启键或者直接拔掉电源重启，都会有概率卡死，很神奇的是有概率还不是每次都卡死（大部分时候卡死），卡死的时候系统最后一行是pppoe拨号的或者是link become ready    如果拔掉网线就卡在link is down

Yantesoft, 2024-01-22:
但是每次重新刷好系统，第一次的时候都能进入不会卡死

Yantesoft, 2024-01-22:
目前重启暂时没问题了，我已经尝试了好多次，好像是突然好了，我也不知道调整了哪里，如果再次出现问题我会在这个帖子再更新症状和一些记录，感谢[jjm2473](https://github.com/jjm2473)的帮助和耐心询问。

jjm2473, 2024-01-23:
安装系统以后马上改密码，以免被攻击，要是中了挖矿木马，也会导致cpu和网络占用异常，如果其他类型的病毒，那出现什么情况都不奇怪

Yantesoft, 2024-01-23:
[image]
很不幸今天又卡死了，正装是这样，首先跑码，我看系统跑码分为两个阶段，一个阶段是字体稍大一些的，另外一个是字体稍微小一些的，字体大的全部可以跑过，字体小的在重启的时候又开始卡了，卡在PPPOE拨号这个阶段，感觉是拨号成功以后卡死的，因为路由器的可以上网的灯是亮的，路由器重启以后，192.168.100.1会可以相应几秒（很短，3秒内），然后就会卡死，路由器的灯也会变成黄色（不可上网，这个显示的会慢一些，因为不是实时刷新的），这个时候跑码的光标不继续闪了，也不会有新的码出现，插入键盘以后键盘灯不亮，处于卡死状态。但是有概率重启是可以正常进入系统的，概率非常低，大约10%左右。密码重装的时候就改了，应该不是木马所为

Yantesoft, 2024-01-23:
上述有一丝错误，不应该是卡在PPPOE，而是拨号成功以后网卡会link is down，两个网卡都会掉，然后系统卡死（大概率是这样）也有小概率直接在PPPOE拨号阶段就卡死。如果拔掉网线会在link becom ready卡死，网卡是 Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller 。系统是[istoreos-22.03.6-2024011910-x86-64-squashfs-combined.img.gz](https://fw0.koolcenter.com/iStoreOS/x86_64/istoreos-22.03.6-2024011910-x86-64-squashfs-combined.img.gz) 主机配置：J1900 2GB+8GBSSD（系统所在盘） 作为主路由，路由器中级模式当AP，上网方式是PPPOE拨号

Yantesoft, 2024-01-23:
[image]
补yizh一张在PPPOE阶段卡死的图

jjm2473, 2024-01-24:
在系统-杂项设置-驱动里面，勾选r8125和r8168，保存应用，然后重启设备

jjm2473, 2024-01-24:
[image]

这个确定是卡死吗？按回车有没有反应？

jjm2473, 2024-01-24:
> 我看系统跑码分为两个阶段，一个阶段是字体稍大一些的，另外一个是字体稍微小一些的

字体大是还没加载显卡驱动。大部分网卡驱动都是后期加载的，也就是跟显卡驱动差不多时间加载，再加上卡死是网络初始化一段时间后才发生的，显卡驱动早已加载完，所以都是在字体小的时候才会卡死。总之字体大小跟网络没什么关系。

jjm2473, 2024-01-24:
你还可以试试不拨号，而是使用dhcp客户端的情况下，重启多次会不会卡死

Yantesoft, 2024-01-24:
> ![pppoe卡死](https://private-user-images.githubusercontent.com/81399642/299060929-7a590907-4dde-471b-9907-1c8654872d44.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwNzMzMTksIm5iZiI6MTcwNjA3MzAxOSwicGF0aCI6Ii84MTM5OTY0Mi8yOTkwNjA5MjktN2E1OTA5MDctNGRkZS00NzFiLTk5MDctMWM4NjU0ODcyZDQ0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI0VDA1MTAxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjYzc0M2NkNTM4YmI0ZWE5N2Y0NjI3NTQxYjUzYTU5YWRhZWVkNDQwM2Q5MmY0NDY2Mjg0NWJkMzQ1Nzg3MjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NF3uoBA6GYE7uza5uy8GBDIWTmKqDeIrobahi9neLDg)
> 
> 这个确定是卡死吗？按回车有没有反应？

确定是卡死，回车无响应，这个时候键盘灯是亮的，如果拔掉键盘重新插键盘就不亮了，说明系统已经卡死了

Yantesoft, 2024-01-24:
> > 我看系统跑码分为两个阶段，一个阶段是字体稍大一些的，另外一个是字体稍微小一些的
> 
> 字体大是还没加载显卡驱动。大部分网卡驱动都是后期加载的，也就是跟显卡驱动差不多时间加载，再加上卡死是网络初始化一段时间后才发生的，显卡驱动早已加载完，所以都是在字体小的时候才会卡死。总之字体大小跟网络没什么关系。

明白的，我只是想描述清楚，好让你了解到系统在什么地方卡死的

Yantesoft, 2024-01-24:
昨天在系统加载过程中，在跑码过程中我按下回车然后输入top有反应，可以看到系统进程，但是因为top3秒才刷新一次，也没看到什么结果，感觉像是在等锁，而且键盘重新插入灯就不亮了，如果是资源耗尽一般情况下灯还是能亮的，感觉应该是是锁死了

Yantesoft, 2024-01-24:
希望系统越来越好，兼容性也越来越好，我也是一个程序员，不过是web方向的，如果需要配合解决问题，可以随时回复我，我会一直配合。

jjm2473, 2024-01-24:
我测试了下h28k，它的一个网卡是pcie的RTL8111HS，驱动也换成了r8169，没复现卡死的问题。可能你的情况只是个例，或者特定型号的网卡才能触发，建议你先使用r8168驱动。

关于调试：

如果你能通过ttl/串口连接设备，那么在卡死的时候，使用sysrq触发内核崩溃，或者触发阻塞任务dump，把内核日志保存下来，这个日志也许能看到代码锁死的位置。

开机时尽快执行 `echo 1 > /proc/sys/kernel/panic_on_rcu_stall` ，当rcu检测到卡死的时候就会自动触发内核崩溃，打印日志，但是也需要ttl/串口才方便保存。

Yantesoft, 2024-01-24:
我不太了解硬件的linux，我对linux的了解来源于云服务器，部署项目的时候会用到，设计硬件知识的地方是我的盲区，ttl/串口连接需要额外的数据线或者其他设备吗？如果只是要系统日志的话我这里有采集卡，也许能把系统卡死前的日志输出全部录屏给你，这样的日志我也不清楚是否完全，也不太方便查看。如果有关于ttl的硬件你可以告诉我需要购买什么东西，我去看一下，另外你说在其他设备复现没能找到bug的问题，我的设备也是，刚刷入系统一切正常，只是在重启的时候会出现，也不是每次出现，大概就是这一段时间重启都没问题，过一段时间重启都会卡死，再多重启几次又有概率能进入系统。

jjm2473, 2024-01-24:
视频采集文件太大了，如果能抓到关键日志也可以，只保留关键部分。为了避免键盘无法输入，开机前先拔掉wan口网线，然后进终端执行`echo kernel.panic_on_rcu_stall=1 >/etc/sysctl.d/00-stall-debug`，然后重启下。

串口/ttl：如果你的设备没有串口，那最简单的方式就是买个USB转ttl的模块，另一端放个电脑接收ttl数据，一般也是用usb转ttl，电脑使用串口通信软件就能连接了（如果是linux建议picocom），设置波特率115200。istoreos启动时选择的grub界面会等待一秒钟，选择“iStoreOS (ttyUSB0,115200)”这一行回车，ttl应该就有输出了，而且能作为终端控制istoreos。

Yantesoft, 2024-01-24:
ttl我好像有一个小东西，当时用来给Arduino下载固件的，是一个东西吗

Yantesoft, 2024-01-24:
3V3 RXD TXD GND 我的ttl转usb上有这些口，还有另外的IO0、IO2、EN和RST，J1900主板上有6个COM口，我应该找到主板的rx、tx和gnd连上是吗

jjm2473, 2024-01-26:
如果有com口，那你可以试试连接第一个com口，然后启动时grub界面选择“iStoreOS (ttyS0)”

marlunzhang, 2024-02-03:
> ![fcabf1a102d4f0ddae45158b38ecd80](https://private-user-images.githubusercontent.com/81399642/299056718-aada2920-6a02-4d42-bfed-0ec754be5338.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY5NDQ1OTUsIm5iZiI6MTcwNjk0NDI5NSwicGF0aCI6Ii84MTM5OTY0Mi8yOTkwNTY3MTgtYWFkYTI5MjAtNmEwMi00ZDQyLWJmZWQtMGVjNzU0YmU1MzM4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAzVDA3MTEzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQwNjdmMzEzNDNjNDhmNDUzMmIwMjU0YmUyYjYwZDI4M2JmMWFjZTY5NmMzNmNiZmQ2MWUxMDc5OTkwMTdiMmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.gpK5uiKR9FdxezP3sJ3ufP_0GMlsQhhrEGMsD6qUb6A) 很不幸今天又卡死了，正装是这样，首先跑码，我看系统跑码分为两个阶段，一个阶段是字体稍大一些的，另外一个是字体稍微小一些的，字体大的全部可以跑过，字体小的在重启的时候又开始卡了，卡在PPPOE拨号这个阶段，感觉是拨号成功以后卡死的，因为路由器的可以上网的灯是亮的，路由器重启以后，192.168.100.1会可以相应几秒（很短，3秒内），然后就会卡死，路由器的灯也会变成黄色（不可上网，这个显示的会慢一些，因为不是实时刷新的），这个时候跑码的光标不继续闪了，也不会有新的码出现，插入键盘以后键盘灯不亮，处于卡死状态。但是有概率重启是可以正常进入系统的，概率非常低，大约10%左右。密码重装的时候就改了，应该不是木马所为

我也碰到了一样的问题，也是J1900日志也一样，从零开始烧录U盘启动就卡死在这个界面，重启几次也进不去，系统是istoreos-22.03.6-2024012613-x86-64-squashfs-combined

Yantesoft, 2024-02-03:
我尝试取日志，没取到，已暂时2改用openwrt 了

qq532669879, 2024-02-25:
我也是j1900小主机，同样的问题，卡在br-lan:port1  eth0/1 link is down，更新了多个版本故障依旧，求解决

jjm2473, 2024-02-25:
可能是网卡驱动导致的死锁，遇到这个问题的人先刷 istoreos-22.03.5-2023122916 版本，然后进后台页面，系统-杂项设置-驱动，勾选 r8168 驱动，保存应用，然后再保留配置刷最新的固件（例如在线升级或者刷已经下载好的固件，但是要保留配置）。之后如果能进系统，就可以证明默认的 r8169驱动有bug

jjm2473, 2024-02-25:
对于已经升级的用户，不想重刷老版本，可以使用以下步骤切换到r8168驱动：
1. 接上显示器和键盘
2. 启动到安全模式。也就是开机时在 GRUB 菜单选择“iStoreOS (failsafe)”（开机过程中键盘重复快速按方向下键，中断自动启动，等出现菜单就可以按方向键上下选择了），然后回车。稍候日志刷屏，等几秒钟屏幕显示“root@(none):/# ”就表示已启动到安全模式的终端了。
3. 安全模式是没有网络的，没法ssh或网页控制，终端只能手打命令。一行一行执行以下命令（输入一行后回车，然后等到屏幕再次显示“root@(none):/# ”才可以输入下一行。如果哪一行输错了，或者屏幕被其他日志干扰，可以按 Ctrl+C 重新输入）：
  ```shell
mount_root
uci add_list kmods.kmods.enable=r8168
uci commit kmods
sync /
  ```
  注意：第一条命令执行需要几秒钟，且会有一些日志输出，不要理会。后面的命令都应该没有输出，如果有输出说明有错误。
4. 最后执行`reboot`，系统会重启到正常状态

qq532669879, 2024-02-26:
感谢您的回复，平台是翔升J1900M-HK主板，安装4GB单条内存，SATA硬盘2块，内置网卡网卡是rtk8111F，外置一块pcieX1的网卡8111E，我按照您给出的方法切换了驱动，卡死的概率大幅降低，但还是会出现卡死情况。以前卡死的概率是90%，要重启很多次才能进入一次。切换驱动后，卡死的概率是30%左右，但还是会有卡死的情况。

 
出问题的网卡型号是什么？

jjm2473, 2024-02-26:
@qq532669879 换驱动以后，把内核启动日志也发出来看看，在后台的状态-系统日志-内核日志里面，开机以后尽快全部复制出来就行

jjm2473, 2024-02-26:
大家是不是拨号的时候才有这个问题？

还有，执行下`lspci -nnk | grep r816 -B2`看看结果是什么

querypeng, 2024-07-11:
我的j1900小主机也有类似问题
[image]
[image]

TZP001, 2024-10-31:
同问，安装的时候就出现了这个问题

xd19861224, 2025-02-21:
同问如上图

foanfan-qq, 2025-05-20:
我也是这个问题，请问解决了吗

jjm2473, 2025-05-23:
分别试试22.03，24.10固件看能不能用。不行的话，就试试openwrt官方的固件有没有这个问题。

wzlqiang, 2025-06-16:
[image]
J1900小鸡，也碰到这个问题了，22.03.5/6/7，24.14都试过了，通通卡死，卡死率100%，进failsafe 输入命令后teboot重启无反应。到底是哪里的问题呢？

jjm2473, 2025-06-16:
@wzlqiang 可能内核不兼容了吧，试试openwrt官方的固件是不是也这样，如果openwrt有办法解决那我们也会合并。

jjm2473, 2025-06-28:
> Openwrt官方固件没问题

你测试的官方固件是什么版本