istoreos/istoreos #1292: 是否可以在istoreos中开起docker的ipv6地址，以及适配KVM虚拟机对ARM环境的适配

814747889, 2024-01-22:
### (2) 路由器型号和固件版本

A：r5s

### (3) 详细日志和/或截图

A：想在开起docker的ipv6地址，以及适配KVM虚拟机对ARM环境的适配

GreenTeodoro839, 2024-02-07:
> ipv6跟系统没关系，本来就支持，只要你会配置。
> 
> kvm支持，你可以关注openwrt，如果openwrt支持了，那istoreos也可以支持。不过我觉得没啥用，因为arm的不同芯片厂商的虚拟化标准都不一样，不像x86就两家。如果你觉得kvm没有硬件加速虚拟化也能接受，那可以给openwrt提下。

把这些添加到target/linux/rockchip/armv8/config-内核版本
``CONFIG_VIRTUALIZATION=y`
CONFIG_KVM=y
CONFIG_KVM_ARM_HOST=y
CONFIG_KVM_GENERIC_DIRTYLOG_READ_PROTECT=y
CONFIG_KVM_INDIRECT_VECTORS=y
CONFIG_KVM_MMIO=y
CONFIG_KVM_VFIO=y
CONFIG_VHOST_NET=y``
然后qemu包需要自己编译，去lede仓库找到软件包，修改Makefile加上aarch64就能编译出来

jjm2473, 2024-02-09:
关键是，没有硬件虚拟化性能估计很差

GreenTeodoro839, 2024-02-09:
> 关键是，没有硬件虚拟化性能估计很差

有啊，这就是开启kvm

jjm2473, 2024-02-18:
KVM只是个虚拟化框架，不一定是硬件虚拟化

jjm2473, 2024-02-18:
你自己有使用其他系统（例如debian）测试过自己的板子的kvm性能吗？

GreenTeodoro839, 2024-02-18:
> 你自己有使用其他系统（例如debian）测试过自己的板子的kvm性能吗？

树莓派4b用debian开虚拟机试过，和真机性能差不多，开发版跑openwrt有点性能过剩，开虚拟机能充分利用资源

jjm2473, 2024-02-18:
有什么场景必须用kvm，而不能用docker实现的？

GreenTeodoro839, 2024-02-19:
> KVM只是个虚拟化框架，不一定是硬件虚拟化

qemu是虚拟化框架，kvm是内核级虚拟化支持

GreenTeodoro839, 2024-02-19:
> 有什么场景必须用kvm，而不能用docker实现的？

ha智能家居debian docker才能用完整版，还有虚拟机可以单独ip装web服务器做网站

jjm2473, 2024-02-20:
docker可以单独ip啊，只是默认情况下使用docker0这个桥而已，如果需要桥接到lan口，那docker创建个网络，把br-lan作为桥就行了，如果需要桥接到wan，docker使用macvlan创建网络也可以。以网络功能来说，虚拟机跟docker不会有任何差别，虚拟机能做的docker一样可以，docker没法做到的虚拟机一样也不行。

至于ha，应该只是因为你使用的镜像不包含systemd而已，找个支持systemd的debian docker镜像，然后里面再启动个ha的docker容器不就行了，虚拟机里面不还是一样要再启动一个docker。

GreenTeodoro839, 2024-02-20:
> docker可以单独ip啊，只是默认情况下使用docker0这个桥而已，如果需要桥接到lan口，那docker创建个网络，把br-lan作为桥就行了，如果需要桥接到wan，docker使用macvlan创建网络也可以。以网络功能来说，虚拟机跟docker不会有任何差别，虚拟机能做的docker一样可以，docker没法做到的虚拟机一样也不行。
> 
> 至于ha，应该只是因为你使用的镜像不包含systemd而已，找个支持systemd的debian docker镜像，然后里面再启动个ha的docker容器不就行了，虚拟机里面不还是一样要再启动一个docker。

openwrt主路由用docker macvlan容器没法联网，但是外面可以访问容器，旁路由没这个问题，istoreos有解决方法吗

jjm2473, 2024-02-20:
那不是废话吗，主路由情况下wan口不是要拨号吗？你用macvlan以后，容器里面有拨号吗？如果没法多播，为什么不直接桥接到br-lan？

jjm2473, 2024-02-20:
还有你说的ha完整功能，用这个镜像应该就行了 https://github.com/home-assistant/supervisor/pkgs/container/aarch64-hassio-supervisor

jjm2473, 2024-02-20:
我测试了下ha supervisor，没啥问题，这还是没使用systemd的情况下，只用了istoreos默认的docker。

[image]

[image]

jjm2473, 2024-02-20:
这是我的部署命令：
```bash
docker run -d \
  --name hassio_supervisor \
  --privileged \
  --restart unless-stopped \
  --dns=172.17.0.1 \
  -e TZ=Asia/Taipei \
  -v /var/run/docker.sock:/var/run/docker.sock \
  -v /var/run/dbus:/var/run/dbus \
  -v /run/udev:/run/udev \
  -v /mnt/data/hassio:/data \
  -e SUPERVISOR_SHARE="/mnt/data/hassio" \
  -e SUPERVISOR_NAME="hassio_supervisor" \
  -e HOMEASSISTANT_REPOSITORY="homeassistant/qemuarm-64-homeassistant" \
  ghcr.io/home-assistant/aarch64-hassio-supervisor:2024.01.1

```

把命令里面的两个`/mnt/data/hassio`改成自己希望在不同容器共享数据的路径就行了

GreenTeodoro839, 2024-02-20:
> 这是我的部署命令：
> 
> ```shell
> docker run -d \
>   --name hassio_supervisor \
>   --privileged \
>   --restart unless-stopped \
>   --dns=172.17.0.1 \
>   -e TZ=Asia/Taipei \
>   -v /var/run/docker.sock:/var/run/docker.sock \
>   -v /var/run/dbus:/var/run/dbus \
>   -v /run/udev:/run/udev \
>   -v /mnt/data/hassio:/data \
>   -e SUPERVISOR_SHARE="/mnt/data/hassio" \
>   -e SUPERVISOR_NAME="hassio_supervisor" \
>   -e HOMEASSISTANT_REPOSITORY="homeassistant/qemuarm-64-homeassistant" \
>   ghcr.io/home-assistant/aarch64-hassio-supervisor:2024.01.1
> ```
> 
> 把命令里面的两个`/mnt/data/hassio`改成自己希望在不同容器共享数据的路径就行了，注意，可能只有`/mnt/`的子目录才支持

这个可以用，那这样kvm就没大有必要了，多开系统本身还浪费资源

138161, 2024-02-26:
第十一、Docker安装最小化linux
1、Docker开启IPV6，注意Docker 版本20.10.22
vim /etc/config/dockerd
config globals 'globals'
        option log_level 'warn'
        option iptables '1'
        option data_root '/mnt/sata2-4/docker'
        list registry_mirrors 'https://docker.mirrors.ustc.edu.cn/'
        option experimental '1'
        option ipv6 '1'
        option ip6tables '1'
        option fixed_cidr_v6 'fd01::/80' #不能使用fd00，否则会 v6无法ping通

config firewall 'firewall'
        option device 'docker0'
        list blocked_interfaces 'wan'
2、重启

jjm2473, 2024-02-27:
@138161  不要使用docker的iptables，除非你不担心docker容器端口被开放到公网。有端口开放需求，用系统的防火墙单独配置开放特定端口

SB-CNM, 2025-03-09:
> [@138161](https://github.com/138161) 不要使用docker的iptables，除非你不担心docker容器端口被开放到公网。有端口开放需求，用系统的防火墙单独配置开放特定端口

太专业了哥，docker用的少就是跑过些ha服务。最近整了个istore想在内置的docker开ipv6咋操作啊。目的是为了让istore内置容器跑的网心云支持v6