istoreos/istoreos #1293: N100 CPU 2.5G网口和千兆网关连接拨号只有百兆

zkxi, 2024-01-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
wan口和光猫千兆网络连接后只有百兆，偶尔重启系统可以解决，现在重启两三次依旧百兆

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：
畅网N100先锋版V2 四个2.5G网口

iStoreOS 22.03.6 2024011910

### (3) 详细日志和/或截图

A：
[image]

gowy222, 2024-01-22:
估计是百兆宽带 或者 光猫是百兆网口.... 毕竟网卡一般都是自动交互协商速率的。企业级交换机可以手动设置网卡速率。

zkxi, 2024-01-22:
> 估计是百兆宽带或者光猫是百兆网口....最后中断一般都是自动协商减速到10兆的。企业级交换机可以手动设置带宽。有时候流量会自动协商降速到10兆 .. .唤醒之后会恢复...

宽带是千兆的，光猫网口也是千兆的，今天更新了最新固件后重启也不行了，之前重启后就会恢复到千兆

jjm2473, 2024-01-22:
一般千兆设备协商到百兆都是因为网线不行。当然，也不排除硬件的问题。不过我遇到和听到的全是因为网线问题。

zkxi, 2024-01-22:
先手动改设置速率了，ethtool -s eth0 speed 1000

zkxi, 2024-01-22:
> 一般千兆设备协商到百兆都是因为网线不行。当然，也不排除硬件的问题。不过我遇到和听到的全是因为网线问题。

换网线试过，刚买的7类网线也不行，现在插得是个超六类

gowy222, 2024-01-22:
> > 一般千兆设备协商到百兆都是因为网线不行。当然，也不排除硬件的问题。不过我遇到和听到的全是因为网线问题。
> 
> 换网线试过，刚买的7类网线也不行，现在插得是个超六类

网卡型号和固件驱动 对的上 有没有排查？

jjm2473, 2024-01-22:
千兆哪需要7类6类，只要网线质量过关，cat5e都行了。建议找个扁网线试试，我没见过扁网线会出问题，圆网线质量参差不齐。

jjm2473, 2024-01-22:
还有，不要随便乱用屏蔽线，屏蔽线要求两端都接地，如果不满足条件，效果可能反而不如非屏蔽线。

zkxi, 2024-01-22:
> 千兆哪需要7类6类，只要网线质量过关，cat5e都行了。建议找个扁网线试试，我没见过扁网线会出问题，圆网线质量参差不齐。

5类，超5，六类，超六，甚至还买了一个七类，之前版本的固件都可以重启解决，iStoreOS 22.03.6 2024011910这个固件更新后，重启也不行了，应该不是网线问题，eth1是内网协商2.5G网速没问题，应该不是网卡和驱动不匹配问题

jjm2473, 2024-01-22:
以前都要重启才能正常识别，说明本来就有问题啊，正常情况下都是协商到最高速度的。你看看网卡是什么型号，驱动是什么，执行`lspci -nnk; ethtool -i eth0;ethtool -i eth1`看看

zkxi, 2024-01-22:
> 以前都要重启才能正常识别，说明本来就有问题啊，正常情况下都是协商到最高速度的。你看看网卡是什么型号，驱动是什么，执行`lspci -nnk; ethtool -i eth0;ethtool -i eth1`看看

00:00.0 Host bridge [0600]: Intel Corporation Device [8086:461c]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
lspci: Unable to load libkmod resources: error -12
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:46d1]
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: i915
00:0d.0 USB controller [0c03]: Intel Corporation Device [8086:464e]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: xhci_hcd
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:54ed]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: xhci_hcd
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:54ef]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:15.0 Serial bus controller [0c80]: Intel Corporation Device [8086:54e8]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:15.1 Serial bus controller [0c80]: Intel Corporation Device [8086:54e9]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:54e0]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1a.0 SD Host controller [0805]: Intel Corporation Device [8086:54c4]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: sdhci-pci
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:54b8]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1c.6 PCI bridge [0604]: Intel Corporation Device [8086:54be]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.0 PCI bridge [0604]: Intel Corporation Device [8086:54b0]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.1 PCI bridge [0604]: Intel Corporation Device [8086:54b1]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.2 PCI bridge [0604]: Intel Corporation Device [8086:54b2]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1d.3 PCI bridge [0604]: Intel Corporation Device [8086:54b3]
	Subsystem: Intel Corporation Device [8086:7270]
	Kernel driver in use: pcieport
00:1e.0 Communication controller [0780]: Intel Corporation Device [8086:54a8]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1e.3 Serial bus controller [0c80]: Intel Corporation Device [8086:54ab]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:5481]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:54c8]
	DeviceName: Onboard - Sound
	Subsystem: Intel Corporation Device [8086:7270]
00:1f.4 SMBus [0c05]: Intel Corporation Device [8086:54a3]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
00:1f.5 Serial bus controller [0c80]: Intel Corporation Device [8086:54a4]
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Device [8086:7270]
01:00.0 Non-Volatile memory controller [0108]: ADATA Technology Co., Ltd. Device [1cc1:633a] (rev 01)
	Subsystem: ADATA Technology Co., Ltd. Device [1cc1:633a]
	Kernel driver in use: nvme
02:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
03:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
04:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
05:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
	Subsystem: Intel Corporation Device [8086:0000]
	Kernel driver in use: igc
06:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983 [144d:a808]
	Subsystem: Samsung Electronics Co Ltd SSD 970 EVO Plus 1TB [144d:a801]
	Kernel driver in use: nvme
driver: igc
version: 5.10.201
firmware-version: 
expansion-rom-version: 
bus-info: 0000:02:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes
driver: igc
version: 5.10.201
firmware-version: 
expansion-rom-version: 
bus-info: 0000:03:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes

zkxi, 2024-01-22:
这个机器网卡型号是Intel i226-V

jjm2473, 2024-01-22:
你尝试过wan口连其他千兆设备吗，会不会出现协商到百兆的情况

zkxi, 2024-01-22:
> 你尝试过wan口连其他千兆设备吗，会不会出现协商到百兆的情况

电脑是千兆网卡插上后协商就是千兆，只有插光猫上偶尔协商百兆，可以通过插拔网线或者重启解决

jjm2473, 2024-01-23:
要不让运营商换个光猫试试，顺便让他们检查下网线 😆

zkxi, 2024-01-23:
> 要不让运营商换个光猫试试，顺便让他们检查下网线 😆

这光猫插电脑上就自动协商到千兆，目前已经联系运营商了，改天来换个光猫试试

zkxi, 2024-03-20:
和光猫的兼容问题，换了一个其他的光猫后自动协商1000M了

closed 2024-04-03 by jjm2473

shen-lan, 2024-10-08:
。。。。刚进控制台发现首页网口那里都不显示了，于是尝试 系统便利工具-修复系统软件， 修复完毕能显示了，但是wan口 eth0显示 百兆(之前都是正常的)  这个怎么破😭

shen-lan, 2024-10-08:
```
root@iStoreOS:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                2500baseT/Full 
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: on
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```