istoreos/istoreos #1294: curl: (60) SSL certificate problem: self signed certificate

Nedved-liao, 2024-01-22:
# 版本和系统信息

iStoreOS 22.03.5, 2023122916

系统时间没问题的，
date查看
Mon Jan 22 17:57:59 CST 2024

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ x ] 没有类似的issue

已有一个相似的issue，不过里面没有相应解决
https://github.com/istoreos/istoreos/issues/1112

## 2. 详细叙述

### (1) 具体问题

A：安装后正常的，过一段时间后opkg update报错
```bash
root@iStoreOS:~# opkg update
Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
curl: (60) SSL certificate problem: self signed certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
*** Failed to download the package list from https://istore.linkease.com/repo/all/compat/Packages.gz

```
使用curl 访问https全都不可用
```bash
root@iStoreOS:~# curl -v https://baidu.com
*   Trying 39.156.66.10:443...
* Connected to baidu.com (39.156.66.10) port 443 (#0)
* ALPN: offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: none
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: self signed certificate
* Closing connection 0
curl: (60) SSL certificate problem: self signed certificate
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

```
使用如下方式尝试，还是不行
1. 使用opkg install ca-bundle  ca-certificates --force-reinstall    --force-overwrite
2. 从https://curl.se/docs/sslcerts.html下载替换ca证书/etc/ssl/certs/ca-certificates.crt
3. curl时候执行指定capath`curl --capath /etc/ssl/certs  -v https://baidu.com`

jjm2473, 2024-01-22:
没发现有这个问题。你引用的那个issue很明显是系统时间不对。

Nedved-liao, 2024-01-22:
我是在pve上面运行的，刚才重新安装了一个发现也是这个问题
[image]

jjm2473, 2024-01-22:
你还是检查下自家网络里有没有去广告插件，而且是https去广告，如果没有，那估计是被黑客攻击了（或运营商攻击）

closed 2024-01-22 by jjm2473