istoreos/istoreos #1296: 应用商店里的应用不能更新？

hui-f, 2024-01-23:
[image]
[image]
应用商店里面的软件更新，总是更新不了

hui-f, 2024-01-23:
处理器：n100，系统：22.03.6 2024011910

jjm2473, 2024-01-23:
[image]

我倒是想问问你用的什么软件源呢，`op.dllkids.xyz`既不是openwrt官方仓库，也不是国内有名的镜像地址，也不是istore提供的软件源，安装出问题关istore屁事？

closed 2024-01-23 by jjm2473

hui-f, 2024-01-23:
> <img alt="image" width="395" src="https://private-user-images.githubusercontent.com/3896703/298823841-e029a5bb-7f73-4f0a-8c33-1e9d2b3ca18c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwMDAyOTcsIm5iZiI6MTcwNTk5OTk5NywicGF0aCI6Ii8zODk2NzAzLzI5ODgyMzg0MS1lMDI5YTViYi03ZjczLTRmMGEtOGMzMy0xZTlkMmIzY2ExOGMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMjNUMDg1MzE3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzQ0MGIxYzY5MmYwNDBlNmJlYjQ0NzE2MjBkMGEzNDI0ZDdiYmM0MTUyYzAxODhmZWFiZWU1YzcxYjIwZDVjZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hEmb5SblnCtWOIofpY3esAwHUp52FhgfsSdBnTddD2I">
> 我倒是想问问你用的什么软件源呢，`op.dllkids.xyz`既不是openwrt官方仓库，也不是国内有名的镜像地址，也不是istore提供的软件源，安装出问题关istore屁事？

换源了还是失败，怎么解决，重新安装吗？
[image]

hui-f, 2024-01-23:
> <img alt="image" width="395" [...]

换源了，还是不行，怎么解决，重新安装吗？
[image]

jjm2473, 2024-01-23:
看看能不能卸载掉之前的，如果不行，那就恢复出厂设置最简单