istoreos/istoreos #1298: 远程安全反馈咨询

pili-book, 2024-01-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我开启了路由的远程访问功能，能够直接远程访问路由，但是安全问题有点担忧，不知道远程安全除了密码保护，还有什么方法保证安全，比如类似威联通之类的能够密码错误5次就锁这个ip5分钟这种

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R6S
iStoreOS 22.03.5 2023121510 / LuCI istoreos-22.03 branch git-23.342.39264-37c783a

### (3) 详细日志和/或截图

A：

jjm2473, 2024-01-23:
ddnsto内网穿透不是本身就有密码保护吗？如果是其他的穿透服务，就看对应的服务有没有验证

jinxiao, 2024-01-25:
先用openvpn，然后访问内网，安全多了，攻击面也小

jjm2473, 2024-01-26:
根据openwrt论坛的说法，要使用ssh或者vpn之类的创建安全连接（端口映射/端口转发），通过这个连接来访问。 https://forum.openwrt.org/t/limiting-login-attempts-was-is-robzrs-beardropper-unmaintained/31695/5

而且luci（uhttpd）默认才允许3个连接数，要是直接开放到公网，就算密码不被攻破，可能也会被攻击耗尽连接数，导致自己没法访问。

pili-book, 2024-01-26:
好的，感谢普及

pili-book, 2024-01-26:
> 根据openwrt论坛的说法，要使用ssh或者vpn之类的创建安全连接（端口映射/端口转发），通过这个连接来访问。 https://forum.openwrt.org/t/limiting-login-attempts-was-is-robzrs-beardropper-unmaintained/31695/5
> 
> 而且luci（uhttpd）默认才允许3个连接数，要是直接开放到公网，就算密码不被攻破，可能也会被攻击耗尽连接数，导致自己没法访问。

istoreos的vpn在哪里了，好像没有自带的？

jjm2473, 2024-02-01:
@pili-book ssh不就可以做端口转发吗

closed 2024-02-02 by jjm2473