istoreos/istoreos #1300: wan_6 这个错误提示还是觉得应该修正下。

haobinnan, 2024-01-23:
别的固件点击编辑没有错误提示。
[image]

[image]

设备型号
FastRhino R66s
固件版本
iStoreOS 22.03.6 2024011910

zctmdc, 2024-01-24:
手动创建一个就好了 :)
1. 创建接口 wan_6
2. 指定协议和设备 建议选定 `pppoe-wan` 而不是 `@wan` 总是获取不到地址，需要修改一下配置才行（比如 修改IPv6 分配长度）
3. 其他按照图
[image]

[image]
[image]
[image]
4. 建议这里按图配置，不然上不了
[image]

jjm2473, 2024-01-26:
你看仔细了，wan6和wan_6不是同一个东西，wan_6是pppoe创建的虚拟接口，没得配置，相关配置都在pppoe里面

closed 2024-01-26 by jjm2473

jjm2473, 2024-01-26:
之所以其他固件没有wan_6，那只是因为其他固件的版本太老了，openwrt 21版本以上，pppoe拨号的情况下都有wan_6。当然，界面上应该要让wan_6的编辑和删除按钮变灰的，openwrt目前好像还没做。

haobinnan, 2024-01-26:
> 之所以其他固件没有wan_6，那只是因为其他固件的版本太老了，openwrt 21版本以上，pppoe拨号的情况下都有wan_6。当然，界面上应该要让wan_6的编辑和删除按钮变灰的，openwrt目前好像还没做。

是的 我上面的截图 的确是21如果能将编辑与删除 禁用掉最好。免得发生奇异。