istoreos/istoreos #1302: 系统日志出现大量 关于 daemon.err nmbd 的信息

haobinnan, 2024-01-24:
Wed Jan 24 09:37:36 2024 daemon.err nmbd[13191]: [2024/01/24 09:37:36.664428,  0] ../../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Wed Jan 24 09:37:36 2024 daemon.err nmbd[13191]:   query_name_response: Multiple (2) responses received for a query on subnet 192.168.1.1 for name WORKGROUP<1d>.
Wed Jan 24 09:37:36 2024 daemon.err nmbd[13191]:   This response was from IP 192.168.1.12, reporting an IP address of 192.168.1.12.

设备型号
FastRhino R66s
固件版本
iStoreOS 22.03.6 2024011910

haobinnan, 2024-01-24:
[image]

haobinnan, 2024-01-25:
将默认的组名改名后好了，没有额外的日志了，请问会有什么别的问题吗？

uci set samba4.@samba[-1].workgroup='OPENWRT'
uci commit samba4

@jjm2473

jjm2473, 2024-01-26:
这个警告应该是因为你的192.168.1.12这台机器也有samba服务，并且工作组的名称一样。如果不影响使用就忽略这个警告

closed 2024-01-26 by jjm2473

haobinnan, 2024-01-26:
> 这个警告应该是因为你的192.168.1.12这台机器也有samba服务，并且工作组的名称一样。如果不影响使用就忽略这个警告

我将咱们系统中的工作组改名没什么影响吧。因为我不用咱们系统共享数据。