istoreos/istoreos #1304: esxi安装istoreos后，网络选项下不显示”无线“菜单，ax200无线网卡

yixmlh90, 2024-01-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
不显示”无线“ 控制面板

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ *] 没有类似的issue
## 2. 详细叙述

在istore-网络选项下，不显示”无线“ 控制面板
### (1) 具体问题

不显示”无线“ 控制面板
A：

### (2) 路由器型号和固件版本
n100，
固件版本
iStoreOS 22.03.6 2024011910
内核版本
5.10.201
A：

### (3) 详细日志和/或截图
[image]

A：

yixmlh90, 2024-01-24:
[image]

已安装的软件包

yixmlh90, 2024-01-24:
**执行  lspci -nnk 结果**

root@iStoreOS:~# lspci -nnk
00:00.0 Host bridge [0600]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX Host bridge [8086:7190] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: agpgart-intel
lspci: Unable to load libkmod resources: error -12
00:01.0 PCI bridge [0604]: Intel Corporation 440BX/ZX/DX - 82443BX/ZX/DX AGP bridge [8086:7191] (rev 01)
00:07.0 ISA bridge [0601]: Intel Corporation 82371AB/EB/MB PIIX4 ISA [8086:7110] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.1 IDE interface [0101]: Intel Corporation 82371AB/EB/MB PIIX4 IDE [8086:7111] (rev 01)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
        Kernel driver in use: ata_piix
00:07.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 08)
        Subsystem: VMware Virtual Machine Chipset [15ad:1976]
00:07.7 System peripheral [0880]: VMware Virtual Machine Communication Interface [15ad:0740] (rev 10)
        Subsystem: VMware Virtual Machine Communication Interface [15ad:0740]
        Kernel driver in use: vmw_vmci
00:0f.0 VGA compatible controller [0300]: VMware SVGA II Adapter [15ad:0405]
        Subsystem: VMware SVGA II Adapter [15ad:0405]
00:11.0 PCI bridge [0604]: VMware PCI bridge [15ad:0790] (rev 02)
        Subsystem: VMware PCI bridge [15ad:0790]
00:15.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:15.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.0 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.1 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.2 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.3 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.4 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.5 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.6 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
00:16.7 PCI bridge [0604]: VMware PCI Express Root Port [15ad:07a0] (rev 01)
        Subsystem: VMware PCI Express Root Port [15ad:07a0]
        Kernel driver in use: pcieport
[... 51 lines cut ...]
        Kernel driver in use: uhci_hcd
02:01.0 Ethernet controller [0200]: Intel Corporation 82545EM Gigabit Ethernet Controller (Copper) [8086:100f] (rev 01)
        DeviceName: Ethernet0
        Subsystem: VMware PRO/1000 MT Single Port Adapter [15ad:0750]
        Kernel driver in use: e1000
02:02.0 USB controller [0c03]: VMware USB2 EHCI Controller [15ad:0770]
        DeviceName: ehci
        Subsystem: VMware USB2 EHCI Controller [15ad:0770]
        Kernel driver in use: ehci-pci
03:00.0 Network controller [0280]: Intel Corporation Wi-Fi 6 AX200 [8086:2723] (rev 1a)
        DeviceName: pciPassthru0
        Subsystem: Intel Corporation Device [8086:008c]
        Kernel driver in use: iwlwifi
0b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru1
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
13:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru2
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc
1b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
        DeviceName: pciPassthru3
        Subsystem: Intel Corporation Device [8086:0000]
        Kernel driver in use: igc

yixmlh90, 2024-01-24:
**执行  dmesg | grep iwlwifi 结果** ，请帮忙看下

root@iStoreOS:~# dmesg | grep iwlwifi
[    8.486231] iwlwifi 0000:03:00.0: enabling device (0000 -> 0002)
[    8.578464] iwlwifi 0000:03:00.0: api flags index 2 larger than supported by driver
[    8.580588] iwlwifi 0000:03:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    8.581267] iwlwifi 0000:03:00.0: loaded firmware version 66.f1c864e0.0 cc-a0-66.ucode op_mode iwlmvm
[   10.939122] iwlwifi 0000:03:00.0: Detected Intel(R) Wi-Fi 6 AX200 160MHz, REV=0x340
[   11.992403] iwlwifi 0000:03:00.0: SecBoot CPU1 Status: 0x5a29, CPU2 Status: 0x3
[   11.994243] iwlwifi 0000:03:00.0: UMAC PC: 0xc00815cc
[   11.994627] iwlwifi 0000:03:00.0: LMAC PC: 0x4f13de
[   11.994994] iwlwifi 0000:03:00.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[   12.242346] iwlwifi 0000:03:00.0: Start IWL Error Log Dump:
[   12.244504] iwlwifi 0000:03:00.0: Transport status: 0x00000042, valid: 6
[   12.244951] iwlwifi 0000:03:00.0: Loaded firmware version: 66.f1c864e0.0 cc-a0-66.ucode
[   12.245410] iwlwifi 0000:03:00.0: 0x00000084 | NMI_INTERRUPT_UNKNOWN       
[   12.245877] iwlwifi 0000:03:00.0: 0x002022F0 | trm_hw_status0
[   12.246344] iwlwifi 0000:03:00.0: 0x00000000 | trm_hw_status1
[   12.246821] iwlwifi 0000:03:00.0: 0x004FAA46 | branchlink2
[   12.247289] iwlwifi 0000:03:00.0: 0x004F13DE | interruptlink1
[   12.247756] iwlwifi 0000:03:00.0: 0x004F13DE | interruptlink2
[   12.248212] iwlwifi 0000:03:00.0: 0x004F976E | data1
[   12.248657] iwlwifi 0000:03:00.0: 0x01000000 | data2
[   12.249103] iwlwifi 0000:03:00.0: 0x00000000 | data3
[   12.249530] iwlwifi 0000:03:00.0: 0x00000000 | beacon time
[   12.249954] iwlwifi 0000:03:00.0: 0x000F837B | tsf low
[   12.250374] iwlwifi 0000:03:00.0: 0x00000000 | tsf hi
[   12.250795] iwlwifi 0000:03:00.0: 0x00000000 | time gp1
[   12.251182] iwlwifi 0000:03:00.0: 0x000FDE9C | time gp2
[   12.251543] iwlwifi 0000:03:00.0: 0x00000001 | uCode revision type
[   12.251900] iwlwifi 0000:03:00.0: 0x00000042 | uCode version major
[   12.252270] iwlwifi 0000:03:00.0: 0xF1C864E0 | uCode version minor
[   12.254051] iwlwifi 0000:03:00.0: 0x00000340 | hw version
[   12.254605] iwlwifi 0000:03:00.0: 0x18C89000 | board version
[   12.254947] iwlwifi 0000:03:00.0: 0x8004F502 | hcmd
[   12.255303] iwlwifi 0000:03:00.0: 0x00020000 | isr0
[   12.255630] iwlwifi 0000:03:00.0: 0x00000000 | isr1
[   12.255950] iwlwifi 0000:03:00.0: 0x00F00002 | isr2
[   12.256259] iwlwifi 0000:03:00.0: 0x00C0000C | isr3
[   12.256567] iwlwifi 0000:03:00.0: 0x00000000 | isr4
[   12.256863] iwlwifi 0000:03:00.0: 0x00000000 | last cmd Id
[   12.257159] iwlwifi 0000:03:00.0: 0x004F976E | wait_event
[   12.257442] iwlwifi 0000:03:00.0: 0x00000000 | l2p_control
[   12.257712] iwlwifi 0000:03:00.0: 0x00000000 | l2p_duration
[   12.257987] iwlwifi 0000:03:00.0: 0x00000000 | l2p_mhvalid
[   12.258282] iwlwifi 0000:03:00.0: 0x00000000 | l2p_addr_match
[   12.258579] iwlwifi 0000:03:00.0: 0x00000009 | lmpm_pmg_sel
[   12.258874] iwlwifi 0000:03:00.0: 0x00000000 | timestamp
[   12.259185] iwlwifi 0000:03:00.0: 0x00B80000 | flow_handler
[   12.259520] iwlwifi 0000:03:00.0: Start IWL Error Log Dump:
[   12.259813] iwlwifi 0000:03:00.0: Transport status: 0x00000042, valid: 7
[   12.260106] iwlwifi 0000:03:00.0: 0x20000066 | NMI_INTERRUPT_HOST
[   12.260384] iwlwifi 0000:03:00.0: 0x00000000 | umac branchlink1
[   12.260650] iwlwifi 0000:03:00.0: 0x80455A96 | umac branchlink2
[   12.260904] iwlwifi 0000:03:00.0: 0x80473F9A | umac interruptlink1
[   12.261148] iwlwifi 0000:03:00.0: 0xC0084A28 | umac interruptlink2
[   12.261382] iwlwifi 0000:03:00.0: 0x01000000 | umac data1
[   12.261609] iwlwifi 0000:03:00.0: 0xC0084A28 | umac data2
[   12.261837] iwlwifi 0000:03:00.0: 0x00000000 | umac data3
[   12.262063] iwlwifi 0000:03:00.0: 0x00000042 | umac major
[   12.262296] iwlwifi 0000:03:00.0: 0xF1C864E0 | umac minor
[   12.262515] iwlwifi 0000:03:00.0: 0x000FDE99 | frame pointer
[   12.264585] iwlwifi 0000:03:00.0: 0xC0887F18 | stack pointer
[   12.264851] iwlwifi 0000:03:00.0: 0x00000000 | last host cmd
[   12.265055] iwlwifi 0000:03:00.0: 0x00000004 | isr status reg
[   12.265272] iwlwifi 0000:03:00.0: IML/ROM dump:
[   12.265477] iwlwifi 0000:03:00.0: 0x00000003 | IML/ROM error/state
[   12.265710] iwlwifi 0000:03:00.0: 0x00005A29 | IML/ROM data1
[   12.265936] iwlwifi 0000:03:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[   12.266157] iwlwifi 0000:03:00.0: Fseq Registers:
[... 149 lines cut ...]
[   17.699628] iwlwifi 0000:03:00.0: 0x000FF32D | frame pointer
[   17.699826] iwlwifi 0000:03:00.0: 0xC0887F18 | stack pointer
[   17.700017] iwlwifi 0000:03:00.0: 0x00000000 | last host cmd
[   17.700208] iwlwifi 0000:03:00.0: 0x00000004 | isr status reg
[   17.700411] iwlwifi 0000:03:00.0: IML/ROM dump:
[   17.700600] iwlwifi 0000:03:00.0: 0x00000003 | IML/ROM error/state
[   17.700790] iwlwifi 0000:03:00.0: 0x00005A2B | IML/ROM data1
[   17.700983] iwlwifi 0000:03:00.0: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[   17.701173] iwlwifi 0000:03:00.0: Fseq Registers:
[   17.701361] iwlwifi 0000:03:00.0: 0x60000000 | FSEQ_ERROR_CODE
[   17.701566] iwlwifi 0000:03:00.0: 0x80290021 | FSEQ_TOP_INIT_VERSION
[   17.701764] iwlwifi 0000:03:00.0: 0x00050008 | FSEQ_CNVIO_INIT_VERSION
[   17.701963] iwlwifi 0000:03:00.0: 0x0000A503 | FSEQ_OTP_VERSION
[   17.702164] iwlwifi 0000:03:00.0: 0x80000003 | FSEQ_TOP_CONTENT_VERSION
[   17.702397] iwlwifi 0000:03:00.0: 0x4552414E | FSEQ_ALIVE_TOKEN
[... cut ...]
[   17.703310] iwlwifi 0000:03:00.0: 0x00000532 | CNVR_AUX_MISC_CHIP
[   17.703549] iwlwifi 0000:03:00.0: 0x05B0905B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[   17.703779] iwlwifi 0000:03:00.0: 0x0000025B | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[   17.704009] iwlwifi 0000:03:00.0: Failed to start RT ucode: -110
[   17.704240] iwlwifi 0000:03:00.0: WRT: Collecting data: ini trigger 13 fired (delay=0ms).
[   19.078127] iwlwifi 0000:03:00.0: Failed to run INIT ucode: -110
[   19.094698] iwlwifi 0000:03:00.0: retry init count 2

13535048320, 2024-08-03:
@jjm2473 我也遇到同样的问题了，请问解决了吗？

jjm2473, 2024-08-03:
没