istoreos/istoreos #1309: 建议首页的‘DNS配置’添加ipv6的dns设置

dcluffy, 2024-01-25:
目前首页的DNS配置只能改ipv4的dns，建议加上ipv6 dns配置，可以节省一些步骤

haobinnan, 2024-01-25:
还可以考虑添加ipv6模式：
[image]

jjm2473, 2024-01-26:
ipv4 dns基本都支持ipv6解析的，我还没见过不支持的，所以只要配置ipv4的dns就足够了。

ipv6模式的话，从istore安装系统便利工具（systools），里面就有开启ipv6的功能，会自动判断当前的联网模式选择正确的ipv6模式。

haobinnan, 2024-01-26:
> ipv4 dns基本都支持ipv6解析的，我还没见过不支持的，所以只要配置ipv4的dns就足够了。
> 
> ipv6模式的话，从istore安装系统便利工具（systools），里面就有开启ipv6的功能，会自动判断当前的联网模式选择正确的ipv6模式。

可以考虑默认将“系统便利工具（systools）”集成到固件中。

haobinnan, 2024-01-26:
> ipv4 dns基本都支持ipv6解析的，我还没见过不支持的，所以只要配置ipv4的dns就足够了。
> 
> ipv6模式的话，从istore安装系统便利工具（systools），里面就有开启ipv6的功能，会自动判断当前的联网模式选择正确的ipv6模式。

刚测试了 用 系统便利工具（systools） 打开ipv6 后 重启路由+电脑 本来有ipv6地址 操作后就彻底没有了

jjm2473, 2024-01-26:
@haobinnan 你确定之前能获取到2xxx开头的ipv6地址？

haobinnan, 2024-01-26:
> @haobinnan 你确定之前能获取到2xxx开头的ipv6地址？

是的 我每天都用ipv6 
uci del dhcp.@dnsmasq[-1].filter_aaaa
我之前只是将 filter_aaaa 删除  别的啥都没做。

haobinnan, 2024-01-26:
我刚刚将我的老配置恢复后就正常了。
[image]

jjm2473, 2024-01-26:
@haobinnan 网络-接口截图看看，还有执行开启ipv6的时候，输出的日志也发来看看

haobinnan, 2024-01-26:
[image]

haobinnan, 2024-01-26:
[image]
这里 没有 systools.sh 工具呢

jjm2473, 2024-01-26:
@haobinnan 直接网页上执行系统便利工具，也有日志

haobinnan, 2024-01-26:
[image]

haobinnan, 2024-01-26:
执行开启ipv6的截图 
[image]

[image]

jjm2473, 2024-01-26:
把 /etc/config/dhcp 提取出来看看，执行开启ipv6前后都行

haobinnan, 2024-01-26:
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option cachesize '1000'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        option filter_a '0'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

config host
        option mac '00:A0:C9:22:60:04'
        option ip '192.168.1.11'
        option name 'PC'
        option dns '1'

config host
        option mac '00:A0:C9:22:60:08'
        option ip '192.168.1.12'
        option name 'HomeServer'
        option dns '1'

config host
        option mac 'FA:A9:8F:C9:EC:3C'
        option ip '192.168.1.13'
        option name 'OrangePi'
        option dns '1'

config host
        option mac '52:54:00:B9:CD:82'
        option ip '192.168.1.14'
        option name 'KVM-PC'
        option dns '1'

jjm2473, 2024-01-26:
@haobinnan 奇怪，看起来跟其他人的pppoe一样啊，但是其他人反而要关闭lan的dhcpv6才能正确分配（通过odhcpd），你应该没卸载odhcpd或者禁用它吧

haobinnan, 2024-01-26:
没有关闭发自我的荣耀手机-------- 原始邮件 --------发件人： Liangbin Lian ***@***.***>日期： 2024年1月26日周五 12:54收件人： istoreos/istoreos ***@***.***>抄送： binnan_hao ***@***.***>, Mention ***@***.***>主    题： Re: [istoreos/istoreos] 建议首页的‘DNS配置’添加ipv6的dns设置 (Issue #1309)
@haobinnan 奇怪，看起来跟其他人的pppoe一样啊，但是其他人反而要关闭lan的dhcpv6才能正确分配（通过odhcpd），你应该没卸载odhcpd或者禁用它吧

haobinnan, 2024-01-26:
没卸载odhcpd或者禁用

haobinnan, 2024-01-26:
额外问题，今天发新固件吗？@jjm2473

jjm2473, 2024-01-26:
@haobinnan 发啊，不过没啥更新，只是小修小补

haobinnan, 2024-01-26:
> @haobinnan 发啊，不过没啥更新，只是小修小补

好的 这个ipv6 还打算处理吗 ？ 我目前 默认设置 就是删除 aaaa选项 就可以正常上网。

jjm2473, 2024-01-26:
@haobinnan 没法区分你的情况啊，你还是用以前的方式就行了。

haobinnan, 2024-01-26:
> @haobinnan 没法区分你的情况啊，你还是用以前的方式就行了。

好吧 我的情况就是有ipv6-pd 有wan口ipv6 地址（可用作nat转发）

jjm2473, 2024-01-26:
pppoe一般都有ipv6-pd，不过大部分人都是前缀长度60，56比较少见

haobinnan, 2024-01-26:
> pppoe一般都有ipv6-pd，不过大部分人都是前缀长度60，56比较少见

因为我这个56导致的？

dcluffy, 2024-01-26:
> ipv4 dns基本都支持ipv6解析的，我还没见过不支持的，所以只要配置ipv4的dns就足够了。
> 
> ipv6模式的话，从istore安装系统便利工具（systools），里面就有开启ipv6的功能，会自动判断当前的联网模式选择正确的ipv6模式。

因为ipv6优先，运营商的dns在某些地方解析又有问题，哪怕我改了ipv4的dns还是会用ipv6的dns来解析，访问依然有问题，所以还是要手动改下ipv6的dns

dcluffy, 2024-01-27:
> > ipv4 dns基本都支持ipv6解析的，我还没见过不支持的，所以只要配置ipv4的dns就足够了。
> > ipv6模式的话，从istore安装系统便利工具（systools），里面就有开启ipv6的功能，会自动判断当前的联网模式选择正确的ipv6模式。
> 
> 因为ipv6优先，运营商的dns在某些地方解析又有问题，哪怕我改了ipv4的dns还是会用ipv6的dns来解析，访问依然有问题，所以还是要手动改下ipv6的dns

我又重置重新配置了一遍，发现只改ipv4的就可以了，之前测试可能哪里没设置好

closed 2024-01-27 by dcluffy