istoreos/istoreos #1313: 关于使用系统便利工具导致的概览页面显示问题

dcluffy, 2024-01-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用系统便利工具的关闭ipv6，在使用打开ipv6，发现概览页面ipv6上游不显示dns，已分配的DHCPv6租约也没有内容（之前都是可以正常显示的）

### (2) 路由器型号和固件版本

A：x86
iStoreOS 22.03.6 2024012613

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-02-01:
如果你原本的ipv6没问题，就不要使用工具打开。你现在可以把lan口的ipv6服务打开试试。

dcluffy, 2024-02-01:
> 如果你原本的ipv6没问题，就不要使用工具打开。你现在可以把lan口的ipv6服务打开试试。

这个问题倒无所谓，毕竟不影响使用。
我把istoreos重置后，只设置pppoe拨号，其他的都不动，仍然有非常搞人心态的问题存在，总体并不是一个可以稳定运行的状态，可能是我设备配置不够吧，所以暂时放弃istoreos了

closed 2024-02-01 by dcluffy