istoreos/istoreos #1319: nat环回不生效

18515381698, 2024-01-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：nat环回已开启，不生效，内网通过域名无法请求内网服务器

### (2) 路由器型号和固件版本

A：[Powered by LuCI openwrt-22.03 branch (git-23.292.78392-9f66674)](https://github.com/openwrt/luci) / [ArgonTheme v2.2.10.9](https://github.com/jerrykuku/luci-theme-argon) / iStoreOS 22.03.6 2024012613

### (3) 详细日志和/或截图

A：
[image]

uibehw, 2024-01-31:
我也遇到这个问题了，刷的23.06-240126最新版X86_64-EFI固件，端口转发一直不成功，折腾了一晚上，搜遍了网络上的各种办法，映射端口就是打不开。

jjm2473, 2024-02-23:
环回源IP改成外部试试？

18515381698, 2024-02-23:
> Contributor

试了，依旧无法内网通过域名访问

jjm2473, 2024-02-23:
你这个内网服务器直接用ip和端口可以访问吗

18515381698, 2024-02-25:
> 你这个内网服务器直接用ip和端口可以访问吗

是可以的。内网通过ip可正常访问，外网通过域名也可正常访问。唯独内网通过域名无法正常访问。nat环回已开启，功能似乎未生效。网上查了很多openwrt的系统的nat环回不生效的例子。有说修改/etc/sysctl.conf的，但也不生效

18515381698, 2024-02-25:
似乎因为我端口转发里没添加一条源区域是lan的记录？我添加上后lan通过域名访问正常了

closed 2024-02-27 by 18515381698

jjm2473, 2024-02-27:
理论上NAT环回就是用来做这件事的

PunkFleet, 2025-02-08:
> 似乎因为我端口转发里没添加一条源区域是lan的记录？我添加上后lan通过域名访问正常了

请问是如何设置端口转发的，我这样对吗：

[image]

King-hsw, 2025-07-04:
> 似乎因为我端口转发里没添加一条源区域是lan的记录？我添加上后lan通过域名访问正常了

请问可以参考一下怎么设置么？我也是这种问题但是不知道怎么配置你说的这个。不过我域名绑定的是ipv6的不知道行不行

jjm2473, 2025-07-15:
ipv6的NAT环回暂不支持，openwrt就是如此。可以使用socat转发，这样就不需要NAT和环回了。