istoreos/istoreos #1322: jellyfin转码缓存 放在内存中

pili-book, 2024-02-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：想把缓存放在内存中，但是设置了tmp的文件路径，安装失败了，不知道如何办，请求高手

[image]

### (2) 路由器型号和固件版本

A：R6s

### (3) 详细日志和/或截图

A：

bcseputetto, 2024-02-01:
想要挂载`tmpfs`，请用以下办法
示例一
```console
--mount type=tmpfs,destination=/transcodes
```
示例二
```console
--tmpfs /transcodes
```

pili-book, 2024-02-01:
tmp文件夹本身就是那个挂载出来的哦，我把转吗文件夹设置在那里就不行，创建失败

 

 
 --tmpfs /transcodes

bcseputetto, 2024-02-01:
请你按照我说的办法挂载`tmpfs`到容器，然后去媒体库后台修改转码临时路径即可。

pili-book, 2024-02-01:
> 请你按照我说的办法挂载`tmpfs`到容器，然后去媒体库后台修改转码临时路径即可。
[image]

我先把文件位置放在docker文件夹下了，但是创建失败了，现在是无法创建

pili-book, 2024-02-01:
你这个是不是创建容器的时候得手动创建了，我是用的istore商店的那个创建的

pili-book, 2024-02-01:
好尴尬，发现正常生成都不能用，我先看看解决能用问题吧，你这个是需要手动创建吧？不能使用istore的那个生成吧

pili-book, 2024-02-02:
> 你这个是不是创建容器的时候得手动创建了，我是用的istore商店的那个创建的
> […](#)

我按照你的方法手动创建了容器，但是在挂载点那里并没有看到这个挂载，不知道到底挂上了没有

jjm2473, 2024-02-02:
确定不是docker数据目录出问题了吗，看报错不是mount，是mkdir，看看docker数据目录能不能写入，有没有满

pili-book, 2024-02-02:
没有满的还有12g呢

jjm2473, 2024-02-02:
那能不能写入呢

pili-book, 2024-02-02:
> 那能不能写入呢

docker run --name jellyfin \
    --privileged \
    --restart=unless-stopped -td \
    --device /dev/rga\
    --device /dev/dri\
    --device /dev/dma_heap\
    --device /dev/mpp_service\
    --dns=172.17.0.1 \
    -e TZ=Asia/Shanghai \
    --mount type=tmpfs,destination=/config/transcodes\
    -v "/mnt:/media" \
   -v /mnt:/mnt:rslave \
    jjm2473/jellyfin-mpp:latest
最后手动创建的，没用那个自动创建的

pili-book, 2024-02-02:
手动建立了，但是感觉扫描速度很慢哦，cpu基本上百分之5左右。。。不知道为啥这么慢

jjm2473, 2024-02-02:
/config/transcodes是jellyfin的默认配置，这个没啥问题，但是  @pili-book 都没配置config，也就是说jellyfin配置和数据不会保留的，重新部署jellyfin就会丢失数据。

@pili-book 你的/mnt/huhu/jellyfinOpenwrt是不是只读的？能写入文件吗？你把一个只读的文件夹作为config目录，有什么用，是不打算启动jellyfin吗？

jjm2473, 2024-02-02:
这个问题的根本原因不是tmpfs，也不是transcodes，也不是缓存目录，而是配置文件夹不可写。

closed 2024-02-02 by jjm2473

pili-book, 2024-02-02:
> 这个问题的根本原因不是tmpfs，也不是transcodes，也不是缓存目录，而是配置文件夹不可写。

docker run --restart=unless-stopped -d -v "/mnt/huhu/jellyfinOpenwrt:/config"     -t     --privileged  --device /dev/rga --device /dev/dri --device /dev/dma_heap --device /dev/mpp_service    --dns=172.17.0.1       -e TZ=Asia/Shanghai --mount type=tmpfs,destination=/mnt/transcodes -v "/mnt/huhu/jellyfinOpenwrt/tmp:/config/transcodes" -v "/mnt:/media" -v /mnt:/mnt:rslave --name jellyfin "jjm2473/jellyfin-mpp:latest"
按照这个写的，又不能用了，加了一句--mount type=tmpfs,destination=/mnt/transcodes

jjm2473, 2024-02-02:
-v "/mnt/huhu/jellyfinOpenwrt/tmp:/config/transcodes" 又是什么？

jjm2473, 2024-02-02:
都说了是配置文件夹有问题，就不会换一个路径？

pili-book, 2024-02-02:
这个不是把内存挂载到mnt的那个文件夹吗

 
为什么用/mnt/transcodes，你知道这些参数干嘛的吗

pili-book, 2024-02-02:
> 为什么用/mnt/transcodes，你知道这些参数干嘛的吗

https://github.com/istoreos/istoreos/issues/1325
这是重新弄得代码但是不对，帮忙看看，那个配置文件夹我现在是可读写的

pili-book, 2024-02-02:
> 我个人建议你不用非得把 transcodes 放在 /config 后面 你/config 文件夹你不映射的？那你每次遇到问题重建容器，配置都是丢失的，每次你都重新去配？
https://github.com/istoreos/istoreos/issues/1325
这是重新弄得代码但是不对，帮忙看看，那个配置文件夹我现在是可读写的

jjm2473, 2024-02-02:
用jellyfin插件部署啊，不是很简单吗，谁让你不懂还乱写命令。配置文件路径选择硬盘的一个文件夹（例如`/mnt/sata1-1/Configs/Jellyfin`），不要使用samba挂载的文件夹，媒体路径不需要填，转码缓存路径可以填 `/tmp/jellyfin/cache` 或者也使用硬盘的文件夹，或者不填都行。

pili-book, 2024-02-02:
> 有插件做好的功能不用，偏要自己写命令，还tm乱写。

插件自己弄得那个不能用啊，我也不想乱写。。但是配置文件夹没办法啊r6s存储小，要使用这个软件必须挂载给samba 啊

pili-book, 2024-02-02:
> > 有插件做好的功能不用，偏要自己写命令，还tm乱写。
> 
> 插件自己弄得那个不能用啊，我也不想乱写。。但是配置文件夹没办法啊r6s存储小，要使用这个软件必须挂载给samba 啊

r6s是板载的32g内存。。很小的

jjm2473, 2024-02-02:
谁说配置文件32g不够的？又不是视频文件

pili-book, 2024-02-02:
> 谁说配置文件32g不够的？又不是视频文件

转码文件应该是很耗硬盘的，32g也耐不住啊，我才想把那个转码缓存文件的放到内存中

bcseputetto, 2024-02-02:
> > 谁说配置文件32g不够的？又不是视频文件
> 
> 转码文件应该是很耗硬盘的，32g也耐不住啊，我才想把那个转码缓存文件的放到内存中

所以我不是跟你讲了，你就正常宿主机硬盘的挂载路径，映射容器 /config，tmpfs挂载一个文件夹到容器根目录，然后去jellyfin控制台修改转码文件临时路径就完了，/config 文件不要用SMB的路径去映射

bcseputetto, 2024-02-02:
假设你自带硬盘是挂载到 /mnt/data_sda4
```console
sudo docker run -d \
    --name jellyfin \
    --network=host \
    -e UID=0 \
    -e GID=0 \
    -e GIDLIST=100 \
    -e TZ="Asia/Shanghai" \
    --device /dev/dri:/dev/dri \
    --mount type=tmpfs,destination=/transcodes \
    -v /mnt/data_sda4/jellyfin/config:/config \
    -v /mnt/hulu:/media \
    --restart=unless-stopped \
    jjm2473/jellyfin-mpp:latest
```
这样 tmpfs 会挂载到容器根目录一个叫 transcodes的文件夹，就不要去用 config下面的那个了。
这样不就行了