istoreos/istoreos #1323: jellyfin 安装后无法使用

pili-book, 2024-02-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
[image]

[image]
显示挂载成功，但是网页打开是空白
### (2) 路由器型号和固件版本
FriendlyElec NanoPi R6S
A：
iStoreOS 22.03.5 2023121510 / LuCI istoreos-22.03 branch git-23.342.39264-37c783a
### (3) 详细日志和/或截图

A：

jjm2473, 2024-02-02:
你确定30秒已经启动完了？

pili-book, 2024-02-02:
> 你确定30秒已经启动完了？

docker run --name jellyfin \
    --privileged \
    --restart=unless-stopped -td \
    --device /dev/rga\
    --device /dev/dri\
    --device /dev/dma_heap\
    --device /dev/mpp_service\
    --dns=172.17.0.1 \
    -e TZ=Asia/Shanghai \
    --mount type=tmpfs,destination=/config/transcodes\
    -v "/mnt:/media" \
   -v /mnt:/mnt:rslave \
    jjm2473/jellyfin-mpp:latest
最后手动创建的。。。。。
但是发现个问题扫库特别慢，cpu都占不了多少，我的媒体库是挂载的nas的都是2.5g网口哦，r6s八核的cpule

jjm2473, 2024-02-02:
我靠，你把jellyfin的配置文件夹设置成samba挂载的文件夹了？而且根据你的另一个issue，你这个文件夹可能还是只读的，能启动才怪了。

jjm2473, 2024-02-02:
而且你自己部署的美丽甚至没有映射config目录，你是不打算保存jellyfin的数据是吧。

closed 2024-02-02 by jjm2473

pili-book, 2024-02-02:
> 我靠，你把jellyfin的配置文件夹设置成samba挂载的文件夹了？而且根据你的另一个issue，你这个文件夹可能还是只读的，能启动才怪了。

这个只读已经改了
https://github.com/istoreos/istoreos/issues/1325 帮忙看看这个问题
实话说，原来不知道怎么的还能用。。删了照你说的改了改不能用了 
[image]