istoreos/istoreos #1325: R6S安装jellyfin ，格式错误，请高手帮忙看看哪里不对

pili-book, 2024-02-02:
[image]
docker run --name jellyfin \
--privileged \
-v /mnt/huhu/jellyfinOpenwrt:/config\
--restart=unless-stopped -td \
--device /dev/rga\
--device /dev/dri\
--device /dev/dma_heap\
--device /dev/mpp_service\
--dns=172.17.0.1 \
-eTZ=Asia/Shanghai \
--mount type=tmpfs,destination=/mnt/transcodes\
-v /mnt/huhu/jellyfinOpenwrt/tmp:/config/transcodes\
-v /mnt:/media \
-v /mnt:/mnt:rslave \
jjm2473/jellyfin-mpp:latest

jjm2473, 2024-02-02:
你能不能别发这么多相同的issue，你以为这是qq群吗

closed 2024-02-02 by jjm2473

pili-book, 2024-02-02:
> 你能不能别发这么多相同的issue，你以为这是qq群吗

这是不同的问题吧。。。。

pili-book, 2024-02-02:
我这个代码到底有啥问题哦

pili-book, 2024-02-02:
插件自动生成的那个不能用，看着成功加载了打开网页空白的。。。。

pili-book, 2024-02-02:
> 你能不能别发这么多相同的issue，你以为这是qq群吗

我想加一句这个--mount type=tmpfs,destination=/mnt/tmp，而且商店插件自己生成的不能用我也不知道为啥，虽然我是把那几个选项选到samb文件夹了，也不应该不能用啊，他还 加载成功。。就是不能用，插件的那个还有个问题就是端口隐射，我住路由不需要端口映射，映射了反而打开端口搞得外端口还占一个，我lucky大吉害的重新选个端口

pili-book, 2024-02-02:
> 还不是同一个问题？都是你要部署jellyfin，却总是搞错，每错一步就发一个issue。本来这种插件的使用问题都不应该发issue的，又不是bug，并且跟固件无关。

同志你帮我看看代码啊，到底哪有问题

pili-book, 2024-02-02:
> 还不是同一个问题？都是你要部署jellyfin，却总是搞错，每错一步就发一个issue。本来这种插件的使用问题都不应该发issue的，又不是bug，并且跟固件无关。

我也没办法啊，走一步发现一个问题，刚开始插件自动生成发现无法使用，后来，代码生成能用了，结果人家说我弄得不对，然后各种改都不鞥成功挂载了，，，，这会又改了半天，还是弹出问题，一个新问题。。。。

pili-book, 2024-02-02:
> 还不是同一个问题？都是你要部署jellyfin，却总是搞错，每错一步就发一个issue。本来这种插件的使用问题都不应该发issue的，又不是bug，并且跟固件无关。

确实config不能挂载到外面，挂载到docker的一个文件夹以后好了，还有个问题想问下，我威联通nas扫描库的时候CPU满载，r6s扫描的时候cpu最多到百分之30左右，不知道docker限制cpu使用没有