istoreos/istoreos #1327: iStore里面的Cups打印问题

lizhi10, 2024-02-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：istore 里面的 cups 功能只支持smb共享，无法支持ipp共享打印

### (2) 路由器型号和固件版本

A：畅网N5105，H28k，m920x with i350t4, r5s

### (3) 详细日志和/或截图

A：使用上述型号的设备和兄弟 T725dw, 兄弟 2550dw， 惠普 227dw，无论有没有pdd文件，都无法使用ipp共享打印，Windows 显示无法添加，macos显示不能与打印机通信。换了olbat/cupsd:latest 的 docker 版，OpenPrinting CUPS 2.4.7, 至少x86设备的配合这三款打印机都可以打印了。所以可能是商店里的 cups 有问题。

gzsdbb, 2024-04-23:
其实不是应用中心的cups问题  应用中心只是安装了这个插件
而问题出在驱动上,openwrt上打印机上的驱动没有安装,其实我也知道怎么安装 ipk也找不到
而pdd他并不是驱动 他只是描述问题,网上倒是有万能驱动gutenprint 可是openwrt就知道怎么去安装这个了

lizhi10, 2024-04-25:
> 其实不是应用中心的cups问题 应用中心只是安装了这个插件 而问题出在驱动上,openwrt上打印机上的驱动没有安装,其实我也知道怎么安装 ipk也找不到 而pdd他并不是驱动 他只是描述问题,网上倒是有万能驱动gutenprint 可是openwrt就知道怎么去安装这个了

你的理解是不对的。cups 是一个协议，大部分打印机都支持。pdd 文件是打印机厂商提供的postcript描述，适应不同打印机的postscript实现和一些特殊功能。ipk 是 openwrt 软件包. linux 基本上都是用cups来打印的，不用单独安装驱动。应用中心里的cups可能做过修改不支持ipp方式共享，至少我是不行的。但是 docker 版的 cups 却可以支持 你要是有类似问题可以试试。

Zhuwenxue2002, 2025-03-30:
是的，我也出现这种情况了，但我通过应用市场方式安装cups可以通过windows添加连接的方式连接打印机，这种方式好像就是IPP协议。但苹果仍无法使用Air隔空打印服务。我尝试使用docker来安装cups 经历了很多意想不到的错误，遂放弃;

Zhuwenxue2002, 2025-03-30:
在linux系统上可以通过安装apt-get -y install avahi-daemon avahi-discover libnss-mdns的方式实现苹果的隔空打印服务，但我不清楚如何在istoreos上面安装

lizhi10, 2025-03-30:
> 在linux系统上可以通过安装apt-get -y install avahi-daemon avahi-discover libnss-mdns的方式实现苹果的隔空打印服务，但我不清楚如何在istoreos上面安装

用 Windows 添加连接打印机是 smb 共享, 不是 ipp, 得手动选择 ipp 才是 ipp. 我用的 docker cups 镜像貌似也有点老, 有些打印机不支持了.

Zhuwenxue2002, 2025-03-30:
> > 在linux系统上可以通过安装apt-get -y install avahi-daemon avahi-discover libnss-mdns的方式实现苹果的隔空打印服务，但我不清楚如何在istoreos上面安装
> 
> 用 Windows 添加连接打印机是 smb 共享, 不是 ipp, 得手动选择 ipp 才是 ipp. 我用的 docker cups 镜像貌似也有点老, 有些打印机不支持了.

直接使用http://192.168.2.2:631/printers/Canon_iP100_series这种连接添加的打印机时smb共享的形式吗？ 我以为这种方式就是ipp添加了

lizhi10, 2025-03-30:
> 直接使用[http://192.168.2.2:631/printers/Canon_iP100_series这种连接添加的打印机时smb共享的形式吗？](http://192.168.2.2:631/printers/Canon_iP100_series%E8%BF%99%E7%A7%8D%E8%BF%9E%E6%8E%A5%E6%B7%BB%E5%8A%A0%E7%9A%84%E6%89%93 [...]

不是, 要选择添加 ipp 打印机才是 ipp

Zhuwenxue2002, 2025-03-30:
> 不是, 要选择添加 ipp 打印机才是 ipp

好吧，那其实使用cups搭建之后仍然用smp协议来使用了，毕竟cups支持smp，那么我的打印机在istoreos的适用范围极小了，基本就支持windows的笔记本和电脑作为网络打印的发起者了

Zhuwenxue2002, 2025-03-31:
我解决了，我换了一个版本很低但是拉取量最高的那个docker镜像，经过测试之后发现，能否使用ipp打印和cups（或者说是容器是否存在该驱动）有很大关系，我的打印机是ip100 cups docker镜像中不包含此驱动程序，我用了原版pdd，但没有原版驱动，是无法使用ipp协议的。后面我拉了很多的依赖，费很大劲把我这个老旧打印机驱动装好了，已经可以正常使用了，并且可以使用airprint来打印了。但还有一个问题，windows扫描到打印机之后无法连接，经查明说是证书存在相关问题。但已经不想折腾了。