istoreos/istoreos #1331: 【BUG】断电重启 很容易openwrt时间回到2017年，导致docker彻底无法使用, 突然倒退几年..任何证书都过期

gowy222, 2024-02-04:
固件版本2024-2月最新的

用docker跑singbox，路由吗..断电硬重启是家常便饭...
'毕竟,重启大法=解决任何bug 233'
结果造成bug了...

重启只会 固件时间退回2017年..(要么固件内置了一个发行日期？还是板子?)

certificate has expired or is not yet valid: current time 2017-08-04T09:03:44Z is before 2023-04-12T00:00:00Z
很多底层依赖ssl的就会爆出证书过期而不能启动，docker尤其多..

后来手动在UI界面和浏览器同步时间，重启docker就恢复正常...
------
所以，固件开机, 需要加固一下 网络通了，先强制同步时间!! 比任何任务服务优先...

haobinnan, 2024-02-05:
我用了这么多年openwrt 时间都是准确的，路由启动后 几秒后 你查看时间也不对吗？

jjm2473, 2024-02-09:
正常情况下联网以后都会同步时间，估计网络还没连接的时候，你的容器已经启动，之后时间同步以后你的容器也没重试机制。

还有，为什么要断电重启，直接软重启不行吗？软重启就肯定不会导致时间重置。

closed 2024-02-09 by jjm2473

gowy222, 2024-02-09:
@jjm2473 
就是专门拔电重启做高可用稳定性测试.(毕竟你人不在家，其他人自会断电重启)....
问题就在于, 开机 连不上ntp服务器...一直卡在那边 只能和浏览器同步时间

gowy222, 2024-02-09:
不是容器启动不了，是一启动就失败(容器)，在无限自动重启...
https://stackoverflow.com/questions/35289802/docker-pull-error-x509-certificate-has-expired-or-is-not-yet-valid

lvkai, 2024-09-07:
每次家里跳闸或停电，都要手动同步浏览器时间。可能是软路由启动比光猫联网快造成的。