istoreos/istoreos #1333: 挂载 CIFS/SMB  重启后就消失一个bug

pili-book, 2024-02-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：挂载 CIFS/SMB 
[image]
[image]
发现个问题，重启后本来挂载的两个只剩下了一个自动挂载上了。。。。我觉得绝对算bug

### (2) 路由器型号和固件版本

A：FriendlyElec NanoPi R6S
iStoreOS 22.03.6 2024012613 / LuCI istoreos-22.03 branch git-24.016.46072-3e9fab7

### (3) 详细日志和/或截图

A：

jjm2473, 2024-02-09:
可能是因为启动过程中网络还没连接好就执行到挂载了

pili-book, 2024-02-09:
那这。。。。。怎解决，固件本身这样子应该是还是bug吧，我自己解决的话应该去哪改改东西

pili-book, 2024-02-11:
这个怎么能调节一下，延迟一下了

pili-book, 2024-02-12:
那我想把他延时一下得去哪改东西了

jjm2473, 2024-02-18:
之后会增加一个延迟的参数配置