istoreos/istoreos #1334: 使用网络向导功能后，系统盘疑似掉盘

ssmengyan, 2024-02-06:
### (1) 具体问题

在安装到本地的nvme M2 500g固态硬盘后，可以正常引导运行系统，但是点了网络向导，配置完之后就会后台直接一大堆报错，网页访问地址也打不开。但是在PE系统中是可以正常查看到该磁盘的。尝试安装其他的openclash系统运行，安装一些docker后也能正常运行。

在把系统写入硬盘后就直接启动了，进入网页管理界面后没有做任何其他动作，直接使用网络向导功能，一键配置旁路由模式。配置完后就直接出错，出现这种疑似掉盘的情况，之后再强行拔电重启后还是打不开系统，也是一直提示类似的报错。

### (2) 路由器型号和固件版本

N5095主板 、16G内存、500G全新DIY nvme m2固态、4T机械硬盘。
固件版本：istoreos-22.03.6-2024012613-x86-64-squashfs-combined-efi.img

### (3) 详细日志和/或截图

后台报错时的日志：
[image]
出错时的网页管理界面打开就是这样的：
[image]
硬盘的检测信息：
[image]
[image]

ssmengyan, 2024-02-07:
应该是底层Linux内核不支持我的这个m2硬盘，安装了unraid、PVE、unbantu都会出现类似的情况。但是装windows后使用无任何异常掉盘或数据错误问题。

jjm2473, 2024-02-09:
你是用usb转nvme？

ssmengyan, 2024-02-10:
> 你是用usb转nvme？

使用的原生nvme，应该是那块硬盘的问题，我又买了块新的m2盘，已经流畅稳定运行istoreOS两天，应该没问题了。

closed 2024-02-10 by ssmengyan