istoreos/istoreos #1335: adguardhome安装后无法正常使用

carsonrz, 2024-02-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：无论是从软件包安装的adguradhome107.21，还是先安装luci-app-adguardhome_1.8-9_all.IPK 界面程序再更新内核到107.44，都无法正常使用。安装后可以打开adguradhome的web界面，但配置完后，只能保存配置，实际adguradhome无法重定向。而且无法使用 luci-app-adguardhome_1.8-9_all界面中的“手动设置”，打开报500错误。

### (2) 路由器型号和固件版本

A：国伟R86S-U2

### (3) 详细日志和/或截图

A：
[image]

DanFreitas94, 2024-02-11:
我看到您已經遇到問題超過兩天了，請使用本教程（[](https://forum.openwrt.org/t/how-to-updated-2021-installing-adguardhome-on-openwrt-manual-and - opkg-method /113904) ）它安裝了更新的版本，對我來說它總是有效。 （請原諒，使用谷歌翻譯從 PT-BR 到 CH）

bcseputetto, 2024-02-12:
@carsonrz 请你按正常流程配置好`AGH(AdGuardHome)`后，不要使用`luci`的重定向功能，如果你是想`作为dnsmasq的上游服务器`，可以到【网络】-【DHCP/DNS】-【DNS转发】处，手动设置`dnsmasq`的上游服务器为`AGH`，假设你的`AGH`的DNS监听端口为`5335`，请在【DNS转发】处设置`127.0.0.1#5335`即可

jinxiao, 2024-02-17:
这个版本的luci-app-adguardhome已经好几年不更新了，不好使。可以网上搜索sirpdboy啥的，那个人在更新这个软件

bcseputetto, 2024-02-18:
提供了 sirpdboy 源码编译的 `luci-app-adguardhome` 欢迎测试
https://github.com/bcseputetto/Are-u-ok/blob/master/packages/README.md

jjm2473, 2024-02-18:
看起来是因为resolv.conf.auto的路径变了，但是插件却还在使用老的路径。试试执行这个命令 `ln -s /tmp/resolv.conf.d/resolv.conf.auto /tmp/resolv.conf.auto` ，“手动设置”页面应该就没问题了。

如果你是从软件包里面安装的插件，那么还可以卸载重装试试。

jjm2473, 2024-02-18:
哦，我看luci-app-adguardhome并不是openwrt提供的，那就只能找个新版本来安装了

carsonrz, 2024-02-18:
> @carsonrz 请你按正常流程配置好`AGH(AdGuardHome)`后，不要使用`luci`的重定向功能，如果你是想`作为dnsmasq的上游服务器`，可以到【网络】-【DHCP/DNS】-【DNS转发】处，手动设置`dnsmasq`的上游服务器为`AGH`，假设你的`AGH`的DNS监听端口为`5335`，请在【DNS转发】处设置`127.0.0.1#5335`即可
> 
> 如果问题已解决请关闭issue

试过这方法了，不行，不知道为什么，原来也不像安装这luci-app-adguardhome界面的，但发现从istoreOS的【软件包】安装adguardhome后，配置完，也无法生效。我目前情况是用一个VM虚拟机装istoreOS，然后再测试AGH的功能，设备是两个网卡，一个有线，一个无线。不知道有没影响，之前试过偶尔能成功过滤。我刚开始接触软路由，所以很多细节都不太熟悉。请看截图。
[image]
[image]
[image]

bcseputetto, 2024-02-18:
@carsonrz 这样配置之后，终端设备向路由器`192.168.5.2:53`查询，路由器向AGH`127.0.0.1:533`查询，AGH的Web后台里没有查询记录吗？理论上应该有查询记录才对，有查询记录就说明DNS流程正确

carsonrz, 2024-02-19:
> sirpdboy

用这个解决了问题，之前不太敢用非官方的东西，怕有后门。不过sirpdboy应该信得过 ，目前设置【重定向53端口到AdGuardHome】经常不生效，所以现在设置【使用53端口替换dnsmasq】使用正常。途中参考了以下文章作为辅助https://post.smzdm.com/p/avx9xzx4/   ，非常感谢istoreos团队的付出，同事感谢各位githup同仁的回复。Guys！Thanks for helping~ : -)

closed 2024-02-19 by carsonrz