istoreos/istoreos #1336: 动态DDNS  cloudflare.com-v4   dnsexit都解析不了

zhb7670, 2024-02-11:
### (1) 具体问题

动态DDNS  cloudflare.com-v4   dnsexit. 都解析不了

### (2) 路由器型号和固件版本

 ESXI N5105

### (3) 详细日志和/或截图

cloudflare.com-v4报错：
 020652  WARN : NO valid IP found
 020652  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
 020652  info : Starting main loop at 2024-02-12 02:06
 020652       : Detect local IP on 'web'
 020652       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/hongbo.dat -o /var/run/ddns/hongbo.err --no-proxy 'http://checkipv6.dyndns.com'
 020652 ERROR : GNU Wget Error: '4'
 020652       : wget: unable to resolve host address 'checkipv6.dyndns.com'
 020652  WARN : Transfer failed - retry 1/0 in 60 seconds
 020752       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/hongbo.dat -o /var/run/ddns/hongbo.err --no-proxy 'http://checkipv6.dyndns.com'
 020752 ERROR : GNU Wget Error: '4'
 020752       : wget: unable to resolve host address 'checkipv6.dyndns.com'
 020752  WARN : Transfer failed - retry 2/0 in 60 seconds

dnsexit报错：
020652  WARN : NO valid IP found
 020652  WARN : Error in 'expand_ipv6()' - invalid IPv6 found: '' expanded: ''
 020652  info : Starting main loop at 2024-02-12 02:06
 020652       : Detect local IP on 'web'
 020652       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/hongbo.dat -o /var/run/ddns/hongbo.err --no-proxy 'http://checkipv6.dyndns.com'
 020652 ERROR : GNU Wget Error: '4'
 020652       : wget: unable to resolve host address 'checkipv6.dyndns.com'
 020652  WARN : Transfer failed - retry 1/0 in 60 seconds

jjm2473, 2024-02-18:
你确定用这个 `http://checkipv6.dyndns.com/` 能获取到公网地址？

closed 2024-02-18 by jjm2473

zhb7670, 2024-02-22:
> 你确定用这个 `http://checkipv6.dyndns.com/` 能获取到公网地址？

确定

zhb7670, 2024-02-22:
[image]

jjm2473, 2024-02-22:
那你在软路由上执行`curl http://checkipv6.dyndns.com/`看看

jjm2473, 2024-02-22:
> `wget: unable to resolve host address 'checkipv6.dyndns.com'`

看你的日志中有这个，你确定软路由的DNS正常且有ipv6吗？执行 `nslookup checkipv6.dyndns.com` 看看

zhb7670, 2024-02-22:
> curl http://checkipv6.dyndns.com/

[image]

zhb7670, 2024-02-22:
> > `wget: unable to resolve host address 'checkipv6.dyndns.com'`
> 
> 看你的日志中有这个，你确定软路由的DNS正常且有ipv6吗？执行 `nslookup checkipv6.dyndns.com` 看看

[image]

zhb7670, 2024-02-22:
我装了好多op系统，用DDNS都没有问题，只有iStoreOS的DDNS，解析不了

zhb7670, 2024-02-22:
ipv4  ipv6都不能解析出来

zhb7670, 2024-02-22:
root@iStoreOS:~# curl http://checkipv6.dyndns.com/
curl: (6) Could not resolve host: checkipv6.dyndns.com
错了，iStoreOS并不能获取

zhb7670, 2024-02-22:
[image]

zhb7670, 2024-02-22:
[image]

zhb7670, 2024-02-22:
[image]

zhb7670, 2024-02-22:
[image]
把这个钩去掉就好了，默认是钩选的