istoreos/istoreos #1338: istoreos_x86_64_intel能否直通pci

zhuozi1212, 2024-02-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
为避免重复issue，请先搜索issue，确认没有类似issue再提交新issue；
注意搜索时**包括已关闭**的issue（删掉搜索框的的“is:open”条件）；

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：查询一下是否能开启iommu通过vfio达到pci直通设备
主板是w680+12500，支持并开启iommu，sr-iov，vt-d，vt-x
图内的通过修改grub的方式开始是不是为方式不对，或者有其他办法也能pci直通到webvirtcloud

### (2) 路由器型号和固件版本

A：istoreos x86_64_efi

### (3) 详细日志和/或截图

A：
[image]

jjm2473, 2024-02-18:
可以修改/boot/grub里面的配置文件，但是这个文件每次升级固件都会被重置。

jjm2473, 2024-02-18:
看你的cmdline不是已经生效了吗

zhuozi1212, 2024-02-18:
> 看你的cmdline不是已经生效了吗

libvirt 错误- unsupported configuration: host doesn't support VFIO PCI passthrough
[image]

jjm2473, 2024-02-18:
那你就按其他教程，修改/boot/grub

zhuozi1212, 2024-02-18:
> 那你就按其他教程，修改/boot/grub

因为没有关于openwrt的方法，因此我只是参考centos的贴文

ks-bao, 2024-02-18:
因为时间不够，所以我没空把显卡直通开发完成。你有兴趣，我提供大概的思路：

### 一 grub 要修改，比如

INTEL 后面要添加：intel_iommu=on iommu=pt video=efifb:off,vesafb:off

或者 AMD后面要添加：amd_iommu=on iommu=pt video=vesafb:off,efifb:off,simplefb:off

### 删除原本的显卡驱动，比如

jjm2473, 2024-02-18:
我看了下，目前22.03固件的内核没开启iommu的支持，好像只有测试固件才开了

jjm2473, 2024-02-18:
下载这里的23.05测试固件，grub一样修改，应该就行了 https://fw.koolcenter.com/iStoreOS/alpha/x86_64_efi/

zhuozi1212, 2024-02-18:
> 我看了下，目前22.03固件的内核没开启iommu的支持，好像只有测试固件才开了

之后的正式版会留一个在杂项设置里面开启的选项吗？

jjm2473, 2024-02-18:
我看看22.03打开iommu以后兼容性怎么样

jjm2473, 2024-02-18:
@zhuozi1212 试试这个固件，开启了iommu，注意还是要手动修改grub配置文件 https://we.tl/t-WZr9OpXPbi

jjm2473, 2024-02-18:
显卡驱动应该可以不删除的，unbind就行了。 https://www.ibm.com/docs/en/linux-on-z?topic=hs-pci

对于intel，差不多是：
```
lspci -n | grep 00:02.0
echo 0000:00:02.0 > /sys/bus/pci/drivers/i915/unbind
echo 8086 4e55 > /sys/bus/pci/drivers/vfio-pci/new_id
```
注意最后一个id需要改成第一条命令里面显示的

journeyaway, 2024-11-15:
想在istoreos里装的homeassistant直通usb用zigbee网关，怎么都能不成功，lsusb后得到
root@iStoreOS:~# lsusb
Bus 003 Device 003: ID 1a86:7523  USB Serial
然后虚拟机的xml增加并删除前面冲突的bus和port
      <source>
        <vendor id='0x1a86'/>
        <product id='0x7523'/>
        <address bus='3' device='3'/>
      </source>
      <address type='usb' bus='0' port='1'/>
    </hostdev>
启动后提示找不到
Unable to stat /dev/bus/usb/003/003: No such file or directory