istoreos/istoreos #1342: opkg无法更新列表

dune1986, 2024-02-14:
更新22.03.06版后，opkg无法更新列表，反馈如下：

Downloading https://istore.linkease.com/repo/all/compat/Packages.gz
Updated list of available packages in /var/opkg-lists/istore_compat
Downloading https://istore.linkease.com/repo/all/compat/Packages.sig
Signature check passed.
Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/targets/x86/64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/targets/x86/64/packages/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/base/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/base/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/luci/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/luci/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/packages/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/routing/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/routing/Packages.gz

Downloading https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/telephony/Packages.gz
*** Failed to download the package list from https://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.6/packages/x86_64/telephony/Packages.gz

恢复出厂设置故障依旧。浏览器可以访问这个仓库，换了阿里、腾讯、中科大的源后故障依旧。

jjm2473, 2024-02-18:
没发现有这个问题，可能已经修复或你自己的网络配置有问题，建议关闭翻墙插件和去广告插件，重启下再试试。

closed 2024-02-18 by jjm2473

cfwvip, 2024-11-16:
x64版本的确如此

Mooling0602, 2024-11-17:
> 没发现有这个问题，可能已经修复或你自己的网络配置有问题，建议关闭翻墙插件和去广告插件，重启下再试试。

root@ImmortalWrt:~# wget https://istore.linkease.com/repo/all/compat/Packages.gz
--2024-11-17 12:00:16--  https://istore.linkease.com/repo/all/compat/Packages.gz
Resolving istore.linkease.com... 119.188.240.179
Connecting to istore.linkease.com|119.188.240.179|:443... connected.
OpenSSL: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
Unable to establish SSL connection.

Mooling0602, 2024-11-17:
请求reopen该issue @jjm2473