istoreos/istoreos #1343: NPS版本迭代建议

Protein-V, 2024-02-15:
## 关于当前商店的NPS服务端/客户端版本更新迭代的问题

Q：是否用关键词搜索了issue? (使用 "x" 选择)
* [x] 没有类似的issue

A：当前商店的NPS版本原作者久未更新，存在诸多重大漏洞，为此我建议以二开维护后版本替换当前商店的版本，作者yisier自fork原仓库一来在间续推进更新维护，原作存在的鉴权问题以及流量加密漏洞基本已得到修复，现在stars的数量也有1.3k并在不断的攀升，我们不否分原作者的贡献但也不能否定原作存在的bug，特此推荐！

二开维护仓库：[yisier/nps](https://github.com/yisier/nps)

jjm2473, 2024-02-18:
这里的软件包测试过吗，是否直接替换可执行文件就能正常使用 https://github.com/yisier/nps/releases/tag/v0.26.17

Protein-V, 2024-02-18:
> 这里的软件包测试过吗，是否直接替换可执行文件就能正常使用 https://github.com/yisier/nps/releases/tag/v0.26.17

直接替换理论可以继续正常使用的，该版本的服务端多了登录验证码校验开关，应是不影响

AlanWang2333, 2024-03-02:
> > 这里的软件包测试过吗，是否直接替换可执行文件就能正常使用 https://github.com/yisier/nps/releases/tag/v0.26.17
> 
> 直接替换理论可以继续正常使用的，该版本的服务端多了登录验证码校验开关，应是不影响

请问，更换web 目录 和 nps 可执行文件，istoreos是在哪个目录

Protein-V, 2024-04-12:
> > > 这里的软件包测试过吗，是否直接替换可执行文件就能正常使用 https://github.com/yisier/nps/releases/tag/v0.26.17
> > 
> > 
> > 直接替换理论可以继续正常使用的，该版本的服务端多了登录验证码校验开关，应是不影响
> 
> 请问大佬怎么用啊，看到里面说：更换web 目录 和 nps 可执行文件，istoreos是在哪个目录

我自己没有手动进行替换过，只是希望istoreos官方可以加入适配，因为原作者最后的版本已经停止维护近两年了，三大运营商对其的鉴权漏洞查封也越来越严重。

partrick, 2024-11-28:
这个二创版本稳定性不行，目前测试下来经常出现异常退出的情况